espressif/esp-idf #3991: wifi: optimize WiFi rx (732e936a) problem

xiruilin, 2019-08-30:
## Environment

- Development Kit:    ESP32-DevKitC v2
- Module or chip used:    ESP32-WROOM-32
- IDF version:    v4.0-dev-1287-gd7e659df2
- Build System:    CMake
- Compiler version:    1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

@liuzfesp
Hi,
When esp-idf updated from commit d7e659d to commit 45dd617
(esp32-wifi-lib updated from commit e37154f to commit 95edac5),
http2 requests have issues for different routers.

The two routers:
1. TP-LINK TL-WR800N
2. xiaomi router 4C

The following is a modified sample file:
"esp-idf/examples/protocols/http2_request/main/http2_request_example_main.c".

When build with commit d7e659d, all routers work fine and all duration(2.5s) is approach.

When build with commit 45dd617, only No.1 router work fine.
If change delay time to 15000ms in line 100, all routers work fine,
but No.2 router is three times longer than commit d7e659d (7.5s vs 2.5s).

### Code to reproduce this issue

```cpp
/* HTTP2 GET Example using nghttp2
 
   Contacts http2.golang.org and executes the GET/PUT requests. A thin API
   wrapper on top of nghttp2, to properly demonstrate the interactions.

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.

 */
#include <string.h>
#include <stdlib.h>
#include <sys/time.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "lwip/apps/sntp.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "protocol_examples_common.h"
#include "tcpip_adapter.h"

#include "sh2lib.h"

/* The HTTP/2 server to connect to */
#define HTTP2_SERVER_URI  "https://www.163.com"
/* A GET request that keeps streaming current time every second */
#define HTTP2_STREAMING_GET_PATH  "/"

static bool streamClose;
static int64_t duration;

int handle_get_response(struct sh2lib_handle *handle, const char *data, size_t len, int flags)
{
    if (len) {
//        printf("[get-response] %.*s\n", len, data);
    }
    if (flags == DATA_RECV_FRAME_COMPLETE) {
        printf("[get-response] Frame fully received\n");
    }
    if (flags == DATA_RECV_RST_STREAM) {
        printf("[get-response] Stream Closed\n");
        printf("Duration: %lld\n", esp_timer_get_time() - duration);
        streamClose = true;
    }
    return 0;
}

static void set_time(void)
{
    struct timeval tv = {
        .tv_sec = 1509449941,
    };
    struct timezone tz = {
        0, 0
    };
    settimeofday(&tv, &tz);

    /* Start SNTP service */
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_init();
}

static void http2_task(void *args)
{
    /* Set current time: proper system time is required for TLS based
     * certificate verification.
     */
    set_time();
    /* HTTP2: one connection multiple requests. Do the TLS/TCP connection first */
    printf("Connecting to server\n");
    struct sh2lib_handle hd;
    if (sh2lib_connect(&hd, HTTP2_SERVER_URI) != 0) {
        printf("Failed to connect\n");
        vTaskDelete(NULL);
        return;
    }
    printf("Connection done\n");

    while (1) {
        duration = esp_timer_get_time();

        /* HTTP GET  */
        sh2lib_do_get(&hd, HTTP2_STREAMING_GET_PATH, handle_get_response);

        streamClose = false;
        while (!streamClose) {
            /* Process HTTP2 send/receive */
            if (sh2lib_execute(&hd) < 0) {
                printf("Error in send/receive\n");
                goto err_exit;
            }
            vTaskDelay(2);
        }

        vTaskDelay(30000 / portTICK_PERIOD_MS);
    }

err_exit:
    sh2lib_free(&hd);
    vTaskDelete(NULL);
}

void app_main(void)
{
    ESP_ERROR_CHECK( nvs_flash_init() );
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());

    xTaskCreate(&http2_task, "http2_task", (1024 * 32), NULL, 5, NULL);
}

```

negativekelvin, 2019-08-30:
https://github.com/espressif/esp-idf/issues/3956

Alvin1Zhang, 2019-10-11:
@xiruilin Thanks for reporting, the issue fix has been merged, would you please help check whether your issue has been resolved? Thanks.

xiruilin, 2019-10-23:
I'm sorry to have replied so long. 
We have used [cf0caae](https://github.com/espressif/esp-idf/commit/cf0caaec11d9acec1b008c13a10d68e7b6631792) test, everything is fine, thank you for your timely update.
We mainly use the master branch. 
Can you synchronize this update from "release/v4.0" to the master?
Thanks

liuzfesp, 2019-10-28:
@xiruilin sure we are doing it, but it still need some days before it can be ready on master branch because of internal software development process.

xiruilin, 2019-11-21:
@liuzfesp
This problem has been solved, thanks

closed 2019-11-21 by xiruilin