espressif/esp-idf #3993: Error when compiling large programs despite CONFIG_ULP_COPROC_RESERVE_MEM=8192

nicholas1230, 2019-08-30:
Please fix issue discussed here:
[https://github.com/duff2013/ulptool/issues/21](url)
Editing "../ulp/ld/esp32.ulp.ld" worked! 
As explained by this comment:
[https://github.com/duff2013/ulptool/issues/21#issuecomment-458273726](url)

igrr, 2019-08-30:
@nicholas1230 The upper part of RTC_SLOW memory is reserved for ESP-IDF use, this limitation is intentional. You can indeed edit the linker script to take advantage of the remaining space, but we can't guarantee that this will keep working in future ESP-IDF versions.

negativekelvin, 2019-08-30:
Seems like this could be accounted for at compile time such as:
"WARNING: CONFIG_ULP_COPROC_RESERVE_MEM > 4096, automatic app state persistence is DISABLED"

closed 2020-11-03 by espressif-bot