espressif/esp-idf #3995: ESP32 tarefa bloqueia quando uso WiFi e Time ISR

elielmarcos, 2019-08-31:
I set the ISR time to 250us for a task on core 0 (for ADC sampling rate) and Wifi on core 1.

But when I use WiFi communication, I realize that when I come back from the timer interrupt, the task at core 0 takes time to get back to work. And when WiFi communication is not being used, it works perfectly.

Why does this happen?

Alvin1Zhang, 2019-09-01:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

elielmarcos, 2019-09-01:
I am using ESP32 - DOIT with IDE Arduino and Esp IDF 1.0.1.

I am running a WiFi Loop routine (CORE 1) and a task to process an ADC signal (CORE 0).

The ADC processing task, in the example, only prints the value of a counter that increments (every 250us) each time the ISR Timer is called and incremented each time it prints in Serial.

My simplified code:

/*******************************************************************************
LIBRARIES
*******************************************************************************/
#include <math.h>
#include <stdio.h>
#include <esp_system.h>
#include <driver/adc.h>
#include <esp_adc_cal.h>
#include <esp_task_wdt.h>
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <ESPmDNS.h>
#include <HTTPClient.h>
#include <WebServer.h>
#include <DNSServer.h>
#include <FS.h>
#include <SPIFFS.h>

/*******************************************************************************
CONSTANT DEFINITION
******************************************************************************/
// Device Name
const String DEVICE_NAME = "SmartMeter";
// System Password
const char PASSWORD[] = "12345678";
// WebServer Port
const byte WEBSERVER_PORT = 80;
// DNS Server Port
const byte DNSSERVER_PORT = 53;
// Access Point IP
const IPAddress AP_IP (192, 168, 1, 1);
// WebServer Headers
const char WEBSERVER_HEADER_KEYS[] = {"User-Agent", "Cookie"};

/*******************************************************************************
ENTITIES
*******************************************************************************/
// Web Server
WebServer server(WEBSERVER_PORT);
// DNS Server
DNSServer dnsServer;
// HTTP Client
HTTPClient http;

/*******************************************************************************
VARIABLES
*******************************************************************************/
// Attribute Timer
hw_timer_t * timer = NULL;
// Mutex Timer
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
// Timer Interrupt Counter
volatile int interruptCounter = 0;

/*******************************************************************************
TASKS
*******************************************************************************/
void SignalProcess( void *pvParameters );

/*******************************************************************************
INTERRUPTIONS
[... 179 lines cut ...]
9
9
8
7
6
5
4
3
2
1
1
1
1
1
1
1
1
:
:
:

Note that the counter is not decremented while something is being requested via WiFi. At the end it is decremented again and always printed "1".
But the two processes (WiFi and Task SignalProcess) are in different CORE's.

Any solution?

negativekelvin, 2019-09-01:
Go to the forum for this type of question esp32.com

1. configMAX_PRIORITIES is 25 so yours is too low
2. Use a semaphore not poll a counter
3. Yield from the isr

elielmarcos, 2019-09-02:
@negativekelvin

Ok Thanks and Sorry!

Delete topic

closed 2019-09-02 by elielmarcos