espressif/esp-idf #3998: Ethernet not working after upgrade to v4.x

Metmacher, 2019-09-02:
## Environment

- Development Kit: none
- Board: Olimex Gateway Rev. C (Phy is the lan8720 with address 0)
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1416-gba0f4f17e
- Build System:  Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply:   USB

## Problem Description

After update the esp-idf from v3.3 to v4.0 the ethernet won't work anymore. Below is the setup with the ethernet example from the esp-idf.

### Steps to repropduce

Working project ist the esp-idf ethernet example (in v4.0 the basic).

Show working with the esp-idf version 3.3:
[image]

After switching to version 4.0:
Add the following to the ethernet_example_main.c.
`phy_config.phy_addr=0;`
And then:
[image]

I tested it also with the actual master branch with the same result.

negativekelvin, 2019-09-02:
> By default Ethernet driver will assume the PHY address to 1, but you can alway reconfigure this value after eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();

Metmacher, 2019-09-02:
After line 75 which containts `eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();` i add `phy_config.phy_addr=0;` (before the compiler if's)

suda-morris, 2019-09-02:
If you can test the signal on GPIO0? Because from your configuration, Ethernet only get worked when there's a valid RMII clock input to GPIO0. If not, it even can't reset itself. 
And have you also selected the PHY device to LAN8720 from the menuconfig, in the `Example Configuration`?

Metmacher, 2019-09-03:
Yes I set the example configuration to lan8720:
[image]

The signal on GPIO0 looks with esp-idf-v4 the same like with esp-idf-v3

suda-morris, 2019-09-03:
What about enlarge the reset timeout value? `phy_config.reset_timeout_ms=1000;`

negativekelvin, 2019-09-03:
@suda-morris isn't it the Mac reset that times out? 

https://github.com/espressif/esp-idf/blob/f16f4297cd59b981ae39b503cfa491642ef24334/components/esp_eth/include/esp_eth_mac.h#L263

Metmacher, 2019-09-03:
@suda-morris  Yes I also tried this with `phy_config.reset_timeout_ms=3000;`. Nothing changed.

suda-morris, 2019-09-03:
hi @negativekelvin Ah sorry, yes, we need to set the internal mac reset timeout.

suda-morris, 2019-09-03:
> @suda-morris Yes I also tried this with `phy_config.reset_timeout_ms=3000;`. Nothing changed.

Sorry it's my bad, as @negativekelvin mentioned, we should try to enlarge the reset timeout of mac, not phy.

Metmacher, 2019-09-03:
Thats it, after setting `mac_config.sw_reset_timeout_ms = 2000;` it worked. Thank you a lot @negativekelvin  and @suda-morris !

closed 2019-09-03 by Metmacher

negativekelvin, 2019-09-03:
@suda-morris how long does internal Mac reset take and why so long?

suda-morris, 2019-09-03:
If we can make sure the RMII clock is ready at the time we reset internal mac, then 100ms is enough. The timeout value only gets tested in our ESP32-Ethernet-Kit, and it might vary from boards to boards. But I also don't think it will cost such a long time.
For example, if we initialize the Ethernet driver in the following way:
```c
esp_eth_install();// because phy doen't poweron for now, it gets blocked at the reset timeout checking.
do_something_to_poweron_phy();// after this, we can go on with ethernet driver install
```
In this case, the timeout value will looks very big. 
I would like to suggest to move the specific PHY power control operation into a hook provided by https://github.com/espressif/esp-idf/blob/master/components/esp_eth/include/esp_eth.h#L100
So that we can make sure the RMII clock is ready before we start to reset mac.

negativekelvin, 2019-09-03:
In this board I think there is 500ms delay from powerup until ext clk is applied to gpio0 so maybe 100ms is too close.  But there is no phy power gpio.