espressif/esp-idf #4000: Don't warn if chip version is greater than CONFIG_ESP32_REV_MIN

ecc1, 2019-09-02:
The whole point of this configuration variable is that it's a _minimum_ revision. There's no need to warn if the actual version is greater, and indeed "suggest to upgrade it" is not only bad English, it's bad software engineering advice.

https://github.com/espressif/esp-idf/blob/master/components/esp32/cpu_start.c#L411-L418

suda-morris, 2019-09-03:
Thanks @ecc1 for your suggestion. We will change it to normal information log ASAP.

jimmo, 2019-09-20:
@ecc1 I thought the idea was that by setting a higher minimum version (i.e. to match your chip revision) then fewer workarounds for old revisions need to be enabled. That sounds like good advice, especially if the workarounds cost performance or code size.

negativekelvin, 2019-09-20:
It's a good hint, although better to receive this info at compile time like "This build supports ESP32 REV:0+".  Also helpful would be if app binary header had this minimum rev and OTA would not flash an incompatible binary (@projectgus).  Ideally workarounds would be selectively available by compile time config but dynamically enabled at runtime based on detected chip version.  Some do have runtime checks, I think.

igrr, 2019-09-20:
> Also helpful would be if app binary header had this minimum rev and esptool or OTA would not flash an incompatible binary

You seem to have gained access to our internal repository somehow :) This has been merged recently, should be on Github soon.

closed 2019-09-30 by igrr