espressif/esp-idf #4002: esp_wifi_80211_tx does not allow sending data frames when in AP mode

seppestas, 2019-09-03:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-S0WD
- IDF version (run ``git describe --tags`` to find it): v3.1.3
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:         external 5V / Battery

## Problem Description

`esp_wifi_80211_tx` does not allow sending arbitrary WiFi data frames when the WiFi interface is configured in AP mode. When sending the same frame when the interface is configured in STA mode the frame is transmitted successfully and `esp_wifi_80211_tx` returns ESP_OK.

Sending management frames as described in e.g [Jeija/esp32-80211-tx](https://github.com/Jeija/esp32-80211-tx/blob/master/main/main.c) works fine.

### Expected Behavior

`esp_wifi_80211_tx` should be able to send arbitrary IEEE802.11 data frames to allow connectionless communication in either STA or AP mode.

### Actual Behavior

`esp_wifi_80211_tx` returns an `0x102`: `ESP_ERR_INVALID_ARG` error and does not transmit the frame.

### Steps to reproduce

1. Configure the ESP32 as in AP mode
2. Create a custom data frame
3. attempt to send the custom data frame using the `esp_wifi_80211_tx` function

### Code to reproduce this issue

```cpp
uint8_t frame_buffer[200];

// First 4 bytes bytes of the frame. The rest of the frame might be Cisco proprietary and thus fall under NDA.
frame_buffer[0] = 0x08; // Version (B0-1): 00 (802.11), Type (B2-3): 10 (Data) Subtype: 0000 (Data) _ 0000100 = 0x08
frame_buffer[1] = 0x03; // To DS: 1 From DS: 1 More Fragments: 0 Retry: 0, Power Management: 0 More Data: 0, Protected Frame_buffer: 0, +HTC Order: 0

// 2-3 Duration - The Duration field must be set to all zeros.
frame_buffer[2] = 0x00;
frame_buffer[3] = 0x00;

// 4-9 Destination address - Multicast address (1)
// 16-21 BSSID - Unused (3), set to 0.0.0.0
// 24-29 Source address - Unused (4), set to 0.0.0.0

void app_main(void) {
    // ...
    // init stuf
    wifi_config_t ap_config = {
        .ap = {
            .ssid = "esp32-beaconspam",
            .ssid_len = 0,
            .password = "dummypassword",
            .channel = 1,
            .authmode = WIFI_AUTH_WPA2_PSK,
            .ssid_hidden = 1,
            .max_connection = 4,
            .beacon_interval = 60000
        }
    };

    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &ap_config));
    ESP_ERROR_CHECK(esp_wifi_start());
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));

    // Note: actual size is calculated
    free80211_send(frame_buffer, sizeof(frame_buffer));
}
```

seppestas, 2019-09-03:
Note that the QoS Control field (bit 7) is set to 0, ergo according to [the API description](https://github.com/espressif/esp-idf/blob/v3.1.3/docs/en/api-guides/wifi.rst#wi-fi-80211-packet-send) `esp_wifi_80211_tx` should be able to send "the non-QoS data frame".

I did not yet test this on the newest (stable or master) version of ESP-IDF since updating the configuration takes time and no relevant changes appear in the changelog. Sadly I don't have a lot more time I can spend on this.

esp-HarveyRong, 2020-03-17:
Hi @seppestas, 
May I ask what is your specific version number ? E.g. commit ID
I used the latest v3.1 and did not reproduce the issue.

liuzfesp, 2020-03-25:
Hi @seppestas I think the invalid argument error is because you call `esp_wifi_80211_tx()` to send the packets to AP interface while the WiFi mode is in STA mode.

For `esp_wifi_80211_tx()`, if you want to send the packet to AP interface, you should call `esp_wifi_set_mode()` to set ESP32 to AP or AP + STA mode first. If you want to send the packet to STA interface, you should call `esp_wifi_set_mode()` to set ESP32 to STA or AP + STA mode first. 

Could you help to check it again?

Alvin1Zhang, 2020-08-20:
@seppestas Thanks for reporting, would you please help share if any updates for this issue? Thanks.

seppestas, 2020-08-20:
Hi

I'm afraid I'm not working on the project I used the ESP32 for anymore and I currently don't have time to look into this.

It looks like `esp_wifi_80211_tx ` changed and no other people seem to be having the same issue. I'll close this issue. Thanks for looking into it.

closed 2020-08-20 by seppestas