espressif/esp-idf #4003: 使用框架带的例子（esp-idf\examples\bluetooth\bluedroid\ble\ble_spp_server），改蓝牙名字不成功

867339205, 2019-09-03:
改变项目esp-idf\examples\bluetooth\bluedroid\ble\ble_spp_server的ble_spp_server.c函数中宏定义SAMPLE_DEVICE_NAME为其他的值，结果重新编译烧写，蓝牙的名字还是原来的名字，并没有改变

Alvin1Zhang, 2019-09-03:
@867339205  Thanks for reporting the issue, please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md). Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

867339205, 2019-09-04:
> @867339205 Thanks for reporting the issue, please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md). Information [...]

Environment
Development Kit: none
Module or chip used: ESP32-WROVER-B
IDF version (run git describe --tags to find it): v4.0-dev-1443-g39f090a4f
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it):(crosstool-NG esp32-2019r1) 8.2.0
Operating System: Windows x64
Power Supply: external 3.3V

gengyuchao, 2019-09-17:
这确实是一个问题，在这个example中使用了 esp_ble_gap_config_adv_data_raw((uint8_t *)spp_adv_data, sizeof(spp_adv_data)); 函数来设置广播数据，这会在广播开始的时候使用 spp_adv_data 中的设置进行广播，而这个数组中设备名称的部分不会被SAMPLE_DEVICE_NAME定义改变。
现在你可以修改 spp_adv_data 中的数据来修改扫描到的设备名称，例如：

```
static const uint8_t spp_adv_data[23] = {
    /* flags */
    0x02,0x01,0x06,
    /* tx power*/
    0x03,0x03,0xF0,0xAB,
    /* device name */
    0x0F,0x09, 'E', 'S', 'P', '_', 'S', 'S', 'P', '_', 'S', 'E', 'R','V', 'E', 'R'
};
```

Alvin1Zhang, 2020-08-20:
Thanks for reporting, sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/a16d7954486b663b7449ac07fde088d4c9f8f5f4, feel free to reopen if the issue still happens. Thanks.

closed 2020-08-20 by Alvin1Zhang