espressif/esp-idf #4004: update the esp32-bt-lib submodule in the relase/v4.0 branch

enricop, 2019-09-03:
@Weijian-Espressif 

Hi, would it be possible to align the [esp32-bt-lib](https://github.com/espressif/esp32-bt-lib) component to the last commit also in the relase/v4.0 branch?

In our project we would like to target the 4.0 relase and since we are using BLE extensively need the recent fixes ( [1](https://github.com/espressif/esp32-bt-lib/commit/717f0c6ec71a016a0b292acffeacf239d007b8ff) and [2](https://github.com/espressif/esp32-bt-lib/commit/a45ff23f0f6682caeb40dbb673f9dc54f0f8e1cd) ) of the ble controller submodule library.

Many thanks for your support
Regards,
Enrico

enricop, 2019-09-10:
any update? thanks

Weijian-Espressif, 2019-09-10:
we have updated the bt lib, please update to the latest release/v4.0 . The bt lib commit id is  
commit a45ff23f0f6682caeb40dbb673f9dc54f0f8e1cd (origin/bugfix/btdm_fix_ble_data_loss_and_send_same_pkt_twice, bugfix/btdm_fix_ble_data_loss_and_send_same_pkt_twice)
Author: zhiweijian <zhiweijian@espressif.com>
Date:   Mon Aug 26 11:13:07 2019 +0800

    component/bt: fix ble data loss and send same pkt twice(a482cdac)

This lib include ``bugfix btdm sleep twice after wakeup request ``.

enricop, 2019-09-10:
@Weijian-Espressif  Thank you very much!

closed 2019-09-10 by enricop