espressif/esp-idf #4005: esp_wifi_80211_tx does not allow sending on a specific channel

seppestas, 2019-09-03:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-S0WD
- IDF version (run ``git describe --tags`` to find it): v3.1.3
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:         external 5V / Battery

## Problem Description

When using the `esp_wifi_80211_tx` function to send arbitrary WiFi frames with the WiFi interface configure in STA mode it is not possible to change the WiFi channel on which the frame is sent. I have tried to change the channel using the `esp_wifi_set_config` with interface set to `WIFI_IF_AP` and `conf->sta.channel` to different channels (tested with channel 6 and 11) and I have tried the `esp_wifi_set_channel` function, but neither work to change the channel on which `esp_wifi_80211_tx` sends the frames.

This issue is similar to #3210, but here @ingframin talks about sending "beacons", presumably management frames sent from a device set up as an access point. I'm trying to send non-QoS data frames (WDS Frames).

### Expected Behavior

`esp_wifi_80211_tx` should be able to send frames on an arbitrary WiFi channel. Either by using the channel configured with `esp_wifi_set_config` or using a different, documented method.

### Actual Behavior

`esp_wifi_80211_tx` always sends frames on channel 1.

### Steps to repropduce

1. Configure the WiFi interface in STA mode
2. Set the STA interface channel to a channel different than 1 using `esp_wifi_set_config`
3. Create a non-QoS data frame
4. Send the data frame using the `esp_wifi_80211_tx` function

### Code to reproduce this issue

```cpp
uint8_t frame_buffer[200];

// First 4 bytes bytes of the frame. The rest of the frame might be Cisco proprietary and thus fall under NDA.
frame_buffer[0] = 0x08; // Version (B0-1): 00 (802.11), Type (B2-3): 10 (Data) Subtype: 0000 (Data) _ 0000100 = 0x08
frame_buffer[1] = 0x03; // To DS: 1 From DS: 1 More Fragments: 0 Retry: 0, Power Management: 0 More Data: 0, Protected Frame_buffer: 0, +HTC Order: 0

// 2-3 Duration - The Duration field must be set to all zeros.
frame_buffer[2] = 0x00;
frame_buffer[3] = 0x00;

// 4-9 Destination address - Multicast address (1)
// 16-21 BSSID - Unused (3), set to 0.0.0.0
// 24-29 Source address - Unused (4), set to 0.0.0.0

void app_main(void) {
    // ...
    // init stuf
    wifi_config_t ap_config = {
        .sta = {
            .channel = 6,
        }
    };

    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &ap_config));
    ESP_ERROR_CHECK(esp_wifi_start());
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));

    // Note: actual size is calculated
    free80211_send(frame_buffer, sizeof(frame_buffer), false);
}
```

Alvin1Zhang, 2019-09-03:
@seppestas Thanks for reporting this. We will take a look at. Thanks.

seppestas, 2019-09-03:
It looks like my use-case is not that secret, or at least there seems to be some information publicly available here: https://www.cisco.com/c/en/us/td/docs/solutions/Enterprise/Mobility/WiFiLBS-DG/wifich6.html

Since my implementation is based on a partner level document I'm afraid I can't share my code publicly, but this should give some more indication about the use-case and the sort of frames I'm trying to send.

It also showcases the need of being able to send frames on different channels.

negativekelvin, 2019-09-03:
Have you tried esp_wifi_set_channel in WiFi null mode with/without promiscuous rx enabled?

seppestas, 2019-09-05:
Hi @negativekelvin 

I now tried putting the WiFi interface in promiscuous mode, however when sending the frame I get the same result as when trying to send the frame in AP mode: `esp_wifi_80211_tx` returns an error code.

However, I found that putting the channel in promiscuous mode, setting the channel using and then switching to station mode seems to work. I will do some more testing and report any workaround I find if it works.

I also changed the way I use the `esp_wifi_set_channel`. before I used `WIFI_SECOND_CHAN_BELOW`, now I use `WIFI_SECOND_CHAN_NONE`. It might be this caused it to work.

esp-HarveyRong, 2019-11-07:
Hi @seppestas Has the problem been solved? If not, I will deal with it in time.

Alvin1Zhang, 2020-01-15:
@seppestas Thanks for reporting. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

seppestas, 2020-01-15:
@Alvin1Zhang yes, I got this working eventually but had to use a workaround I believe. I’ll have a look to see if I can find back the code, but I’m afraid I now work for a different company.

I’ll close this issue for now.

closed 2020-01-15 by seppestas

esp-HarveyRong, 2020-03-17:
Hi @seppestas, 
Like #4002 , what is your specific version number? E.g. commit ID
I used the latest v3.1 and did not reproduce the issue.

liuzfesp, 2020-03-24:
This issue has been fixed, now you can call `esp_wifi_set_channel()' to configure the channel number.

- master - 36c726de
- v4.1 - 4d10eb20
- v3.3 - c9e7a04a

The fix will be back-ported to v3.1/v3.2/v4.0 soon.