espressif/esp-idf #4006: esp32 will not run code when powered by 3.3vpin it requires pushing reset button to work

athleticmistud, 2019-09-03:
our issue is this,  we are using a:
wemos D1 pin d4 connecting to a D 32 with pin 16 and the power from the 
D 32 is coming off the 3.3 V and ground on the d1 when we powered that way with the USB being plugged into the D1 we cannot make a connection to a Bluetooth speaker without pressing the reset button on the d32.  then something from pressing the reset does someting and then it will connect to a bluetooth speaker,

however if I power in through the usb on the D 32 and then power the d1 from the D 32  from 3.3v and ground i could connect to a Bluetooth speaker not requiring the reset button to be pushed which is what we need as we can not access the board normally so pushing the reset button is not an option

[image]
[image]
[natee.txt](https://github.com/espressif/esp-idf/files/3569676/natee.txt)
[nate.txt](https://github.com/espressif/esp-idf/files/3569682/nate.txt)

----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2019-09-03:
Hardware questions belong on the forum esp32.com.  you can try adding a capacitor between EN and GND.

athleticmistud, 2019-09-03:
Sorry I was told by one of your engineers to post it here.

I have a cap already
What can we put in the code thst would do the same as a reset?

Do you have 3.2 parsed for Arduino ide I need help with that

Thanks
Nick

negativekelvin, 2019-09-03:
You have a capacitor on 3.3v but not on EN?

athleticmistud, 2019-09-04:
You mean like this??  Which is + and which gets the - of the cap?

Alvin1Zhang, 2019-09-04:
@athleticmistud Thanks for reporting this. @negativekelvin Thanks for the notes.

closed 2019-09-04 by Alvin1Zhang