espressif/esp-idf #4008: ESP-IDF on WSL

tatsutaigu, 2019-09-04:
Hi,

I'm using ESP-IDF on Windows Subsystem on Linux.
I would like to report it.

stable version(ESP-IDF : v3.2.2)
- cannot run openOCD. 
 WSL cannot find FTDI device on USB.
 I'm using FT2232D USB-UART for JTAG adapter.
 If it can resolve it, windows users are free from slowly mingw system and toolchain-2.0.
 or, if you know JTAG adapter which can use on WSL, please let me know.

latest version(ESP-IDF : master)
- cannot run openOCD. 
- cannot flash
  idf.py(esptool.py?) can find and connect ESP32 via COM port.
  but, it cannot flash it.

> tatsutaigu@ABCDE-FGHI:~/esp/master_pj/hello_world$ idf.py -p /dev/ttyS20 flash
> Checking Python dependencies...
> Python requirements from /home/tatsutaigu/esp/esp-idf/requirements.txt are satisfied.
> Adding flash's dependency "all" to list of actions
> Executing action: flash
> Running esptool.py in directory /home/tatsutaigu/esp/master_pj/hello_world/build
> Executing "/home/tatsutaigu/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyS20 -b 460800 --after hard_reset write_flash @flash_project_args"...
> esptool.py -p /dev/ttyS20 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
> esptool.py v2.8-dev
> Serial port /dev/ttyS20
> Connecting.......
> Detecting chip type... ESP32
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 30:ae:a4:ee:bc:60
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 460800
> Changed.
> Traceback (most recent call last):
>   File "/home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3167, in <module>
>     _main()
>   File "/home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 3160, in _main
>     main()
>   File "/home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2882, in main
>     esp.change_baud(args.baud)
>   File "/home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 104, in inner
>     return func(*args, **kwargs)
>   File "/home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 707, in change_baud
>     self._set_port_baudrate(baud)
>   File "/home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 257, in _set_port_baudrate
>     self._port.baudrate = baud
>   File "/home/tatsutaigu/.espressif/python_env/idf4.1_py2.7_env/local/lib/python2.7/site-packages/serial/serialutil.py", line 295, in baudrate
>     self._reconfigure_port()
>   File "/home/tatsutaigu/.espressif/python_env/idf4.1_py2.7_env/local/lib/python2.7/site-packages/serial/serialposix.py", line 438, in _reconfigure_port
>     [iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
> termios.error: (5, 'Input/output error')
> esptool.py failed with exit code 1

JMike360, 2019-09-05:
I'm having the exact same issue. Using esptool.py on WSL to flash a micropython .bin file to an esp32 board fails with the exact same serialutil.py traceback and termios.error.

$:  esptool.py --chip esp32 --port /dev/ttyS6 erase_flash
works fine, but

$: esptool.py --chip esp32 --port /dev/ttyS6 --baud 460800 write_flash -z 0x1000 esp32-20190905-v1.11-274-g06661890d.bin

fails with the same traceback and termios.error messages as the OP

projectgus, 2019-09-06:
Hi @tatsutaigu @JMike360 ,

Although we don't specifically support WSL, we'd be happy to find ways to get these features to work.

> cannot run openOCD.
> WSL cannot find FTDI device on USB.
> I'm using FT2232D USB-UART for JTAG adapter.

OpenOCD uses libusb to access USB devices. Unfortunately it looks like WSL does not yet support libusb:
https://wpdev.uservoice.com/forums/266908-command-prompt-console-windows-subsystem-for-l/suggestions/31920928-add-libusb-support-to-wsl

So running openocd as a native windows .exe is the only option for now. The good news is that because openocd is just a daemon and all communication is via TCP sockets, you should be able to run Windows openocd in a separate Command Prompt window, put that window to the background, and then use everything else inside WSL.

> idf.py(esptool.py?) can find and connect ESP32 via COM port.
> but, it cannot flash it.

```
 File "/home/tatsutaigu/.espressif/python_env/idf4.1_py2.7_env/local/lib/python2.7/site-packages/serial/serialutil.py", line 295, in baudrate
self._reconfigure_port()
File "/home/tatsutaigu/.espressif/python_env/idf4.1_py2.7_env/local/lib/python2.7/site-packages/serial/serialposix.py", line 438, in _reconfigure_port
[iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
termios.error: (5, 'Input/output error')
```

This seems like a limitation of WSL not supporting some aspect of `tcsetattr()` , as this is what the underlying pyserial library calls on POSIX:
https://github.com/pyserial/pyserial/blob/v3.4/serial/serialposix.py#L435

The [WSL serial implementation](https://blogs.msdn.microsoft.com/wsl/2017/04/14/serial-support-on-the-windows-subsystem-for-linux/) seems to encourage use to termios controls to configure the serial port, so it seems like this should work.

Probably there is a minimal test case like this:

```
#!/usr/bin/env python
import serial
p = serial.Serial("/dev/ttyS20")
p.baudrate = 9600
print("Changed baud rate!")
```
... does that also crash the same way?

 If so, you could try reporting this to WSL and see if they have any ideas: https://github.com/Microsoft/WSL/issues/

Otherwise you can try the workaround of using the default baud rate, just a bit slow to flash. :(

> stable version(ESP-IDF : v3.2.2)

It's interesting that esptool.py can flash on the stable ESP-IDF, maybe there is an older pyserial version that works on WSL. How did you install `pyserial` when setting up version 3.2.2?

tatsutaigu, 2019-09-09:
Hi @projectgus
 
**about GDB**
I tried openocd on mingw32 and gdb on WSL.
gdb on WSL can connect to openocd mingw32.
however, gdb on WSL stacked.

Here is gdb message on WSL.
> tatsutaigu@ABCDE-FGHI:~/esp/hello_world$ xtensa-esp32-elf-gdb -x gdbinit ./build/hello-world.elf
> GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
> Copyright (C) 2015 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from ./build/hello-world.elf...done.
> 0x40009081 in ?? ()
> JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
> esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
> Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
> esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
> esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
> esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
> Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
> Hardware assisted breakpoint 1 at 0x400d216c: file /home/tatsutaigu/esp/hello_world/main/hello_world_main.c, line 17.

gdb message is stopped this line.
it does not show (gdb).

Here is openocd message on mingw32.
> tatsutaigu@ABCDE-FGHI MINGW32 ~/esp/openocd-esp32
> $ ./bin/openocd.exe -s ./share/openocd/scripts/ -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp-wroom-32.cfg
> Open On-Chip Debugger  v0.10.0-esp32-20190708 (2019-07-08-11:04)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> none separate
> adapter speed: 20000 kHz
> Info : Configured 2 cores
> esp32 interrupt mask on
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
> Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
> Info : clock speed 20000 kHz
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : Target halted. PRO_CPU: PC=0x40009081 (active)    APP_CPU: PC=0x400076E2
> Info : Listening on port 3333 for gdb connections
> Info : accepting 'gdb' connection on tcp/3333
> Error: No symbols for FreeRTOS
> Info : Target halted. PRO_CPU: PC=0x400916EE (active)    APP_CPU: PC=0x400076E2
> Info : Flash mapping 0: 0x10020 -> 0x3f400020, 29 KB
> Info : Flash mapping 1: 0x20018 -> 0x400d0018, 73 KB
> Info : Target halted. PRO_CPU: PC=0x400916EE (active)    APP_CPU: PC=0x400076E2
> Info : Auto-detected flash size 4096 KB
> Info : Using flash size 4096 KB
> Info : Target halted. PRO_CPU: PC=0x400916EE (active)    APP_CPU: PC=0x400076E2
> Info : Flash mapping 0: 0x10020 -> 0x3f400020, 29 KB
> Info : Flash mapping 1: 0x20018 -> 0x400d0018, 73 KB
> Info : Using flash size 76 KB
> Info : Target halted. PRO_CPU: PC=0x400916EE (active)    APP_CPU: PC=0x400076E2
> Info : Flash mapping 0: 0x10020 -> 0x3f400020, 29 KB
> Info : Flash mapping 1: 0x20018 -> 0x400d0018, 73 KB
> Info : Using flash size 32 KB
> Warn : negative reply, retrying
> Warn : acknowledgment received, but no packet pending
> undefined debug reason 7 - target needs reset
> Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
> Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
> Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
> Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
> Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
> Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
> Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400

**stable version(ESP-IDF : v3.2.2)** 
 I follow below URL on WSL.
[https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#get-esp-idf](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#get-esp-idf)
I did not do any special installation.

tatsutaigu, 2019-09-10:
Hi @projectgus 

**about GDB**
I tried openocd on mingw32 and gdb on WSL again.
I could debug with openocd on mingw32 and gdb on WSL.

Thank you for your good information.

However, root cause is unknown...

> tatsutaigu@ABCDE-FGHI:~/esp/hello_world$ xtensa-esp32-elf-gdb -x gdbinit ./build/hello-world.elfGNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10Copyright (C) 2015 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from ./build/hello-world.elf...done.
> 0x400e24ee in esp_pm_impl_waiti () at /home/tatsutaigu/esp/esp-idf-v3.2/components/esp32/pm_esp32.c:487
> 487         asm("waiti 0");
> JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
> esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
> Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
> esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
> esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
> esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
> Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
> Hardware assisted breakpoint 1 at 0x400d216c: file /home/tatsutaigu/esp/hello_world/main/hello_world_main.c, line 17.
> Target halted. PRO_CPU: PC=0x400D216C (active)    APP_CPU: PC=0x400E24EE
> [New Thread 1073437944]
> [New Thread 1073436044]
> [New Thread 1073440600]
> [New Thread 1073428252]
> [New Thread 1073412856]
> [New Thread 1073413596]
> [New Thread 1073429384]
> [Switching to Thread 1073433928]
> 
> Temporary breakpoint 1, app_main () at /home/tatsutaigu/esp/hello_world/main/hello_world_main.c:17
> 17      {
> (gdb)

tatsutaigu, 2019-09-10:
Hi @projectgus 

>Probably there is a minimal test case like this:
```
#!/usr/bin/env python
import serial
p = serial.Serial("/dev/ttyS20")
p.baudrate = 57600
print("Changed baud rate!")
```
> ... does that also crash the same way?

It is no problem.
I hope latest esptool.py removes`tcsetattr()` because esptool.py v2.6 can flash.

projectgus, 2019-09-10:
> I tried openocd on mingw32 and gdb on WSL again.
> I could debug with openocd on mingw32 and gdb on WSL.
>
> Thank you for your good information.
>
> However, root cause is unknown...

I'm glad you're able to debug now. There are a few posts here that say different things, so I'm not sure what "root cause" you are referring to?

> I hope latest esptool.py removestcsetattr() because esptool.py v2.6 can flash.

esptool doesn't call this code directly, the third party library pyserial calls this code: https://github.com/pyserial/pyserial/

If you can find a way to install the same version of pyserial via `pip` that is also installed when you run under IDF v3.2.2 (this version is probably the version of pyserial installed via apt-get), then it may work.

tatsutaigu, 2019-09-10:
I'm sorry for confusion.
"root cause" means
I cannot understand reason why openocd on mingw32 and gdb on WSL failed previous trying.
It may be I forget to reset ESP32 after flashing. 

```
latest version(ESP-IDF : master)
- cannot flash
   idf.py(esptool.py?) can find and connect ESP32 via COM port.
   but, it cannot flash it.
```
esptool.py on latest ESP-IDF failed at detecting flash size.

trace log is here.

> Running stub...
> TRACE +0.001 command op=0x06 data len=8 wait_response=1 timeout=0.050 data=0000000068f50940
> TRACE +0.000 Write 18 bytes:
>     c000060800000000 000000000068f509 | .............h..
>     40c0                              | @.
> TRACE +0.015 Read 1 bytes: c0
> TRACE +0.001 Read 19 bytes:
>     010604006c879bc4 00000000c0c04f48 | ....l.........OH
>     4149c0                            | AI.
> TRACE +0.009 Received full packet: 010604006c879bc400000000
> TRACE +0.000 Received full packet: 4f484149
> Stub running...
> Changing baud rate to 460800
> TRACE +0.000 command op=0x0f data len=8 wait_response=1 timeout=3.000 data=0008070000c20100
> TRACE +0.000 Write 18 bytes:
>     c0000f0800000000 000008070000c201 | ................
>     00c0                              | ..
> TRACE +0.004 Read 1 bytes: c0
> TRACE +0.001 Read 11 bytes: 010f0200000000000000c0
> TRACE +0.001 Received full packet: 010f0200000000000000
> Changed.
> Configuring flash size...
> TRACE +0.052 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=1c200060
> TRACE +0.000 Write 14 bytes: c0000a0400000000001c200060c0
> TRACE +3.001 Timed out waiting for packet header
> 
> A fatal error occurred: Timed out waiting for packet header

And, I switched IDF ver. on WSL.
pyserial ver. on ESP-IDF v3.2.2 is same as latest ESP-IDF master.
There are v3.4.

> tatsutaigu@ABCDE-FGHI:~/esp/esp-idf-v3.2$ pip freeze | grep pyserial
> pyserial==3.4
>

tatsutaigu, 2019-09-10:
I could flash by latest ESP-IDF(master).
it must set baud rate 921600.

please revise idf.py default baud rate 460800 to 921600.
and, please add `idf.py menuconfig` -> `serial flasher config` to `Default baud rate`.

> tatsutaigu@ABCDE-FGHI:~/esp/master_pj/hello_world$ idf.py -p /dev/ttyS20 **-b 921600** flash
> Checking Python dependencies...
> Python requirements from /home/tatsutaigu/esp/esp-idf/requirements.txt are satisfied.
> Adding flash's dependency "all" to list of actions
> Executing action: flash
> Running esptool.py in directory /home/tatsutaigu/esp/master_pj/hello_world/build
> Executing "/home/tatsutaigu/.espressif/python_env/idf4.1_py2.7_env/bin/python /home/tatsutaigu/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyS20 -b 921600 --after no_reset write_flash @flash_project_args"...
> esptool.py -p /dev/ttyS20 -b 921600 --after no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
> esptool.py v2.8-dev
> Serial port /dev/ttyS20
> Connecting.....
> Detecting chip type... ESP32
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 24:0a:c4:9b:87:6c
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Configuring flash size...
> Compressed 3072 bytes to 82...
> Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 1535.0 kbit/s)...
> Hash of data verified.
> Compressed 24752 bytes to 14733...
> Wrote 24752 bytes (14733 compressed) at 0x00001000 in 0.2 seconds (effective 1123.3 kbit/s)...
> Hash of data verified.
> Compressed 147152 bytes to 76622...
> Wrote 147152 bytes (76622 compressed) at 0x00010000 in 1.2 seconds (effective 1021.8 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Staying in bootloader.
> Executing action: all (aliases: build)
> Running ninja in directory /home/tatsutaigu/esp/master_pj/hello_world/build
> Executing "ninja all"...
> [1/3] Performing build step for 'bootloader'
> ninja: no work to do.
> Done
>

projectgus, 2019-09-10:
> I could flash by latest ESP-IDF(master).
> it must set baud rate 921600.

That is strange. Can you please repeat the simple Python test from earlier comment using both 460800 and 921600 instead of 57600 and report the results?

> please revise idf.py default baud rate 460800 to 921600.
> and, please add idf.py menuconfig -> serial flasher config to Default baud rate.

We have this set conservatively because not all development boards work at 921600 or higher.

You can override the default value on your local machine by setting the `ESPBAUD` environment variable in your `.profile` or `.bashrc` file. If this is set, the value in menuconfig is ignored.

tatsutaigu, 2019-09-10:
**460800 is NG.**
> $ ../../esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyS20 -b 460800 --after no_reset write_flash --flash_mode dio --flash_size
> 4MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello-world.bin
> esptool.py v2.8-dev
> Serial port /dev/ttyS20
> Connecting........_
> Detecting chip type... ESP32
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 24:0a:c4:9b:87:6c
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 460800
> Changed.
> Configuring flash size...
> 
> A fatal error occurred: Timed out waiting for packet header

**921600 is OK.**
> $ ../../esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyS20 -b 921600 --after no_reset write_flash --flash_mode dio --flash_size4MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello-world.bin
> esptool.py v2.8-dev
> Serial port /dev/ttyS20
> Connecting........__
> Detecting chip type... ESP32
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 24:0a:c4:9b:87:6c
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Configuring flash size...
> Compressed 24752 bytes to 14733...
> Wrote 24752 bytes (14733 compressed) at 0x00001000 in 0.2 seconds (effective 1123.0 kbit/s)...
> Hash of data verified.
> Compressed 3072 bytes to 82...
> Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 1496.4 kbit/s)...
> Hash of data verified.
> Compressed 147152 bytes to 76622...
> Wrote 147152 bytes (76622 compressed) at 0x00010000 in 1.2 seconds (effective 1022.1 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Staying in bootloader.
> 

**57600 is OK.**
> $ ../../esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyS20 -b 921600 --after no_reset write_flash --flash_mode dio --flash_size4MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello-world.bin
> esptool.py v2.8-dev
> Serial port /dev/ttyS20
> Connecting........__
> Detecting chip type... ESP32
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 24:0a:c4:9b:87:6c
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Configuring flash size...
> Compressed 24752 bytes to 14733...
> Wrote 24752 bytes (14733 compressed) at 0x00001000 in 0.2 seconds (effective 1123.0 kbit/s)...
> Hash of data verified.
> Compressed 3072 bytes to 82...
> Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 1496.4 kbit/s)...
> Hash of data verified.
> Compressed 147152 bytes to 76622...
> Wrote 147152 bytes (76622 compressed) at 0x00010000 in 1.2 seconds (effective 1022.1 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Staying in bootloader.
> 

Thank you for your good information.
I could set default value 921600 by `ESPBAUD`.

tatsutaigu, 2019-09-10:
I'm sorry, I misunderstand.

460800, 921600 and 57600 are OK. 

> tatsutaigu@ABCDE-FGHI:~$ cat ./test.sh
> #!/usr/bin/env python
> import serial
> p = serial.Serial("/dev/ttyS20")
> p.baudrate = 460800
> print("Changed baud rate!")
> tatsutaigu@ABCDE-FGHI:~$ ./test.sh
> Changed baud rate!

> tatsutaigu@ABCDE-FGHI:~$ cat ./test.sh
> #!/usr/bin/env python
> import serial
> p = serial.Serial("/dev/ttyS20")
> p.baudrate = 921600
> print("Changed baud rate!")
> tatsutaigu@ABCDE-FGHI:~$ ./test.sh
> Changed baud rate!
 
> tatsutaigu@ABCDE-FGHI:~$ cat ./test.sh
> #!/usr/bin/env python
> import serial
> p = serial.Serial("/dev/ttyS20")
> p.baudrate = 57600
> print("Changed baud rate!")
> tatsutaigu@ABCDE-FGHI:~$ ./test.sh
> Changed baud rate!

projectgus, 2019-09-10:
@tatsutaigu Interesting, thanks!

windowsair, 2021-02-26:
Same question. Specifying the baud rate to `921600` solved this issue.

MichaelDu9226, 2021-03-31:
Same question too. Specifying the baud rate to 921600 solved this issue.

tawalke, 2021-08-18:
I am having the same issue. If I flash with 460800 I have issue but if I flash with any other baud  baud rate, it works fine.  Any update on the fix for this?

jakub-kocka, 2024-01-22:
I have been testing WSL flashing with ESP32 and ESP32-C2 and openocd with ESP32-WROVER-KIT.
I haven't been able to reproduce the described issue.

For flashing it was sufficient to use the esp_rfc2217_server.py script (from [esptool](https://github.com/espressif/esptool/tree/master)) in the host, connect to the port from WSL and everything seems to work as expected.

For openocd I have done port forwarding with USBIPD (https://learn.microsoft.com/en-us/windows/wsl/connect-usb)
To be able to use COM port such as /dev/ttyUSBX and it also works as expected, including flashing.

IMHO I don't see the benefits of using WSL on nowadays Windows with PowerShell included. For example on my machine using WSL is multiple times slower. Also, this use case is for a bit of experienced users, so I think we don't have to edit the documentation. However, it seems to be possible and working so I believe it can be useful for someone.
We will close this issue for now.