espressif/esp-idf #4009: 32d will not connect to bluetooth with out reset button when powered by anything but micro usb

athleticmistud, 2019-09-04:
-wemos D1 pin d4 connecting to a D 32 with pin 16 and the power from the
D 32 is coming off the 3.3 V and ground on the d1 when we powered that way with the USB being plugged into the D1 we cannot make a connection to a Bluetooth speaker without pressing the reset button on the d32. then something from pressing the reset does someting and then it will connect to a bluetooth speaker,

however if I power in through the usb on the D 32 and then power the d1 from the D 32 from 3.3v and ground i could connect to a Bluetooth speaker not requiring the reset button to be pushed which is what we need as we can not access the board normally so pushing the reset button is not an option
[3.0main.txt](https://github.com/espressif/esp-idf/files/3573306/3.0main.txt)

athleticmistud, 2019-09-05:
[naten2.txt](https://github.com/espressif/esp-idf/files/3577632/naten2.txt)

i attached serial monitor

here is the data from serial monitor on the d32 here is what we did

i load code on d32 

then load our code on d1   only connection between the 2 is on pin 16 on d32
our device connects to speaker but no sound
then i press our test button on our config page our light works but no sound
then i press the reset button on the d 32 the d32 disconnects and reconnects to the speaker
then i press our test button on our config page and it works both the light and sound play 

so it is something to do when i press reset on the d32 that fixes the code issue?
  i attached it here

esp-wzh, 2025-04-29:
Hi, @athleticmistud  Not sure if this problem still exists. The information you provided is not sufficient, but from the description, the problem should be on the application layer, because resetting esp32 can solve the problem. If D32 is only used as a slave of wemos D1 for bluetooth communication, perhaps you can use an IO of D1 to control D1's EN to enable and disable the esp32 module.

Alvin1Zhang, 2025-04-30:
Thanks for reporting and sorry for the very slow turnaround, feel free to reopen.

closed 2025-04-30 by Alvin1Zhang