espressif/esp-idf #4010: Cache disabled when copying from SPIFFS to SPIRAM

HanVertegaal, 2019-09-04:
#3401  Environment

- Module or chip used:  Analog Lamb ALB32
- IDF version: v3.3-rc-dirty (modified portmalloc.h)
- Build System:         Make
- Compiler version: 1.22.0-98-g4638c4f
- Operating System:     Windows/WSL
- Power Supply:         Battery

## Problem Description

When I copy data from SPIFFS to SPIRAM, the software panics.

### Expected Behavior

No panic

### Actual Behavior

`
Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:4248
load:0x40078000,len:10996
load:0x40080400,len:6112
entry 0x400806e0
I (208) psram: This chip is ESP32-D0WD
I (209) spiram: Found 32MBit SPI RAM device
I (209) spiram: SPI RAM mode: flash 80m sram 40m
I (211) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (219) cpu_start: Pro cpu up.
I (222) cpu_start: Application information:
I (227) cpu_start: Project name:     aether
I (232) cpu_start: App version:      1
I (236) cpu_start: Compile time:     Sep  4 2019 10:31:10
I (243) cpu_start: ELF file SHA256:  953a24fc02961b39...
I (248) cpu_start: ESP-IDF:          v3.3-rc-dirty
I (254) cpu_start: Starting app cpu, entry point is 0x400813e0
0x400813e0: call_start_cpu1 at /home/han/esp/esp-idf-v3.3-rc/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
D (264) memory_layout: Checking 8 reserved memory ranges:
D (269) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
D (276) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (282) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb4690
D (289) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (295) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (302) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (308) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/han/esp/esp-idf-v3.3-rc/components/freertos/xtensa_vectors.S:1779

D (314) memory_layout: Reserved memory range 0x40080000 - 0x4008ef63
0x40080000: _WindowOverflow4 at /home/han/esp/esp-idf-v3.3-rc/components/freertos/xtensa_vectors.S:1779

D (321) memory_layout: Building list of available memory regions:
D (327) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (334) memory_layout: Available memory region 0x3ffb4690 - 0x3ffb8000
D (340) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
D (347) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
D (353) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
D (360) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
D (367) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (373) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (380) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
[... 89 lines cut ...]

0x400eea39: std::__basic_file<char>::xsgetn(char*, int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/esp32-psram/libstdc++-v3/src/c++98/basic_file.cc:211

0x400ef01a: std::basic_filebuf<char, std::char_traits<char> >::xsgetn(char*, int) at 
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/esp32-psram/libstdc++-v3/include/fstream:758

0x40119f11: std::basic_streambuf<char, std::char_traits<char> >::sgetn(char*, int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/esp32-psram/libstdc++-v3/include/streambuf:436

0x400f9622: std::istream::read(char*, int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/esp32-psram/libstdc++-v3/include/bits/streambuf_iterator.h:113

0x400d6301: CopyTest() at /mnt/c/Users/Han/Documents/Projects/esp32bug/main/main.cpp:46

0x400d6405: app_main at /mnt/c/Users/Han/Documents/Projects/esp32bug/main/main.cpp:27
0x400d1a72: main_task at /home/han/esp/esp-idf-v3.3-rc/components/esp32/cpu_start.c:529

0x4008bfc9: vPortTaskWrapper at /home/han/esp/esp-idf-v3.3-rc/components/freertos/port.c:435

`

### Steps to repropduce
### Code to reproduce this issue

Gist here: https://gist.github.com/HanVertegaal/cfd356161df072c3d2d5f3b65726989e

[sdkconfig .txt](https://github.com/espressif/esp-idf/files/3573738/sdkconfig.txt)

mahavirj, 2019-09-06:
@HanVertegaal Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/external-ram.html#restrictions. SPIRAM access is not possible when flash cache disabled (due to flash read operation during SPIFFS read), and hence you will have to use internal memory as buffering scheme (i.e. to copy to internal memory first and then accumulate in external SPIRAM).

negativekelvin, 2019-09-06:
@mahavirj true but can vfs handle this better by checking pointer and returning an error?

mahavirj, 2019-09-06:
@negativekelvin Yes, was about to mark for `enhancement` w.r.t documentation or better error handling.

HanVertegaal, 2019-09-06:
@mahavirj Thanks for pointing this out to me. I must say however that even after re-reading that section it is not obvious to me that I cannot read from the flash while accessing the external RAM - as opposed to writing to the flash. A more explicit statement of this restriction would indeed be helpful.

mahavirj, 2019-09-06:
@HanVertegaal Reopening, until we have some explicit code checks and/or documentation improvement.

mahavirj, 2019-09-13:
@HanVertegaal Apologies, seems like `spi_flash_read` can indeed support reading to external SPIRAM buffer (with the help of small buffer from internal memory) but it has issue when read size < 16. Can you please try attached fix (on top of `release/v3.3` branch) and let me know if it helps to fix your issue?
[0001-flash_ops-fix-spi_flash_read-with-source-buffer-not-.zip](https://github.com/espressif/esp-idf/files/3608833/0001-flash_ops-fix-spi_flash_read-with-source-buffer-not-.zip)

HanVertegaal, 2019-09-13:
@mahavirj Yes, this fixes the problem. Thanks!

closed 2019-10-25 by igrr