espressif/esp-idf #4012: BLE stack freeze due to forced 1 second delay in esp_nimble_hci.c

nyronium, 2019-09-04:
## Environment
[ IRRELEVANT]
- Development Kit:      none
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-141-ga7e8d87d3
- Build System:         CMake
- Operating System:     Linux

## Problem Description

When using the NimBLE bluetooth stack, under certain circumstances the whole stack will freeze for 1 second. I have found the issue, which is a call to `vTaskDelay(1000 / portTICK_PERIOD_MS);` inside the `esp_nimble_hci.c` implementation.

[HERE](
https://github.com/espressif/esp-idf/blob/a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c#L119)  is a link to the relevant line in the source code.

The problem is this loop, which apparently waits for some buffer space or tx precondition.

    while (!esp_vhci_host_check_send_available()) {
	vTaskDelay(1000 / portTICK_PERIOD_MS);
    }

In the same file, a similar loop exists without the `vTaskDelay`, making it a busy loop.
I don't know about the instrinsics of `esp_vhci_host_check_send_available()`, but depending on the exact conditions, it might be beneficial to convert this to a busy loop, or to call `taskYIELD();` instead of vTaskDelay`.

### Expected Behavior

Alternative waiting implementation that does not freeze the stack.

### Actual Behavior

1 Second guaranteed freeze.

### Steps to repropduce

Call `ble_gattc_notify();` in quick succession (2-3 times) on a suitable characteristic.
The characteristic must be subscribed by at least one device to force a tx transaction.

Example:
1. create 16 bytes characteristic
2. connect a device which subscribes to it
3. call `ble_gattc_notify()` for the connection and characteristic in quick succession.

prasad-alatkar, 2019-09-05:
Hi @nyronium, I did not observe the '1 second guaranteed BLE stack freeze' after calling `ble_gattc_notify()` consecutively 3 times. The time elapsed after 3 successive `ble_gattc_notify()` was ~16msec for me. I used `examples/bluetooth/nimble/blehr` to reproduce the issue and added 3 successive `ble_gattc_notify` in `blehr_tx_hrate`.  Can you please let me know any additional changes/setting to reproduce the issue ?

prasad-alatkar, 2019-09-27:
Hi @nyronium Do you have any update on this issue ?

nyronium, 2019-09-27:
Sorry for my late answer, unfortunately I haven't had the time to create an isolated test environment for this issue.

I'm not quite sure why you weren't able to reproduce the issue, but probably there is another condition necessary to trigger this. Unfortunately I haven't got the knowledge of your BLE driver internals to tell you what events have to happen for this to occur.

In terms of BLE my application isn't a lot more complex than what I have described:
* Single service (MIDI service), single characteristic with read, write, notify, write_without_response
* One connected peer (Android device to play midi from BLE), the device is therefore subscribed to the characteristic
* The nimble task is created as a pinned task to core 0,
as core 1 is under 100% constant load. Core 0 also runs another task, which has to do frequent work, but it also frequently yields (100 times per second) and definitely does not starve the BLE task. BLE is prioritized.
* The busy task on core 1 will notify the characteristic to send 5 bytes (MIDI) with varying frequency (up to 20 times a second in peak conditions) Especially repeated notifications in quick succession reliably trigger the code path with the delay for me.

The main issue is also undoubtedly that a 1 second delay (or any longer hardcoded delay for that matter) should never exist in any low level driver. This is essentially just creating an opportunity for the driver to freeze itself for the given delay. So this is definitely a design error. The ideal driver only waits for the necessary event which allows it to continue its work.

While it'd be great to reproduce the issue, I think it can be easily seen that there exists a design error that must be fixed. I'm sorry that I could not prepare an example for you.

prasad-alatkar, 2019-09-30:
Hi @nyronium , Yes we don't need `vTaskDelay(1000 / portTICK_PERIOD_MS)` call, in fact we don't need the while loop as well, I have raised an internal MR to fix these unnecessary calls. 

However, I don't think this `vTaskDelay(1000 / portTICK_PERIOD_MS)` is the cause for your `BLE stack freeze` as ideally this **one second delay** line of code is not supposed to be executed anyway, as `esp_vhci_host_check_send_available` API is more of a sanity check (Is configuration correct ? etc.) If I may ask, did you observe any improvement after removal of `vTaskDelay(1000 / portTICK_PERIOD_MS)` in your code ? If you have any other observations regarding the `BLE stack freeze`, please let me know.

prasad-alatkar, 2019-10-24:
Hi @nyronium, the issue is fixed with commit: 19375a67. Can you please confirm ?

nyronium, 2019-11-12:
@prasad-alatkar Thank you. While I can confirm that this particular issue is fixed, it also seems like a new issue has been introduced. When calling notify a lot, now the overflowing internal queue seems to cause a panic (StoreProhibited). I have opened a new issue (#4328) for this problem.

closed 2019-11-12 by nyronium