espressif/esp-idf #4014: Compile error in asio

vortex314, 2019-09-04:
After latest eps-idf upgrade, I'm not abvle to compile anymore existing code.

App "akkaEsp32" version: v0.2-71-g2cdcee9-dirty
CXX build/asio/asio/asio/src/asio.o
In file included from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:22:0,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/signal_blocker.hpp:25,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/impl/select_reactor.ipp:28,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:49,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/Users/lieven/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/csignal:52:11: error: '::sig_atomic_t' has not been declared
   using ::sig_atomic_t;
           ^
/Users/lieven/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/csignal:53:11: error: '::signal' has not been declared
   using ::signal;
           ^
/Users/lieven/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/csignal:54:11: error: '::raise' has not been declared
   using ::raise;
           ^
In file included from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/signal_blocker.hpp:25:0,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/impl/select_reactor.ipp:28,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:49,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:75:3: error: 'sigset_t' does not name a type
   sigset_t old_mask_;
   ^
/Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp: In constructor 'asio::detail::posix_signal_blocker::posix_signal_blocker()':
/Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:40:5: error: 'sigset_t' was not declared in this scope
     sigset_t new_mask;
     ^
/Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:41:17: error: 'new_mask' was not declared in this scope
     sigfillset(&new_mask);
                 ^
/Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:41:25: error: 'sigfillset' was not declared in this scope
     sigfillset(&new_mask);
                         ^
/Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:42:33: error: 'SIG_BLOCK' was not declared in this scope
     blocked_ = (pthread_sigmask(SIG_BLOCK, &new_mask, &old_mask_) == 0);
                                 ^
/Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:42:56: error: 'old_mask_' was not declared in this scope
     blocked_ = (pthread_sigmask(SIG_BLOCK, &new_mask, &old_mask_) == 0);
                                                        ^

vortex314, 2019-09-04:
Maybe related : Lievens-MacBook-Pro:akkaEsp32 lieven$ make
Toolchain path: /Users/lieven/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /Users/lieven/esp/esp-idf/requirements.txt are satisfied.
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.

X-Ryl669, 2019-09-04:
Yes, you need gcc 8.2 now to build master. Run `./install.sh` in esp-idf directory to install the dependency.

vortex314, 2019-09-04:
X-Ryl669 thanks,I tried as you suggested.
However I get an assertion error , I'm still doing something wrong ?
==>     assert tool_version is not None

lieven@pi3:~/esp/esp-idf $ ./install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp32-2019r1-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-armel.tar.gz to /home/lieven/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-armel.tar.gz.tmp
Done
Extracting /home/lieven/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-armel.tar.gz to /home/lieven/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0
Traceback (most recent call last):
  File "/home/lieven/esp/esp-idf/tools/idf_tools.py", line 1250, in <module>
    main(sys.argv[1:])
  File "/home/lieven/esp/esp-idf/tools/idf_tools.py", line 1246, in main
    action_func(args)
  File "/home/lieven/esp/esp-idf/tools/idf_tools.py", line 1038, in action_install
    assert tool_version is not None
AssertionError

X-Ryl669, 2019-09-04:
The error you're getting is likely due to a missing tool*chain* for ARM. I
though you had a mac (here it works with my mac).
Please try on a mac or linux first. There is no reason it'd be loading a
"xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-armel.tar.gz" file on a mac.

If you need to run the tool on ARM, then you'll need to check if such
(compiled) toolchain already exists there and if not, you'll need to build
one (not a simple task... unfortunately).

Alvin1Zhang, 2019-09-05:
Thanks for reporting the issue @vortex314 . Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

vortex314, 2019-09-05:
I'm indeed developing on a Mac, indeed I removed the last comment which was about a Raspberry Pi similar test.
After deploying the latest release of the toolchain with install.sh  , I still got the same issue on the Mac.
Lievens-MacBook-Pro:akkaEsp32 lieven$ make
Toolchain path: /Users/lieven/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /Users/lieven/esp/esp-idf/requirements.txt are satisfied.

/Users/lieven/workspace/akkaEsp32/components/ArduinoJson/component.mk:17: warning: undefined variable `component_compile_rules'
/Users/lieven/workspace/akkaEsp32/components/Common/component.mk:17: warning: undefined variable `component_compile_rules'
/Users/lieven/workspace/akkaEsp32/components/DWM1000/component.mk:18: warning: undefined variable `component_compile_rules'
App "akkaEsp32" version: v0.2-71-g2cdcee9-dirty
CXX build/asio/asio/asio/src/asio.o
In file included from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/posix_signal_blocker.hpp:22,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/signal_blocker.hpp:25,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/detail/impl/select_reactor.ipp:28,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:49,
                 from /Users/lieven/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/Users/lieven/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/csignal:52:11: error: '::sig_atomic_t' has not been declared
   using ::sig_atomic_t;

vortex314, 2019-09-05:
ISSUE TEMPLATE
`Environment
Development Kit: ESP32-DevKitC
Module or chip used: ESP32-WROOM-32
IDF version : v4.1-dev-141-ga7e8d87d3
Build System: Make
Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: macOS
Power Supply: USB
Problem Description
compiling my own code , leads to a compile error in the ESP-IDF framework
Expected Behavior : no compile errors in the ESP-IDF framework
Actual Behavior : compile error
Steps to reproduce:
1/ install esp-idf
2/ install toolchain
3/ clone project : https://github.com/vortex314/akkaEsp32 ( with dependencies )
4/ execute Make

X-Ryl669, 2019-09-05:
It seems you are missing an include here (probably `<signal.h>`) since all the errrors are related to this file. Be aware that ESP does not provide any POSIX signal (you only have wrapper for the functions which `abort()` when called, see `newlib/syscalls.c`). So, if I were you, I would double check the code using this to make sure it's never called, or even better, avoid being used at all.

vortex314, 2019-09-05:
I honestly doubt that it has something to do with my code itself, because it's compiling asio.c at that time , which is part of the ESP-IDF framework.
CXX build/asio/asio/asio/src/asio.o
And the missing type sig_atomic_t ,it complains about, appears in the newlib part of esp-idf. 
> grep -r sig_atomic_t $HOME/esp
./esp-idf/components/newlib/include/signal.h:typedef int	sig_atomic_t;		/* Atomic entity type (ANSI) */
./esp-idf/components/newlib/include/stdint.h:/* This must match sig_atomic_t in <signal.h> (currently int) */
I'll keep looking further.

vortex314, 2019-09-05:
@X-Ryl669 , it seems you were right. I had a rogue 'Signal.h' on my include path. Apparently the #include <signal.h> also picked up this one with a capital S. 
Removing this file solved the issue.
Thanks !

closed 2019-09-05 by vortex314