espressif/esp-idf #4015: i2s_read with external I2S microphone always returns zero

morganrallen, 2019-09-04:
I've spent weeks trying dozens of variation but cannot seem to get i2s_read to return anything but zeros when reading from ICS-43434. My code is very similar to an ESP32 Arduino 1.0.2 example that works, and even if I use `i2s_read_bytes` to match it even more closely I cannot get it to work, which has me wondering if newer versions of the IDF have a bug related to external I2S inputs.
## Environment

- Development Kit:      https://www.tindie.com/products/microwavemont/esp32-simple-alexa/
- Module or chip used:  ESP32-WROVER
- IDF version: v4.0-dev-1443-g39f090a4f
- Build System:         CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
`i2s` initialization goes as expected but when i2s_read, or i2s_read_bytes are called, they always return zero.

### Expected Behavior
Get non-zero response

### Actual Behavior
Zeros

My test case that does not work against the IDF listed above.
https://gitlab.com/morganrallen/i2s-test-case
Although this code when compiled against Arduino ESP32 1.0.2 works as expected.
https://github.com/maspetsberger/esp32-i2s-mems/blob/master/examples/VUMeterDemo/VUMeterDemo.ino

Alvin1Zhang, 2019-09-11:
@morganrallen Thanks for reporting the issue, we will look into it.

amirgon, 2019-09-14:
I see the same problem.

cacaodev, 2019-10-11:
Same here with an ESP32-WROVER on a LOLIN D32 PRO and a INMP441 MEMS mic.

koobest, 2019-10-18:
Hi, @morganrallen 
If the DIN pin is connected to vcc, is the read data still 0?

thanks !!

leleb, 2019-12-07:
Me too having this issue after rebuilding an old working project.
for the moment I solved rolling back to espressif platform version 1.9.0.
In platform.ini set:
  platform = espressif32@1.9.0
using later platforms (1.10.0 or 1.11.0 or latest 1.11.1) makes my I2S microphone not working.

leleb, 2019-12-21:
Solved: using latest platform must swap left and right channel in configuration structure
i2s_config_t i2s_config = {
...
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT, 
...

while in older platform works using I2S_CHANNEL_FMT_ONLY_RIGHT

closed 2020-05-28 by Alvin1Zhang

miketeachman, 2021-02-27:
Not sure why this was closed as it is a bug that was fixed in other versions.

Younes-SadatNejad, 2021-03-08:
Yes I am experiencing the same problem when trying to read I2S MEMS Microphone data,. 
Changing the channel does not solve the problem 
Would you kindly provide an update @Alvin1Zhang

ginkgm, 2021-03-08:
Hi @miketeachman @Younes-SadatNejad ,

Have you tried @leleb 's solution above? And could you please provide the IDF version you are using, and maybe a simple piece of code to reproduce the issue? If you are working on the Arduino, please also provide details for that. It will help us locate the issue more quickly.

miketeachman, 2021-03-08:
Hi @ginkgm,

Changing to I2S_CHANNEL_FMT_ONLY_RIGHT is a workaround to this bug.   I a new issue on the bug.
https://github.com/espressif/esp-idf/issues/6625.   

I discovered this bug while writing an I2S driver for MicroPython.  I was able to add two lines of low-level C code to fix the problem:
https://github.com/miketeachman/micropython/blob/i2s/ports/esp32/machine_i2s.c#L550

 ```
// apply workaround for bug in some ESP-IDF versions that swap
 // the left and right channels
 // https://github.com/espressif/esp-idf/issues/6625
 REG_SET_BIT(I2S_CONF_REG(self->i2s_id), I2S_TX_MSB_RIGHT);
 REG_SET_BIT(I2S_CONF_REG(self->i2s_id), I2S_RX_MSB_RIGHT);
```

Younes-SadatNejad, 2021-03-08:
Hello @ginkgm , 
Thank you so much for getting back to me. 
I am using v4.4-dev-4-g73db14240 for IDF version
Here is my code, I am trying to read the samples of MEMS  microphone. Using LEFT or RIGHT channel does not change the result, I am getting all zero. While a similar code in Arduino with I2S_pip_sample is working  fine. 
Commenting the sample shift results in getting random values that does not change by triggering of microphone ... 

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <math.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/queue.h"
#include "driver/i2s.h"
#include "driver/gpio.h"

#define TAG "I2teSt"
#define I2S_SAMPLE_RATE   (44100)
#define I2S_NUM           (I2S_NUM_0)
#define I2S_READ_LEN      (1 << 7)

static char* i2s_read_buff;

int32_t sample=0;
void i2s_init() {
  esp_err_t err;

  const i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX),
    .sample_rate = I2S_SAMPLE_RATE,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_STAND_I2S,//(i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 8,
    .dma_buf_len = I2S_READ_LEN
  };

  const i2s_pin_config_t pin_config = {
    .bck_io_num = 14,
    .ws_io_num = 15,
    .data_out_num = -1,
    .data_in_num = 32
  };

  err = i2s_driver_install(I2S_NUM, &i2s_config, 0, NULL);
  if (err != ESP_OK) {
    ESP_LOGI(TAG, "Failed installing driver: %d\n", err);
    while (true);
  }
  err = i2s_set_pin(I2S_NUM, &pin_config);
  if (err != ESP_OK) {
    ESP_LOGI(TAG, "Failed setting pin: %d\n", err);
    while (true);
  }
  ESP_LOGI(TAG, "I2S driver installed.");
}

void app_main() {
  i2s_init();

  size_t bytes_read;

  i2s_read_buff = (char*) calloc(I2S_READ_LEN, sizeof(char));

   i2s_start(I2S_NUM);

  while(1) {
     ESP_ERROR_CHECK(i2s_read(I2S_NUM, (void*) i2s_read_buff, I2S_READ_LEN, &bytes_read, portMAX_DELAY));

    for(int i = 0; i < bytes_read; i++) {
      // all 0 output
     	sample = (int32_t)i2s_read_buff[i];
    	sample = ~(sample>>14)+1;
    	printf("%d\n",(int16_t)sample);
    }
  }
};

FedericoBusero, 2023-05-21:
Is this issue fixed? I still experience this in Arduino esp32 board version 2.09 (based on IDF 4.4.4)