espressif/esp-idf #4016: FLAG_ATTR is breaking C++ compilation

X-Ryl669, 2019-09-04:
Using master (a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea), there is an issue with the FLAG_ATTR macro used for enum for flags in C++ builds.

It expands to:
```
#define FLAG_ATTR_IMPL(TYPE, INT_TYPE) \
constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
constexpr TYPE operator| (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a | (INT_TYPE)b); } \
constexpr TYPE operator& (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a & (INT_TYPE)b); } \
constexpr TYPE operator^ (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a ^ (INT_TYPE)b); } \
constexpr TYPE operator>> (TYPE a, int b) { return (TYPE)((INT_TYPE)a >> b); } \
constexpr TYPE operator<< (TYPE a, int b) { return (TYPE)((INT_TYPE)a << b); } \
TYPE& operator|=(TYPE& a, TYPE b) { a = a | b; return a; } \
TYPE& operator&=(TYPE& a, TYPE b) { a = a & b; return a; } \
TYPE& operator^=(TYPE& a, TYPE b) { a = a ^ b; return a; } \
TYPE& operator>>=(TYPE& a, int b) { a >>= b; return a; } \
TYPE& operator<<=(TYPE& a, int b) { a <<= b; return a; }
```

If the included file is used in multiple `.cpp` unit, you'll get multiple definition of these functions and it breaks linking.

Thus, the code should read:
```
#define FLAG_ATTR_IMPL(TYPE, INT_TYPE) \
inline constexpr TYPE operator~ (TYPE a) { return (TYPE)~(INT_TYPE)a; } \
inline constexpr TYPE operator| (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a | (INT_TYPE)b); } \
inline constexpr TYPE operator& (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a & (INT_TYPE)b); } \
inline constexpr TYPE operator^ (TYPE a, TYPE b) { return (TYPE)((INT_TYPE)a ^ (INT_TYPE)b); } \
inline constexpr TYPE operator>> (TYPE a, int b) { return (TYPE)((INT_TYPE)a >> b); } \
inline constexpr TYPE operator<< (TYPE a, int b) { return (TYPE)((INT_TYPE)a << b); } \
inline TYPE& operator|=(TYPE& a, TYPE b) { a = a | b; return a; } \
inline TYPE& operator&=(TYPE& a, TYPE b) { a = a & b; return a; } \
inline TYPE& operator^=(TYPE& a, TYPE b) { a = a ^ b; return a; } \
inline TYPE& operator>>=(TYPE& a, int b) { a >>= b; return a; } \
inline TYPE& operator<<=(TYPE& a, int b) { a <<= b; return a; }
```
Notice that `inline` here is better than `static` since the former does what you'd expect, that is, it's replacing the function with the code (and since the used type is an enum, it's a no-op), while the latter create actual functions that might be called and not inlined (and thus, use binary space). `inline` prevents multiple definition, `static` does not (it just prevent colliding definitions).

Alvin1Zhang, 2019-09-05:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks. @X-Ryl669

X-Ryl669, 2019-09-05:
Commit id is in the initial post. The error is generic and not linked with any version, you can't define a function multiple time in a header without it being inline or static.

closed 2019-11-06 by Alvin1Zhang