espressif/esp-idf #4017: Linking error with conflicting _raise_r definitions

X-Ryl669, 2019-09-04:
Using latest master (a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea), and running updated toolchain (that was installed via `./install.sh`), I now got this linking error:
```
/Users/cyril/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/cyril/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc_nano.a(lib_a-raise.o): in function `_raise_r':
/builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/signal/raise.c:65: multiple definition of `_raise_r'; /Users/cyril/esp/project/build/newlib/libnewlib.a(syscalls.o):/Users/cyril/esp/esp-idf/components/newlib/syscalls.c:31: first defined here
```

X-Ryl669, 2019-09-04:
This is fixed for me by:
```
diff --git a/components/newlib/syscalls.c b/components/newlib/syscalls.c
index c3da628..3a04b84 100644
--- a/components/newlib/syscalls.c
+++ b/components/newlib/syscalls.c
@@ -27,6 +27,7 @@ int _system_r(struct _reent *r, const char *str)
     return -1;
 }

+int _raise_r(struct _reent *r, int sig) __attribute__((weak));
 int _raise_r(struct _reent *r, int sig)
 {
     abort();
```
Typically, since I can't change the implementation in the cross compiler, I'm saying that the second one in `newlib` is a weak symbol, and this will be replaced by the one in the cross compiler.

Since you've passed your build tests on your side (I guess), this means that you don't have such symbol in your cross compiler toolchain. If I just remove it, it would break on your side (in that case, the symbol will be missing for you).

Ideally, the cross compiler toolchain should not contain such symbol, but that's not something we can fix in this repository.

Alvin1Zhang, 2019-09-05:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks. @X-Ryl669

X-Ryl669, 2019-09-05:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): see commit id in first post
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
   
- Operating System:     [macOS]
- Power Supply:         [USB|external 5V]

igrr, 2020-09-09:
Fixed in https://github.com/espressif/esp-idf/commit/08286ac20f28a58fffe7acacc34c3566dededdef.

closed 2020-09-09 by igrr