espressif/esp-idf #4018: Fail to install on mac OS

X-Ryl669, 2019-09-04:
Using latest master(a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea), `install.sh` fails with:
```
$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Installing Python environment and packages
fatal: The empty string is not a valid path
Traceback (most recent call last):
  File "/Users/cyril/esp/esp-idf/tools/idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "/Users/cyril/esp/esp-idf/tools/idf_tools.py", line 1313, in main
    action_func(args)
  File "/Users/cyril/esp/esp-idf/tools/idf_tools.py", line 1112, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "/Users/cyril/esp/esp-idf/tools/idf_tools.py", line 813, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['git', '--work-tree=', '/Users/cyril/esp/esp-idf', 'describe', '--tags']' returned non-zero exit status 128
```
 Please notice that `-C` is not a valid command for git (you need the generic `--work-tree=`). I've replaced `${IDF_PATH}/tools/idf_tools.py` with this:
```
diff --git a/tools/idf_tools.py b/tools/idf_tools.py
index 3d23412..ef292cf 100755
--- a/tools/idf_tools.py
+++ b/tools/idf_tools.py
@@ -810,7 +810,7 @@ def get_python_env_path():
         with open(version_file_path, "r") as version_file:
             idf_version_str = version_file.read()
     else:
-        idf_version_str = subprocess.check_output(['git', '-C', global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
+        idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
     match = re.match(r'^v([0-9]+\.[0-9]+).*', idf_version_str)
     idf_version = match.group(1)
```
And it passed.

Alvin1Zhang, 2019-09-05:
@X-Ryl669 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

X-Ryl669, 2019-09-05:
Se #4017

makermuyi, 2019-09-08:
I'm sorry. According to the method you provided, I did not get the correct result and running "./install.sh" still has an error.
`Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Installing Python environment and packages
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "/home/macro/esp/esp-idf-master/tools/idf_tools.py", line 1320, in <module>
    main(sys.argv[1:])
  File "/home/macro/esp/esp-idf-master/tools/idf_tools.py", line 1316, in main
    action_func(args)
  File "/home/macro/esp/esp-idf-master/tools/idf_tools.py", line 1115, in action_install_python_env
    idf_python_env_path, _, virtualenv_python = get_python_env_path()
  File "/home/macro/esp/esp-idf-master/tools/idf_tools.py", line 815, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['git', '--work-tree=/home/macro/esp/esp-idf-master', 'describe', '--tags']' returned non-zero exit status 128‘
I would be grateful if you could provide a more detailed solution to this problem.

igrr, 2019-09-08:
Have you downloaded ESP-IDF as a .zip archive from GitHub and extracted it? This will not work, you need to use Git to clone the repository. Please refer to the steps in the Getting Started guide: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#installation-step-by-step

gabrielantao, 2019-09-08:
I followed exactly all the steps described there, but I got this same problem.

X-Ryl669, 2019-09-08:
You must clone the repository (via git clone). If you do so, you'll not get this error: 
```
fatal: not a git repository (or any of the parent directories): .git
```
@gabrielantao : Please report the error you get exactly.

gabrielantao, 2019-09-09:
@X-Ryl669 I tested those steps here on Ubuntu 18.04 and it worked fine. I got this when I try step 3 when I try on Mint.

gabrielantao@gabriel-antao ~/esp/esp-idf $ ./install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Installing Python environment and packages
Installing Python packages from /home/gabrielantao/esp/esp-idf/requirements.txt
/home/gabrielantao/.espressif/python_env/idf4.1_py2.7_env/bin/python: No module named pip
Traceback (most recent call last):
  File "/home/gabrielantao/esp/esp-idf/tools/idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "/home/gabrielantao/esp/esp-idf/tools/idf_tools.py", line 1313, in main
    action_func(args)
  File "/home/gabrielantao/esp/esp-idf/tools/idf_tools.py", line 1136, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "/home/gabrielantao/anaconda2/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/gabrielantao/.espressif/python_env/idf4.1_py2.7_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/gabrielantao/esp/esp-idf/requirements.txt']' returned non-zero exit status 1

X-Ryl669, 2019-09-09:
Python error trace are a real pain to decipher. But here, the line that's important is:
```
/home/gabrielantao/.espressif/python_env/idf4.1_py2.7_env/bin/python: No module named pip
```
This is not related to this bug. You'd install pip `apt-get install pip` or `port install pip` if you are on a Mac

makermuyi, 2019-09-16:
This problem is caused by the use of macOS.

X-Ryl669, 2019-09-16:
As the title said?

igrr, 2019-09-16:
@X-Ryl669 what is the version of git when this error happened?

X-Ryl669, 2019-09-16:
I'll tell you later today. In all cases, `--work-tree` is working on all versions (it is what you intended, that is, ask git to extract the description of a given work tree) while `-C` is *relatively* new and not your intention (it's a base directory that's prependded to all other directory-based optionn like `work-tree` or `git-dir`). I'd say that the fact it is working initially is pure luck that git use that path for looking its `.git` subdirectory instead of `$PWD`.

X-Ryl669, 2019-09-17:
git version 1.8.3.4

closed 2019-09-30 by igrr

Quentinb, 2019-12-13:
I've followed this step-by-step, more than once and get the same error:

`quentinb@Quentins-MacBook-Pro-2 esp-idf % . ./export.sh                                             
Adding ESP-IDF tools to PATH...
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "/Users/quentinb/.espressif/tools/idf_tools.py", line 1349, in <module>
    main(sys.argv[1:])
  File "/Users/quentinb/.espressif/tools/idf_tools.py", line 1345, in main
    action_func(args)
  File "/Users/quentinb/.espressif/tools/idf_tools.py", line 980, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "/Users/quentinb/.espressif/tools/idf_tools.py", line 840, in get_python_env_path
    idf_version_str = subprocess.check_output(['git', '--work-tree=' + global_idf_path, 'describe', '--tags'], cwd=global_idf_path, env=os.environ).decode()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['git', '--work-tree=/Users/quentinb/.espressif', 'describe', '--tags']' returned non-zero exit status 128
quentinb@Quentins-MacBook-Pro-2 esp-idf % 
quentinb@Quentins-MacBook-Pro-2 esp-idf % 
quentinb@Quentins-MacBook-Pro-2 esp-idf % git --work-tree=/Users/quentinb/.espressif describe --tags
v4.1-dev-1086-g93a8603c5
quentinb@Quentins-MacBook-Pro-2 esp-idf % `

igrr, 2019-12-13:
@Quentinb please check that your ESP-IDF directory is a valid Git repository. Does running `git describe --tags --always` in the ESP-IDF repository show an error? If it does, please make sure that you have downloaded ESP-IDF as explained in the Getting Started guide (`git clone --recursive https://github.com/espressif/esp-idf.git`).

Quentinb, 2019-12-15:
I've managed to get it to work by adding a IDF_PATH=~/esp/esp-idf into my user profile.

trentmillar, 2020-02-18:
> Python error trace are a real pain to decipher. But here, the line that's important is:
>
> ```
> /home/gabrielantao/.espressif/python_env/idf4.1_py2.7_env/bin/python: No module named pip
> ```
>
> This is [...]

I had this error even though pip was installed. This resolved the issue for me,

`curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py`
`python get-pip.py`

0mars, 2020-06-28:
this worked for me
```
curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py
/Users/xxxxx/.espressif/python_env/idf3.3_py2.7_env/bin/python get-pip.py
```