espressif/esp-idf #4022: [BLE Mesh] Crash in some cases after the "BT_MESH_SETTINGS" option is enabled

zhanlej, 2019-09-05:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): [v4.0-dev-301-gc7c419da4(ble_mesh_v0.6.1)]
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): [1.22.0-80-g6c4433a]
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

由于项目需要，我需要实现ble mesh的配置信息可持久化存储功能，即设备重启后依旧能自动加入mesh网络。因此我在menuconfig中配置打开了：``Blutooth Mesh support --> Store Bluetooth Mesh key and configuration persistenl``，在测试中我发现有2个情况会导致设备崩溃重启：

1. 绑定第二个app_key的时候，设备崩溃重启。

我在``sdkconfig.defaults``中修改了``CONFIG_BLE_MESH_APP_KEY_COUNT=3``让设备最多可以持有3个app_key，然后在对model绑定第2个app_key的时候设备就会发生崩溃重启并打印堆栈信息

```
/examples_c7c419d/bluetooth/ble_mesh/ble_mesh_node_store$ make -j4 flash monitor
Toolchain path: /home/zhanlej/WorkSpace/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "ble_mesh_node" version: 1
Python requirements from /home/zhanlej/WorkSpace/esp/git_test/ble_mesh/esp-idf_v0.6.1/requirements.txt are satisfied.

Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.7-dev
Serial port /dev/ttyUSB0
Connecting.......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:0a:c4:c1:06:2c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 24576 bytes to 14655...
Wrote 24576 bytes (14655 compressed) at 0x00001000 in 1.3 seconds (effective 151.8 kbit/s)...
Hash of data verified.
Compressed 846432 bytes to 471236...
Wrote 846432 bytes (471236 compressed) at 0x00010000 in 41.7 seconds (effective 162.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 961.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6472
load:0x40078000,len:11308
load:0x40080400,len:6700
entry 0x40080764
I (28) boot: ESP-IDF v4.0-dev-301-gc7c419da4 2nd stage bootloader
I (29) boot: compile time 11:03:10
I (29) boot: Enabling RNG early entropy source...
[... 431 lines cut ...]
E (10211) task_wdt:  - IDLE0 (CPU 0)
E (10211) task_wdt: Tasks currently running:
E (10211) task_wdt: CPU 0: Btc_task
E (10211) task_wdt: CPU 1: IDLE1
E (15211) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (15211) task_wdt:  - IDLE0 (CPU 0)
E (15211) task_wdt: Tasks currently running:
E (15211) task_wdt: CPU 0: Btc_task
E (15211) task_wdt: CPU 1: IDLE1
E (20211) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20211) task_wdt:  - IDLE0 (CPU 0)
E (20211) task_wdt: Tasks currently running:
E (20211) task_wdt: CPU 0: Btc_task
E (20211) task_wdt: CPU 1: IDLE1
E (25211) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (25211) task_wdt:  - IDLE0 (CPU 0)
E (25211) task_wdt: Tasks currently running:
E (25211) task_wdt: CPU 0: Btc_task
E (25211) task_wdt: CPU 1: IDLE1
```

## Other items if possible

- [ble_mesh_node_store_test.zip](https://github.com/espressif/esp-idf/files/3577388/ble_mesh_node_store_test.zip)
- [sdkconfig.zip](https://github.com/espressif/esp-idf/files/3577390/sdkconfig.zip)

WCCWCC, 2019-09-06:
Hi zhanlej ，
     我按照你的步骤进行操作，均是正常的。 
            或许你应该执行一下  make erase_flash。

zhanlej, 2019-09-06:
Hi WCCWCC，

- [ble_mesh_store_log.txt](https://github.com/espressif/esp-idf/files/3582049/ble_mesh_store_log.txt)

上面完整的log信息，在log中能确定我在每次重新烧写之前都有执行 make erase_flash。

由于之前我的操作流程没有描述清楚这里我再补充说明一下。

- 首先我使用的APP是安卓版本的 nRF Mesh V2.0.2（不清楚是否和APP版本有关）
- 然后我使用APP扫描到设备进行provisioner配置
- 配置成功后在设备列表中点击进入该设备，点击进入“APP Keys”菜单，添加"Application Key 2"
- 退出找到“Elements”菜单，点击“Element: 0x0002”，点击“Generic On Off Server”
- 点击“BIND KEY”先绑定“Application Key 1”，然后左滑删除“Application Key 1”
- 再点击“BIND KEY”绑定“Application Key 2”，此时设备崩溃重启！

然后我测试了 provision 后直接绑定“Application Key 2”确实也没问题，但是删除“Application Key 2”后再绑定“Application Key 1”同样也会崩溃，因此我认为问题出现在删除原来的 APP KEY 后添加新的APP KEY 时。

```
另外我不知道你有没有测试使用我提供的main,c然后在APP上操作“Reset Node”操作，然后再进行 provisioner 操作，这个操作我每次都会崩溃并且导致设备不能正常工作
```

WCCWCC, 2019-09-06:
Hi zhanlej,

    问题复现了，我会尽快答复你。

zhanlej, 2019-09-06:
Hi WCCWCC,

谢谢，另外我认为我的第二种崩溃情况可以参考#3408，我认为我们是同一个问题。

Campou, 2019-09-09:
@zhanlej 
* Thanks for the feedback, we will provide the bugfix **in this week**. Since the current FCB mode of flash used in ESP BLE Mesh is not appropriate enough, we need to refactor the whole NVS storage implementation.

Best Regards

enricop, 2019-09-09:
Hi @Campou ,
we are using BLE mesh in our project too, this fix would be usefull also for us. Thanks for your support.

zhanlej, 2019-09-16:
@Campou @WCCWCC 

Thanks, I tested the problem mentioned earlier and it solved it, but now there is a new problem:

> The APP can no longer reconnect to the device when the device is restarted because it is not in the scan list.

Alvin1Zhang, 2019-09-16:
Hi. Thanks for the updates, if this issue has been resolved, would you please help close this one? For the new problem, please help open a new issue. Thanks.

zhanlej, 2019-09-16:
@Alvin1Zhang 

However, the new problem is normal before the code is modified, but it is abnormal after the modification. I think it is a bug that they generated during the process of modifying the code, so I think it should be placed in the same problem.

Campou, 2019-09-16:
Hi @zhanlej,

You mean after the device been restarted, it can not be reached in the nRF Mesh APP's **Scanner** (i.e. scan list)?

After the device is provisioned successfully and restarted, the APP will not show it in the **Scanner**, instead we need to click the **CONNECT** in the top right and the APP will start to scan the proxy advertising of the device.

I have tested using the branch **ble_mesh_release/esp-ble-mesh-v0.6.1** with NVS storage on, each time after the device is restarted, the APP can connect with it successfully and control the RGB light.

The attachment is the test log and the APP screenshot.

[ble_mesh_node_nvs_test.log](https://github.com/espressif/esp-idf/files/3615989/ble_mesh_node_nvs_test.log)

[image]

zhanlej, 2019-09-18:
Hi @Campou ,

I don't mean to click ``scanner`` but to click ``CONNECT``. The app I used was Android's ``nRF Mesh v2.0.2``, and I retested it many times and found the same results as I did before.

Then I used ``iphone 6s`` to test the ``nRF Mesh v1.0.4`` from the IOS downloaded from ``APP STORE``. I found that although the device can be reconnected after rebooting, when I delete APP_KEY_1 and add APP_KEY_2, the device has an exception.

Finally, I found that your APP screenshot is different from mine, so I downloaded the v1.2.1 version of the app from ``https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/releases`` to test it. I found that the test was successful as the test you described.

[image]

[image]

So now I find this issue related to the version of the app?

The Android APP apk is as fllows:

- [nRF.Mesh.-.V.1.2.1.zip](https://github.com/espressif/esp-idf/files/3626039/nRF.Mesh.-.V.1.2.1.zip)
- [nRF.Mesh.-.V.-.2.0.2.zip](https://github.com/espressif/esp-idf/files/3626045/nRF.Mesh.-.V.-.2.0.2.zip)

enricop, 2019-09-18:
> 
> 
> Hi @zhanlej,
> 
> You mean after the device been restarted, it can not be reached in the nRF Mesh APP's **Scanner** (i.e. scan list)?
> 
> After the device is provisioned successfully and restarted, the APP will not show it in the **Scanner**, instead we need to click the **CONNECT** on the top right and the APP will start to scan the proxy advertising of the device.
> 
> I have tested using the branch **ble_mesh_release/esp-ble-mesh-v0.6.1** with NVS storage on, each time after the device is restarted, the APP can connect with it successfully and control the RGB light.
> 
> The attachment is the test log and the APP screenshot.
> 
> [ble_mesh_node_nvs_test.log](https://github.com/espressif/esp-idf/files/3615989/ble_mesh_node_nvs_test.log)
> 
> [image]

We are using the new code in release/v4.0 branch without problems (for MESH node provisioning)

zhanlej, 2019-10-22:
@Campou @WCCWCC 

Is there any new progress?

Campou, 2019-12-02:
Hi @zhanlej 

Sorry for the late reply.

Please try with the latest code in the branch `release/v4.0`.

Thanks & Regards

Alvin1Zhang, 2020-07-27:
@zhanlej Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-07-27 by Alvin1Zhang