espressif/esp-idf #4023: RMT error, status 0x04000000

vgonet, 2019-09-05:
Hello,

I've implemented a code which is able to decode RC5 protocole. It's working but sometimes, I've the following error.

E (468926) rmt: RMT[0] ERR
E (468926) rmt: status: 0x04000000

Can you please help me to know what it means ?

Thank you and best regards.
Vincent

Alvin1Zhang, 2019-09-05:
@vgonet Thanks for reporting, will close this due to the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md) has been ignored. We suggest you could post your questions on our forum at https://esp32.com. Thanks.

Alvin1Zhang, 2019-11-04:
@vgonet Will close this ticket due to short of feedback, feel free to reopen if you could help provide more details. Thanks.

closed 2019-11-04 by Alvin1Zhang