espressif/esp-idf #4024: tinfl_decompress_mem_to_callback is missing in actual rom

123swk123, 2019-09-05:
## Environment
- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1443-g39f090a4f
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
as per the esp32_rom.ld tinfl_decompress_mem_to_callback is @ 0x40060090 but as per the asm dump at that location it simply returns -1, so i think this function is missing in the ROM or may be its miss placed.

//Detailed problem description goes here.

### Steps to repropduce
1. run the below code with any sdkconfig

### Code to reproduce this issue

```cpp
#include "esp_log.h"
#include "esp32/rom/miniz.h"

const uint8_t COMP_SAMP_DATA[] = {
0x78,0x9c,0xcb,0xc8,0xcc,0x04,0x00,0x02,0x76,0x01,0x3b
};

uint32_t gu32CountOfCbk = 0;
int cbkInflator(const void *pBuf, int len, void *pUser)
{
  ESP_LOGI(TAG, "%u th instance data size: %d", gu32CountOfCbk++, len);
  ESP_LOGI(TAG, "%s", pBuff); //should print "hii"
  return 1; //return success
}

void app_main()
{
   //esp rom function - does not work
  size_t szCompLen = sizeof(COMP_SAMP_DATA);
   int sts = tinfl_decompress_mem_to_callback(COMP_SAMP_DATA, &szCompLen, cbkInflator, NULL, TINFL_FLAG_PARSE_ZLIB_HEADER);
  ESP_LOGI(TAG, "status of inflation: %d, %u", sts, szCompLen);
}

```

123swk123, 2019-09-05:
[rom hex dump assembly file](https://github.com/espressif/esp-idf/files/3578012/rom_dump.hex.asm.txt) from address 0x3ff96e6c to 0x40061fff

X-Ryl669, 2019-09-05:
For what it's worth, here the source code of the function:
```
int tinfl_decompress_mem_to_callback(const void *pIn_buf, size_t *pIn_buf_size, tinfl_put_buf_func_ptr pPut_buf_func, void *pPut_buf_user, int flags)
{
  int result = 0;
  tinfl_decompressor decomp;
  mz_uint8 *pDict = (mz_uint8*)MZ_MALLOC(TINFL_LZ_DICT_SIZE); size_t in_buf_ofs = 0, dict_ofs = 0;
  if (!pDict)
    return TINFL_STATUS_FAILED;
  tinfl_init(&decomp);
  for ( ; ; )
  {
    size_t in_buf_size = *pIn_buf_size - in_buf_ofs, dst_buf_size = TINFL_LZ_DICT_SIZE - dict_ofs;
    tinfl_status status = tinfl_decompress(&decomp, (const mz_uint8*)pIn_buf + in_buf_ofs, &in_buf_size, pDict, pDict + dict_ofs, &dst_buf_size,
      (flags & ~(TINFL_FLAG_HAS_MORE_INPUT | TINFL_FLAG_USING_NON_WRAPPING_OUTPUT_BUF)));
    in_buf_ofs += in_buf_size;
    if ((dst_buf_size) && (!(*pPut_buf_func)(pDict + dict_ofs, (int)dst_buf_size, pPut_buf_user)))
      break;
    if (status != TINFL_STATUS_HAS_MORE_OUTPUT)
    {
      result = (status == TINFL_STATUS_DONE);
      break;
    }
    dict_ofs = (dict_ofs + dst_buf_size) & (TINFL_LZ_DICT_SIZE - 1);
  }
  MZ_FREE(pDict);
  *pIn_buf_size = in_buf_ofs;
  return result;
}
```
In your dump, there is no function prolog at the given address. Are you sure about your dump ?
If the function does not crash and return -1, it's failing at the MZ_MALLOC stage, since `TINFL_STATUS_FAILED` == -1.

X-Ryl669, 2019-09-05:
Please notice that `TINFL_LZ_DICT_SIZE` is 32768, so it's a large memory buffer to allocate.
Can you try to allocate such a buffer and assert it's working (instead of calling the function) ?

projectgus, 2019-09-06:
Hi @123swk123 ,

Thanks for the clear issue report.

@X-Ryl669 is correct, miniz was built with `MINIZ_NO_MALLOC` flag:
https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp32/rom/miniz.h#L29

As a result, the compiler optimizes `tinfl_decompress_mem_to_callback` down to a function that returns -1. Here's the disassembly dump:

```
40060090 <tinfl_decompress_mem_to_callback>:
40060090:       004136          entry   a1, 32
40060093:       f27c            movi.n  a2, -1
40060095:       f01d            retw.n
```

Will add some comments & warnings in the header for the miniz functions which can't be used.

As @X-Ryl669 has said, `tinfl_decompress_mem_to_mem` with a pre-allocated buffer should work.

123swk123, 2019-09-09:
Hi all,
I get it since the rom was image was created with MINIZ_NO_MALLOC setting and hence compiler has optimized the code by just returning -1. 
so as suggested it would be helpful if the documentation reflects that all miniz functions with malloc are not supported.

by the way I am actually using the modified source as mentioned by @X-Ryl669

123swk123, 2019-09-09:
> For what it's worth, here the source code of the function:
> 
> ```
> int tinfl_decompress_mem_to_callback(const void *pIn_buf, size_t *pIn_buf_size, tinfl_put_buf_func_ptr pPut_buf_func, void *pPut_buf_user, int flags)
> {
>   int result = 0;
>   tinfl_decompressor decomp;
>   mz_uint8 *pDict = (mz_uint8*)MZ_MALLOC(TINFL_LZ_DICT_SIZE); size_t in_buf_ofs = 0, dict_ofs = 0;
>   if (!pDict)
>     return TINFL_STATUS_FAILED;
>   tinfl_init(&decomp);
>   for ( ; ; )
>   {
>     size_t in_buf_size = *pIn_buf_size - in_buf_ofs, dst_buf_size = TINFL_LZ_DICT_SIZE - dict_ofs;
>     tinfl_status status = tinfl_decompress(&decomp, (const mz_uint8*)pIn_buf + in_buf_ofs, &in_buf_size, pDict, pDict + dict_ofs, &dst_buf_size,
>       (flags & ~(TINFL_FLAG_HAS_MORE_INPUT | TINFL_FLAG_USING_NON_WRAPPING_OUTPUT_BUF)));
>     in_buf_ofs += in_buf_size;
>     if ((dst_buf_size) && (!(*pPut_buf_func)(pDict + dict_ofs, (int)dst_buf_size, pPut_buf_user)))
>       break;
>     if (status != TINFL_STATUS_HAS_MORE_OUTPUT)
>     {
>       result = (status == TINFL_STATUS_DONE);
>       break;
>     }
>     dict_ofs = (dict_ofs + dst_buf_size) & (TINFL_LZ_DICT_SIZE - 1);
>   }
>   MZ_FREE(pDict);
>   *pIn_buf_size = in_buf_ofs;
>   return result;
> }
> ```
> 
> In your dump, there is no function prolog at the given address. Are you sure about your dump ?
> If the function does not crash and return -1, it's likely your rom dump is wrong.
> It's failing at the MZ_MALLOC stage, since `TINFL_STATUS_FAILED` is -1.

yes i too notice that function prolog 'entry xxx' was missing in asm, I didn't have much time to investigate on it. its probably because of memory alignment issue with the hex dump and hence objdump tool could not disassemble it correctly