espressif/esp-idf #4026: Custom Bootloader in Projektfolder is no longer considered (V3.3, V3.2 ... )

ESP32DE, 2019-09-05:
hi guys

[ESP32.COM Post](https://esp32.com/viewtopic.php?f=2&t=12133)

[Problem Describe](https://esp32.com/viewtopic.php?f=2&t=12133#p48837)

I use GNU Make, tested this with the last Release V3.3, also V3.2 also V3.2 beta
i tested this in windows also in Ubuntu(Linux)

Custom Bootloader component is no longer considered

Steps for test:
feel free to check it out, use example the hello_world

Steps:

a) copy the hello world example to your test folder
b) create in the hello world the folder "components"
c) copy from orgin used ESP-IDF ( example V3.3 ) the components bootloader and bootloader_support to your project components
d) edit the bootloader_init.c file in (yourproject)/components/bootloader_support/src/bootloader_init.c
e) jump to static esp_err_t bootloader_main(void)
f) edit the func and add this 6 lines code between the lines 

```
static esp_err_t bootloader_main() {
    bootloader_common_vddsdio_configure();
    /* Read and keep flash ID, for further use. */
    g_rom_flashchip.device_id = bootloader_read_flash_id();
/*1*/	// break me down
/*2*/	rtc_wdt_disable();
/*3*/    REG_SET_FIELD(TIMG_WDTWPROTECT_REG(0), TIMG_WDT_WKEY,  TIMG_WDT_WKEY_VALUE);
/*4*/    REG_CLR_BIT( TIMG_WDTCONFIG0_REG(0), TIMG_WDT_FLASHBOOT_MOD_EN );
/*5*/	ets_printf("here we are in custom project\n");
/*6*/	while(1);
    esp_image_header_t fhdr;
    if (bootloader_flash_read(ESP_BOOTLOADER_OFFSET, &fhdr, sizeof(esp_image_header_t), true) != ESP_OK) {

```

g) save it,
h) check and controll with 

`make list-components`

$ make list-components
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
*******************************************************************************
COMPONENT_DIRS (components searched for here)
/home/sysop/esp/V3.3/hello_world/components
/h/Releases/esp-idf-v3.3/components
/home/sysop/esp/V3.3/hello_world/main
*******************************************************************************
TEST_COMPONENTS (list of test component names)

*******************************************************************************
TEST_EXCLUDE_COMPONENTS (list of test excluded names)

*******************************************************************************
COMPONENT_PATHS (paths to all components):
/h/Releases/esp-idf-v3.3/components/app_trace
/h/Releases/esp-idf-v3.3/components/app_update
/h/Releases/esp-idf-v3.3/components/asio
/h/Releases/esp-idf-v3.3/components/aws_iot
**/home/sysop/esp/V3.3/hello_world/components/bootloader
/home/sysop/esp/V3.3/hello_world/components/bootloader_support**
/h/Releases/esp-idf-v3.3/components/bt
/h/Releases/esp-idf-v3.3/components/coap
/h/Releases/esp-idf-v3.3/components/console
/h/Releases/esp-idf-v3.3/components/cxx
/h/Releases/esp-idf-v3.3/components/driver
/h/Releases/esp-idf-v3.3/components/efuse
/h/Releases/esp-idf-v3.3/components/esp-tls
/h/Releases/esp-idf-v3.3/components/esp32
/h/Releases/esp-idf-v3.3/components/esp_adc_cal
/h/Releases/esp-idf-v3.3/components/esp_event
/h/Releases/esp-idf-v3.3/components/esp_http_client
/h/Releases/esp-idf-v3.3/components/esp_http_server
/h/Releases/esp-idf-v3.3/components/esp_https_ota
/h/Releases/esp-idf-v3.3/components/esp_https_server
/h/Releases/esp-idf-v3.3/components/esp_ringbuf
/h/Releases/esp-idf-v3.3/components/espcoredump
/h/Releases/esp-idf-v3.3/components/esptool_py
/h/Releases/esp-idf-v3.3/components/ethernet
/h/Releases/esp-idf-v3.3/components/expat
/h/Releases/esp-idf-v3.3/components/fatfs
/h/Releases/esp-idf-v3.3/components/freemodbus
/h/Releases/esp-idf-v3.3/components/freertos
/h/Releases/esp-idf-v3.3/components/heap
/h/Releases/esp-idf-v3.3/components/idf_test
/h/Releases/esp-idf-v3.3/components/jsmn
/h/Releases/esp-idf-v3.3/components/json
/h/Releases/esp-idf-v3.3/components/libsodium
/h/Releases/esp-idf-v3.3/components/log
/h/Releases/esp-idf-v3.3/components/lwip
/home/sysop/esp/V3.3/hello_world/main
/h/Releases/esp-idf-v3.3/components/mbedtls
/h/Releases/esp-idf-v3.3/components/mdns
/h/Releases/esp-idf-v3.3/components/micro-ecc
/h/Releases/esp-idf-v3.3/components/mqtt
/h/Releases/esp-idf-v3.3/components/newlib
/h/Releases/esp-idf-v3.3/components/nghttp
/h/Releases/esp-idf-v3.3/components/nvs_flash
/h/Releases/esp-idf-v3.3/components/openssl
/h/Releases/esp-idf-v3.3/components/partition_table
/h/Releases/esp-idf-v3.3/components/protobuf-c
/h/Releases/esp-idf-v3.3/components/protocomm
/h/Releases/esp-idf-v3.3/components/pthread
/h/Releases/esp-idf-v3.3/components/sdmmc
/h/Releases/esp-idf-v3.3/components/smartconfig_ack
/h/Releases/esp-idf-v3.3/components/soc
/h/Releases/esp-idf-v3.3/components/spi_flash
/h/Releases/esp-idf-v3.3/components/spiffs
/h/Releases/esp-idf-v3.3/components/tcp_transport
/h/Releases/esp-idf-v3.3/components/tcpip_adapter
/h/Releases/esp-idf-v3.3/components/ulp
/h/Releases/esp-idf-v3.3/components/unity
/h/Releases/esp-idf-v3.3/components/vfs
/h/Releases/esp-idf-v3.3/components/wear_levelling
/h/Releases/esp-idf-v3.3/components/wifi_provisioning
/h/Releases/esp-idf-v3.3/components/wpa_supplicant
/h/Releases/esp-idf-v3.3/components/xtensa-debug-module

that your **custom bootloader component** is used for the compile

now compile and flash your project
and`? is the project component used or the esp-idf path component?

here in my example and tests 
the custom bootloader is no longer considered.

windows 7 , 10
ubunto 16, 18, 

ESP-IDF
tested things in 
V3.2
V3.3

can bee that the problem is in earlier Versions too.
not tested just in time.

thank you
best wishes
rudi ;-)

attached is a earlier custom test from me, where the summary problem is showed.
i tested, which bootloader is used, if i use a custom bootloader in the custom project.
custom bootloader does not override original, ( used GNU Make ) 

[image]

projectgus, 2019-09-06:
Hi @ESP32DE,

Thanks for the detailed report.

I'm able to reproduce this, specifically that `bootloader_support` component can't be overridden in the project when using GNU Make. The source files from IDF_PATH are always used.

I am able to override the `bootloader` component and also other components from IDF by copying them to the IDF project components directory, but not `bootloader_support`.

As a workaround, can you can move the customizations into the bootloader component? You should be able to edit subproject/main/bootloader_start.c and insert custom initialization code here:

```
void __attribute__((noreturn)) call_start_cpu0()
{
    // 1. Hardware initialization
    if (bootloader_init() != ESP_OK) {
        bootloader_reset();
    }

    // Add custom init functionality here <------

    // 2. Select the number of boot partition
```

And this should work.

This is actually how we recommend customizing ESP-IDF bootloader, as it means that when IDF is updated the `bootloader_support` functions are automatically updated as well, and minimal or no changes should be needed in the custom bootloader itself.

If you have a customization that can't be applied in `bootloader`, please let us know and we'll try to find a way to support that in IDF as well.

We will also try to find the root cause of why `bootloader_support` component is not overrid-eable in the project, as this should work.

negativekelvin, 2019-09-06:
If you put it here 
`project_folder/components/bootloader/subproject/components/bootloader_support` 
instead of here
 `project_folder/components/bootloader_support`
 it should work

Check with
`make bootloader-list-components`

ESP32DE, 2019-09-06:
@projectgus 
@negativekelvin 
thank so much you guys, i will try your suggestions and let you know asap with an update here.
( i am just in time on other place ).

best wishes
rudi ;-)

ESP32DE, 2019-09-06:
update:
@projectgus 
thank you, the workaround works for me , also 
@negativekelvin, 
if i put the bootloader_support as component under the **component bootloader** in the project. this is from doing i think the best choice in the moment, so the bootloader_support is under the bootloader component and do the work how it is named and we can work in this files.

`make bootloader-list-components
`
works.

@projectgus 
I am newfound to learn _the root cause of why **bootloader_support component** is not overrid-eable in the project, as this should work_. perhabs i get few time for a deeper search here.

thank you for your support!
best wishes
rudi ;-)

ESP32DE, 2019-09-06:
@projectgus @negativekelvin 
feel free to close or let open the issues.
for me the things work just in time with your suggestion.
thank you !!

projectgus, 2019-09-09:
Hi @ESP32DE ,

Glad to hear the workaround is useful for you.

> feel free to close or let open the issues.

Will leave this open for now as we should either fix this case so that when overriding a component in the project it also is overriden in the bootloader subproject, or at least document this as a known limitation.

igrr, 2022-04-07:
The functionality for overriding bootloader components has been added in https://github.com/espressif/esp-idf/commit/a79acb413e4b34885f5004701e99d71fcca7d41f, sorry for not updating the issue in time.

closed 2022-04-07 by igrr