espressif/esp-idf #4027: esp32s2

martinberlin, 2019-09-05:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

martinberlin, 2019-09-05:
Sorry opened this without intent. I'm compiling successfully but still not able to see it booting correctly. I will double check my installation and then make a proper ticket in case I don't make it. 
ESP-ROM ver:v1.0.1
Build:Nov 22 2018 17:03:19
rst:0x7 (TG0WDT_SYS_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
PC:0x4000bf50
configspi: 0x00000000, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:0x4
load:0x3fff001c,len:0x19b0
load:0x40078000,len:0x29c0
load:0x40080400,len:0x1a70
SHA-256 comparison failed:
Calculated: cbb9ddf798e457e3579dda54552a7da922cf6ba231b7cf5ac5ebc1e309929a5d
Expected: a705b333b9bcdf4141f3a3ebc8044b1eb6ce97a005789113d20c7c2a58befc13
Attempting to boot anyway...
entry 0x4008076c
Panic at 4008076e
Fatal exception (0): IllegalInstruction
epc1=0x4000f49c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

closed 2019-09-05 by martinberlin

Alvin1Zhang, 2019-09-06:
@martinberlin Thanks for reporting this.

martinberlin, 2019-09-06:
Actually I wanted to search for ESP32S2 issues and instead open a new one...
It compiles good! The issue is that at the moment you need to always delete /builds directory and sdkconfig to run this line properly:
 idf.py -D IDF_TARGET=esp32s2beta menuconfig

Without that it will compile but not boot since the esp32s2beta is not set by default. My suggestion is that this is done automatically since it's possible to detect what chip it is. 
I'm now researching if I can also compile C/C++ with Platformio so I left a ticket in their github:
https://github.com/platformio/platformio-core/issues/2979