espressif/esp-idf #4028: nimble: please fix or ignore multiple "discards 'const' qualifier" warnings

ecc1, 2019-09-05:
Compiling the components/bt/host/nimble library produces numerous warnings like this:
```
In file included from /opt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:57:
/opt/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os_mempool.h:174:69: note: expected 'char *' but argument is of type 'const char *'
                            uint32_t block_size, void *membuf, char *name);
                                                               ~~~~~~^~~~
/opt/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4800:65: warning: passing argument 6 of 'stats_init_and_reg' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         STATS_SIZE_32), STATS_NAME_INIT_PARMS(ble_gattc_stats), "ble_gattc");
                                                                 ^~~~~~~~~~~
```

Please correct these (by including the const declaration in the types of the relevant formal parameters) or ignore them (by adding the -Wno-discarded-qualifiers flag for gcc for this component).

Alvin1Zhang, 2019-09-06:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ecc1, 2019-09-06:
Development Kit: none
Module or chip used: N/A problem occurs during build
IDF version (run git describe --tags to find it): v4.1-dev-141-ga7e8d87d3
Build System: problem occurs only with (legacy) make, not with cmake
Compiler version (run xtensa-esp32-elf-gcc --version to find it): (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Linux
Power Supply: N/A problem occurs during build

Problem Description: see first message in thread

Build output from make and sdkconfig file attached.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3585932/sdkconfig.txt)
[make-log.txt](https://github.com/espressif/esp-idf/files/3585936/make-log.txt)

prasad-alatkar, 2019-09-10:
Hi @ecc1 , once you do `make menuconfig`, please disable `Enable -Wwrite-strings warning flag` from `Compiler options-->`.  Otherwise, you can simply comment out `CONFIG_COMPILER_WARN_WRITE_STRINGS=y` from your sdkconfig.

ecc1, 2019-09-10:
Thanks for this workaround. (I still think it would be better to fix the few places that aren't respecting "const" declarations in the code, or to apply this compiler flag only to the nimble component until it's fixed, since this warning can detect real problems.)

prasad-alatkar, 2019-09-16:
Yes @ecc1 , I will fix this issue by raising PR on NimBLE upstream stack.

prasad-alatkar, 2019-10-23:
Hi @ecc1, This issue should be fixed with commit- 19375a67

mahavirj, 2019-10-24:
Closing per above.