espressif/esp-idf #4029: tools/mass_mfg/mfg_gen.py生产工具不可用

enric-lee, 2019-09-06:
tools/mass_mfg/mfg_gen.py执行出错，v4.0版本错误log:

> python mfg_gen.py generate multipule_mfg_config.csv multipule_mfg_values.csv scgw 0x4000
> Created CSV file: ===> ./csv/scgw-1.csv
> **'module' object has no attribute 'generate'**

v3.2版本错误log:
> python mfg_gen.py --conf multipule_mfg_config.csv --values  multipule_mfg_values.csv --prefix scgw --size 0x4000
> CSV Generated:  ./csv/scgw-1.csv
> **nvs_part_gen() got an unexpected keyword argument 'input_size'**

multipule_mfg_config.csv文件内容：
scgw-key,namespace,
DeviceName,data,string
DeviceSecret,data,string
ProductKey,data,string

multipule_mfg_values.csv文件内容：
id,DeviceName,DeviceSecret,ProductKey
1,d8905151,Piwto064haIA59x3Z0,a1TGu0u
2,d8905152,Piwto064haIA59x3Z1,a1TGu0u
3,d8905153,Piwto064haIA59x3Z2,a1TGu0u
4,d8905154,Piwto064haIA59x3Z3,a1TGu0u
5,d8905155,Piwto064haIA59x3Z4,a1TGu0u

Alvin1Zhang, 2019-09-11:
@enric-lee Thanks for reporting this, we have looked into this and will provide updates if available. Thanks.

enric-lee, 2019-09-12:
It's a good project. I hope it's getting better and better. ^_^

shivani-tipnis, 2019-09-25:
Hi @enric-lee 
Could you please confirm your IDF_PATH. 
Say if you are running **mfg_gen.py** on **v4.0**, the **IDF_PATH** needs to be set to the **same version(v4.0)**. 
Similarly, if you are running it on **v3.2**, **IDF_PATH** needs to be set to the **same version(v3.2).**

Alvin1Zhang, 2019-10-11:
@enric-lee Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

enric-lee, 2019-10-12:
> Hi @enric-lee
> Could you please confirm your IDF_PATH.
> Say if you are running **mfg_gen.py** on **v4.0**, the **IDF_PATH** needs to be set to the **same version(v4.0)**.
> Similarly, if you are running it on **v3.2**, **IDF_PATH** needs to be set to the **same version(v3.2).**

**that's what i do:**

> root@iZuf6bkar79cnzuht5c71uZ:~/esp-idf/esp-idf-v3.2# ls
> root@iZuf6bkar79cnzuht5c71uZ:~/esp-idf/esp-idf-v3.2# git clone --recursive  https://github.com/espressif/esp-idf.git -b v3.2
> Cloning into 'esp-idf'...
> remote: Enumerating objects: 44, done.
> remote: Counting objects: 100% (44/44), done.
> remote: Compressing objects: 100% (36/36), done.
> remote: Total 112379 (delta 8), reused 16 (delta 3), pack-reused 112335
> Receiving objects: 100% (112379/112379), 80.18 MiB | 274.00 KiB/s, done.
> Resolving deltas: 100% (81273/81273), done.
> Checking connectivity... done.
> Note: checking out '286202caa31b61c2182209f37f8069a0b60fb942'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
> Submodule 'components/aws_iot/aws-iot-device-sdk-embedded-C' (https://github.com/espressif/aws-iot-device-sdk-embedded-C.git) registered for path 'components/aws_iot/a
> Submodule 'components/bt/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/lib'
> Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
> Submodule 'components/esp32/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp32/lib'
> Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
> Submodule 'components/expat/expat' (https://github.com/libexpat/libexpat.git) registered for path 'components/expat/expat'
> Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
> Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
> Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
> Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
> Submodule 'components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/micro-ecc/micro-ecc'
> Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
> Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
> Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c) registered for path 'components/protobuf-c/protobuf-c'
> Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
> Cloning into 'components/asio/asio'...
> remote: Enumerating objects: 45479, done.
> remote: Total 45479 (delta 0), reused 0 (delta 0), pack-reused 45479
> Receiving objects: 100% (45479/45479), 15.38 MiB | 246.00 KiB/s, done.
> Resolving deltas: 100% (30490/30490), done.
> Checking connectivity... done.
> Submodule path 'components/asio/asio': checked out '55efc179b76139c8f9b44bf22a4aba4803f7a7bd'
> Cloning into 'components/aws_iot/aws-iot-device-sdk-embedded-C'...
> remote: Enumerating objects: 1292, done.
> remote: Total 1292 (delta 0), reused 0 (delta 0), pack-reused 1292
> Receiving objects: 100% (1292/1292), 878.44 KiB | 479.00 KiB/s, done.
> Resolving deltas: 100% (716/716), done.
> Checking connectivity... done.
> Submodule path 'components/aws_iot/aws-iot-device-sdk-embedded-C': checked out '299183238ffe7a3e6a5ca0af9db19c10eaca62cf'
> Cloning into 'components/bt/lib'...
> remote: Enumerating objects: 39, done.
> remote: Counting objects: 100% (39/39), done.
> remote: Compressing objects: 100% (27/27), done.
> remote: Total 652 (delta 25), reused 26 (delta 12), pack-reused 613
> Receiving objects: 100% (652/652), 2.67 MiB | 828.00 KiB/s, done.
> Resolving deltas: 100% (439/439), done.
> Checking connectivity... done.
> Submodule path 'components/bt/lib': checked out '6c3379db92017f5713e807b60208b62c205d2e05'
> Cloning into 'components/coap/libcoap'...
> remote: Enumerating objects: 32, done.
> remote: Counting objects: 100% (32/32), done.
[... 189 lines cut ...]
>   Stored in directory: /root/.cache/pip/wheels/92/a2/51/5ed9eedca157b6decba18a0cf880380c9fd9ed6dca46fa791c
> Successfully built pycparser
> Installing collected packages: enum34, pycparser, cffi, asn1crypto, ipaddress, cryptography, pyopenssl
> Successfully installed asn1crypto-1.0.1 cffi-1.12.3 cryptography-2.7 enum34-1.1.6 ipaddress-1.0.22 pycparser-2.19 pyopenssl-19.0.0
> WARNING: You are using pip version 19.1.1, however version 19.2.3 is available.
[... cut ...]
> You should consider upgrading via the 'pip install --upgrade pip' command.
> 
> 
> root@iZuf6bkar79cnzuht5c71uZ:~/esp-idf/esp-idf-v3.2/esp-idf/tools/mass_mfg# 
> root@iZuf6bkar79cnzuht5c71uZ:~/esp-idf/esp-idf-v3.2/esp-idf/tools/mass_mfg# python mfg_gen.py --conf multipule_mfg_config.csv --values multipule_mfg_values.csv --prefix scgw --size 0x4000
> CSV Generated:  ./csv/scgw-1.csv
> nvs_part_gen() got an unexpected keyword argument 'input_size'
> root@iZuf6bkar79cnzuht5c71uZ:~/esp-idf/esp-idf-v3.2/esp-idf/tools/mass_mfg#

shivani-tipnis, 2019-10-15:
Hi @enric-lee 
Thanks for the sending in the steps. 
Could you please checkout to branch release/v3.2 or update to v3.2.3 bugfix release, you should be able to run it then.
Thanks.

Alvin1Zhang, 2019-12-13:
@enric-lee Thanks for reporting, will close due to short of feedback, feel free to reopen if the issue still exists. Thanks.

closed 2019-12-13 by Alvin1Zhang