espressif/esp-idf #4030: Crash when vfs select is frequently invoked

KNpTrue, 2019-09-06:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.1-dev-141-ga7e8d87d3
- Build System:         Make
- Compiler version:  esp32-2019r1
- Operating System:     macOS
- Power Supply:         USB

## Crash when vfs select is frequently invoked

### Expected Behavior
Running normally.
### Actual Behavior
Crash. Log reference below. Every time it print a different log.
### Steps to repropduce

Running the test example, then it crash.

### Code to reproduce this issue
Test example in https://github.com/KNpTrue/select_err_example, it copy from esp-idf select example but remove the print log and cancel delay. We should frequently wake up select to allow problems to occur.

## Debug Logs

```
/Users/zebin/Documents/esp/esp-idf/components/freertos/queue.c:1305 (xQueueGiveFromISR)- assert failed!
abort() was called at PC 0x40088792 on core 1
0x40088792: ppCalTxAMPDULength at ??:?

ELF file SHA256: a0386d652b0d2d5ae49285b3b6af61d2ed2c4a4b7b5cee875867e48eef6a0105

Backtrace: 0x40085a11:0x3ffb0b90 0x40085d89:0x3ffb0bb0 0x40088792:0x3ffb0bd0 0x400e24f9:0x3ffb0bf0 0x400e2829:0x3ffb0c20 0x400e4779:0x3ffb0c40 0x40082119:0x3ffb0c90 0x4000bfed:0x3ffbb1a0 |<-CORRUPTED
0x40085a11: ram_read_sar_dout at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:265

0x40085d89: set_chan_dig_gain at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:1900

0x40088792: ppCalTxAMPDULength at ??:?

0x400e24f9: _svfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1603

0x400e2829: _svfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1695 (discriminator 11)

0x400e4779: select_notif_callback_isr at /Users/zebin/Documents/esp/esp-idf/components/vfs/vfs_uart.c:392

0x40082119: _xt_coproc_exc at /Users/zebin/Documents/esp/esp-idf/components/freertos/xtensa_vectors.S:1090
```

dobairoland, 2019-09-09:
Hi @KNpTrue. Thank you for reporting the issue. I reproduced it and fixed with the following patch.

[patch.zip](https://github.com/espressif/esp-idf/files/3590622/patch.zip)

Could you please confirm that the patch works for you as well?

KNpTrue, 2019-09-10:
> Hi @KNpTrue. Thank you for reporting the issue. I reproduced it and fixed with the following patch.
> 
> [patch.zip](https://github.com/espressif/esp-idf/files/3590622/patch.zip)
> 
> Could you please confirm that the patch works for you as well?

Thanks @dobairoland. This patch fixes the problem in the sample, but in my project, it often crashesit always when I use both lwip select and uart select. Let me see how to reproduce the problems encountered in the project. 
This is the log in my project:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008e98d  PS      : 0x00060530  A0      : 0x80149d99  A1      : 0x3ffcb6b0  
0x4008e98d: xQueueGenericSend at /Users/zebin/Documents/esp/esp-idf/components/freertos/queue.c:720 (discriminator 1)

A2      : 0x800d4f3b  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffd6734  A7      : 0x3ffd6664  A8      : 0x00185000  A9      : 0x00002000  
A10     : 0x0000ff00  A11     : 0x3ffd671c  A12     : 0x3ffd6734  A13     : 0x000000ff  
A14     : 0x00000000  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d4f7b  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

ELF file SHA256: fce9cec521fdf97a3c5dddc8f9b6d5cf79a3c21ff6790b26dd13f8a9c328edcb

Backtrace: 0x4008e98a:0x3ffcb6b0 0x40149d96:0x3ffcb6f0 0x4014b9f2:0x3ffcb710 0x4014c70c:0x3ffcb730 0x40137b7d:0x3ffcb750 0x40137c24:0x3ffcb770 0x4008e479:0x3ffcb7a0
0x4008e98a: xQueueGenericSend at /Users/zebin/Documents/esp/esp-idf/components/freertos/queue.c:720 (discriminator 2)

0x40149d96: sys_sem_signal at /Users/zebin/Documents/esp/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:133

0x4014b9f2: lwip_netconn_do_writemore at /Users/zebin/Documents/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c:1846

0x4014c70c: lwip_netconn_do_write at /Users/zebin/Documents/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c:1891

0x40137b7d: tcpip_thread_handle_msg at /Users/zebin/Documents/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:168

0x40137c24: tcpip_thread at /Users/zebin/Documents/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

0x4008e479: vPortTaskWrapper at /Users/zebin/Documents/esp/esp-idf/components/freertos/port.c:143
```

dobairoland, 2019-09-10:
Thank you @KNpTrue for the feedback. I have no clue for you regarding this new Guru Meditation Error. Can I ask you to open a new issue when you narrow down the problem? I think I will go forward and merge this fix.

KNpTrue, 2019-09-10:
Ok, I will open a new issue.

closed 2019-09-10 by KNpTrue

martinling, 2019-11-20:
@KNpTrue were you able to diagnose or solve the LoadProhibited crash above?

I am getting crashes with that same stack trace on IDF v3.3 right now, in a project using select() on several LWIP sockets and one UART.

elcritch, 2020-06-30:
I’m getting the same issue and stack trace when using uart and select. IDF v4.0.1.

dobairoland, 2020-06-30:
@elcritch The fix was backported to v4.0: https://github.com/espressif/esp-idf/commit/63c2834d7bcb44706ec1ba1cb83d170465350b84

If you have this fix then please open a new issue for this.