espressif/esp-idf #4031: pacman -S mingw-w64-i686-python2-<package_name>

shibaoyi, 2019-09-06:
The recommended way to install a packages is via "pacman". Please run "pacman -Ss <package_name>" for searching the package database and if found then "pacman -S mingw-w64-i686-python2-<package_name>" for installing it.
NOTE: You may need to run "pacman -Syu" if your package database is older and run twice if the previous run updated "pacman" itself.
Please read https://github.com/msys2/msys2/wiki/Using-packages for further information about using "pacman"
WARNING: The cryptography package have dependencies on system packages so please make sure you run "pacman -Syu" followed by "pacman -S mingw-w64-i686-python2-cryptography".
Alternatively, you can run "D:/msys32/mingw32/bin/python.exe -m pip install --user -r D:/msys32/home/root/esp/esp-idf/requirements.txt" for resolving the issue.
make: *** [D:/msys32/home/root/esp/esp-idf/make/project.mk:479：check_python_dependencies] 错误 1

negativekelvin, 2019-09-07:
Alternatively, you can run `D:/msys32/mingw32/bin/python.exe -m pip install --user -r D:/msys32/home/root/esp/esp-idf/requirements.txt` for resolving the issue.

Did you do it? What is your question?

dobairoland, 2019-09-09:
@shibaoyi 

Which packages are not installed? Please give us the full log.

The cryptography package cannot be installed by `pip` in MSYS. You have to run the following command: `pacman -S mingw-w64-i686-python2-cryptography`

The installed version can be checked by:
`pacman -Qs mingw-w64-i686-python2-cryptography`
which prints an output similar to this:
```
local/mingw-w64-i686-python2-cryptography 2.7-1
    A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
```
where you can see that version 2.7 is installed (in my case). ESP-IDF requires version 2.1.4 or newer. If you have an older version then you have to update with the command: `pacman -Syu`.

shibaoyi, 2019-09-09:
> @shibaoyi
> 
> 哪些软件包没有安装？请把完整的日志给我们。
> 
> 无法通过`pip`在MSYS。您必须运行以下命令：`pacman -S mingw-w64-i686-python2-cryptography`
> 
> 已安装的版本可以通过以下方式进行检查：
> `pacman -Qs mingw-w64-i686-python2-cryptography`
> 输出与此类似的输出：
> 
> ```
> local/mingw-w64-i686-python2-cryptography 2.7-1
>     A package designed to expose cryptographic recipes and primitives to Python developers (mingw-w64)
> ```
> 
> 您可以看到安装了2.7版本(在我的例子中)。esp-以色列国防军需要2.1.4或更高版本。如果您的版本较旧，则必须使用以下命令进行更新：`pacman -Syu`.

### **Yes, thank you.**

dobairoland, 2019-09-09:
@shibaoyi so your issue got solved?

shibaoyi, 2019-09-09:
> @shibaoyi so your issue got solved?

Yes, it's settled.

dobairoland, 2019-09-09:
I'm happy that it got solved!

closed 2019-09-09 by dobairoland

julred, 2021-05-12:
I updated the pacman repository via pacman -Syu, but I have problems to install the cryptography module
```
$ pacman -S mingw-w64-i686-python2-cryptography
error: target not found: mingw-w64-i686-python2-cryptography
```

Any suggestions?

dobairoland, 2021-05-12:
@julred The package is now called mingw-w64-i686-python-cryptography. For other dependencies please see: https://github.com/espressif/esp-idf/blob/master/tools/windows/windows_install_prerequisites.sh#L36.

There is also a pre-ZIPed environment ready to use: https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started/windows-setup.html#toolchain-setup (for ESP-IDF v3.3 and earlier)