espressif/esp-idf #4033: ESP32 CAM will not connect to Access Point in station mode

jhinklejr, 2019-09-08:
I've never had a connection issue before.

Just got an ESP32 CAM - using IDF to load the WIFI - Station sample.

Set LWIP to always be on core 0.

I can write and execute code just fine in core 1.

I Put the SID and Password in the simple WIFI Station example and it fails to connect.

All my previous experience has been with the ESP8266 so this is the first time attempting to get the ESP32 working.

I have the crystal identified as 40 mhz.

Any help or comments appreciated.

Thanks.

Joe

Alvin1Zhang, 2019-09-09:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks. @jhinklejr

jhinklejr, 2019-09-09:
I was hoping you would say ... "Dummy - you forgot this ..."

Question - how do you search the issue list (open and close) to see if an issue had been previously addressed?

I upgraded to the latest IDF - I have attached my main.c, sdkconfig.h and an output capture of the boot up process showing the failure.

I'm hoping its a bad sdkconfig setting.

Thanks in advance for your help.

Joe

[WIFI_No_Connect_AP_Issue.zip](https://github.com/espressif/esp-idf/files/3590421/WIFI_No_Connect_AP_Issue.zip)

jhinklejr, 2019-09-10:
works now - no idea of roor cause

closed 2019-09-10 by jhinklejr

espxiehang, 2019-11-06:
I used the code you provided and found no problems.