espressif/esp-idf #4034: running builds from a readonly esp-idf directory (compile error during config)

hamishcunningham, 2019-09-09:
hi

I'm using ESP IDF with a class of 120 students, and need to avoid having a complete copy of everything for each one, so we're trying to have a read-only `esp-idf` directory. This triggers a compile error during config:

> make[1]: Entering directory '/opt/unphone/sdks/esp-idf/tools/kconfig'
> cc -c  -DCURSES_LOC="<ncurses.h>" -DLOCALE -MMD  /opt/unphone/sdks/esp-idf/tools/kconfig/mconf.c -o mconf.o
> /opt/unphone/sdks/esp-idf/tools/kconfig/mconf.c:1053:1: fatal error: opening dependency file mconf.d: Permission denied
>  }
>  ^
> compilation terminated.
> Makefile:171: recipe for target 'mconf.o' failed
> make[1]: *** [mconf.o] Error 1
> make[1]: Leaving directory '/opt/unphone/sdks/esp-idf/tools/kconfig'
> make: *** No rule to make target '/opt/unphone/sdks/esp-idf/tools/kconfig/conf-idf', needed by '/opt/ltsphome/cs51ltsp01/unphone-master/examples/BigDemoIDF/build/include/config/auto.conf'. Stop.
> 

Any ideas? Is there a way to configure the build to only write to the project's `build/` directory?
Thanks!
Hamish Cunningham

dobairoland, 2019-09-11:
Hi @hamishcunningham. Please try to run `make` in `esp-idf/tools/kconfig` before making it read-only. It should work after that.

hamishcunningham, 2019-09-15:
Thanks @dobairoland!  I get a bit further now, but then:

> make app
> GENCONFIG
> 
> *** Error during update of the configuration.
> 
> make: *** No rule to make target '/opt/ltsphome/ac1hc/unphone/examples/BigDemoIDF/build/include/config/auto.conf', needed by '/opt/ltsphome/ac1hc/unphone/examples/BigDemoIDF/build/app_trace/component_project_vars.mk'. Stop.

Any ideas?
Thanks!

dobairoland, 2019-09-16:
Probably `/opt/ltsphome/ac1hc/unphone/examples/BigDemoIDF/build/include/config/auto.conf` was written with higher privileges and the current user is not allowed to write it. 

`sudo rm -r /opt/ltsphome/ac1hc/unphone/examples/BigDemoIDF/build` would probably help but better start from the beginning. Here is how i did it:

1. Clone ESP-IDF and initialize submodules as a regular user (developer).
2. Run menuconfig for `hello_world` example application and build the application just to make sure that everything is working.
3. Copy ESP-IDF to a read-only place. I did `sudo cp -r $IDF_PATH /opt`.
4. I was able to build applications as a regular user after setting the IDF_PATH to the new (read-only) location.

Does these steps sufficient for you?

hamishcunningham, 2019-09-16:
Great, thanks, that works for me :)

closed 2019-09-16 by hamishcunningham