espressif/esp-idf #4036: BLE get disconnected ::: cif=3 connected=0 conn_id=3 reason=0x0013

tharunjoy, 2019-09-09:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version:  v3.3-dev-270-g6f8e2b018
- Build System:         Make
- Compiler version :   1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
I am running ESP IDF as a Gatt client that is connected to a remote. I was able to connect to the remote and read the uuids for battery and HID profiles. But immediately, the client is getting disconnected with 
`:::: 
I (11286) GATT_CLIENT :::: ESP_GATTC_REG_FOR_NOTIFY_EVT
I (11296) GATT_CLIENT :::: EVT 38, gattc if 3
I (11296) GATT_CLIENT :::: ****HID Profile Event****
I (11286) GATT_CLIENT ::::
I (11366) GATT_CLIENT :::: ESP_GATTC_REG_FOR_NOTIFY_EVT
I (11376) GATT_CLIENT :::: EVT 38, gattc if 4
I (11376) GATT_CLIENT :::: ****Battery Profile Event****
I (11386) GATT_CLIENT :::: 
I (11386) GATT_CLIENT :::: ESP_GATTC_REG_FOR_NOTIFY_EVT
D (14556) BT_GATT: GATT   ATT protocol channel with BDA: 000XXXXXXX16 is disconnected
D (14556) BT_GATT: gatt_is_bda_in_the_srv_chg_clt_list :00-0X-XX-XX-XX-16
D (14566) BT_GATT: gatt_cleanup_upon_disc 
D (14566) BT_GATT: found p_tcb 
D (14576) BT_GATT: gatt_set_ch_state: old=4 new=0
D (14576) BT_GATT: gatt_free_pending_ind
D (14576) BT_GATT: gatt_free_pending_enc_queue
D (14586) BT_GATT: gatt_free_pending_prepare_write_queue
D (14586) BT_GATT: found p_reg tcb_idx=0 gatt_if=1  conn_id=0x1
D (14596) BT_GATT: gatt_connect_cback: from 000XXXXXXX16 connected:0 conn_id=1 reason = 0x0013
D (14606) BT_GATT: gatt_get_ch_state: ch_state=0
I (14606) BT_GATT: GATT_GetConnIdIfConnected status=0

D (14616) BT_GATT: found p_reg tcb_idx=0 gatt_if=2  conn_id=0x2
D (14626) BT_GATT: found p_reg tcb_idx=0 gatt_if=3  conn_id=0x3
W (14626) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013
D (14636) BT_GATT: found p_reg tcb_idx=0 gatt_if=4  conn_id=0x4
W (14646) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0013
D (14646) BT_GATT: gatt_delete_dev_from_srv_chg_clt_list
D (14656) BT_GATT: gatt_is_bda_in_the_srv_chg_clt_list :00-0X-XX-XX-XX-16
D (14666) BT_GATT: ATT disconnected
I (14666) GATT_CLIENT :::: EVT 41, gattc if 3
I (14676) GATT_CLIENT :::: ****HID Profile Event****
I (14676) GATT_CLIENT :::: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (14686) GATT_CLIENT :::: EVT 5, gattc if 3
I (14686) GATT_CLIENT :::: ****HID Profile Event****
Unhandled GATT Profile Event: 5
I (14696) GATT_CLIENT :::: EVT 41, gattc if 4
I (14706) GATT_CLIENT :::: ****Battery Profile Event****
I (14706) GATT_CLIENT :::: ESP_GATTC_DISCONNECT_EVT, reason = 0x13
I (14716) GATT_CLIENT :::: EVT 5, gattc if 4
I (14716) GATT_CLIENT :::: ****Battery Profile Event****
Unhandled GATT Profile Event: 5
`
I found the client is getting disconnected when the server stop advertising.

How to resolve this issue?
I tried auto-connect by putting the flag 'false' in the 'open' function, which does not initiate any connections.

Alvin1Zhang, 2019-09-11:
@tharunjoy  Thanks for reporting the issue and we will look into it. Thanks.

Alvin1Zhang, 2019-09-16:
@tharunjoy Thanks for the updates, we have looked into this issue, and will update once we have. Thanks.

junexunxp, 2019-09-16:
Hi @tharunjoy,
Could you tell me the type of the remote device you use? Is it a BLE mouse or BLE keyboard or a BT mouse or keyboard?

junexuesp, 2019-09-16:
Hi @tharunjoy,
Could you tell me the type of the remote device you use? Is it a BLE mouse or BLE keyboard or a BT mouse or keyboard?
Thanks

tharunjoy, 2019-09-16:
Hi @junexuesp ,
The remote which I am using is BLE mouse and Keyboard.
The image of the remote is attached here [https://attinadsoftware-my.sharepoint.com/:i:/p/tharun_joy/EXoLBC3ep2tDpRSkPYgOTUUBTZiuxe45hStUnMJ0Bn761Q?e=YsEbuD](url).

The code is attached here [https://attinadsoftware-my.sharepoint.com/:u:/p/tharun_joy/EQbayrTPqedCrgISWUQ05SYBAoKNdoMISkIGKL73gqOweA?e=hKpfp2](url). Please tell me if I am doing something wrong.

The specification of bluetooth(ATB110x) is as below:
- Complies with bluetooth ® Low Energy Core Specification V5.0（QDID:112994 and 103105）
- Low power and excellent performance 2.4GHz transceiver built-in balun for compact layout area Master and slave mode support.
- Industry leading power consumption BLE Active Rx: 3.1mA（with 
- DCDC）BLE Active Tx: 3.5mA@0dBm（with DCDC）
- BLE Active Rx: 4.2 mA（without DCDC）
- BLE Active Tx: 6.9mA@0dBm（without DCDC）,11mA@5dBm（without DCDC）
- BLE Sleep mode: 700 nA
- Standby mode with all wakeup sources (with SRAM retention) : <15uA
- Standby mode without any wakeup sources (CHIPEN=0) :< 300nA (without flash memory)
- RX sensitivity: -93.5dBm(30.8% PER, Dirty,Mix Mode）
- TX power range: -20 to +5 dBm(2/4 layer PCB)
- BT qualified, FCC & ETSI compliant
- Support standalone mode
- Support HCI mode with standard HCI interface
 
Thank you.

junexuesp, 2019-09-16:
Hi @tharunjoy,
The disconnection error code in your example is 0x13 which means the remote device terminated the connection. Did you the same security method both on the device and esp32?
It would be best if you can send me the sniffer packets(air interface). If not I think you should give me the whole project thant I can run on my side to repeat your problem.
Best
June

Alvin1Zhang, 2019-10-11:
@tharunjoy Thanks for reporting, would you please help share if any updates, sniffer packets, or more details of the issue? Thanks.

tharunjoy, 2019-10-14:
Hy Sorry for the late reply. We don't have bluetooth sniffer with us. We are using the same bluetooth gatt_security_client example and a similar remote [(https://www.banggood.in/Q5-bluetooth2_4GHz-WIFI-Voice-Remote-Control-Air-Mouse-With-USB-Receiver-For-Smart-TV-Android-Box-p-1301925.html?akmClientCountry=IN&gmcCountry=IN&currency=INR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=lijing&utm_campaign=ssc-ing-all-newcustom-1-0822&gclid=CjwKCAjwlovtBRBrEiwAG3XJ-zUC4Te0ictC1DJddTRqzYkuvXs0rydk_pJ2BIMfbqe-piZz8X7XcRoC4FUQAvD_BwE&ID=532942&cur_warehouse=CN)]

Alvin1Zhang, 2019-10-14:
@tharunjoy Thanks for the updates. Feel free to reopen if you still have the issue, thanks.

closed 2019-10-14 by Alvin1Zhang