espressif/esp-idf #4037: make error

shibaoyi, 2019-09-09:
make：python：命令未找到
make: *** [D:/msys32/home/Administrator/esp32/esp-idf-v3.2/make/project.mk:479：check_python_dependencies] 错误 127

478: ifndef IS_BOOTLOADER_BUILD
479:	$(PYTHON) $(IDF_PATH)/tools/check_python_dependencies.py
480: endif

Alvin1Zhang, 2019-09-09:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks. @shibaoyi

dobairoland, 2019-09-10:
Based on the available information I would say that either you don't have a Python interpreter installed or it is not accessible. `CONFIG_PYTHON` in your `sdkconfig` file defines "python" so it must be a valid executable. If you running this from console then `python --version` should print the version of your Python interpreter. 

Please visit https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started/index.html#setup-toolchain for further information.

dobairoland, 2019-09-25:
Hi @shibaoyi. Do you have any update on this issue?

dobairoland, 2019-10-02:
I'm closing this for inactivity. Please feel free to reopen it if the issue is still unsolved.

closed 2019-10-02 by dobairoland