espressif/esp-idf #4038: Linking Issue

fvizzon, 2019-09-09:
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version 
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.1-dev-141-ga7e8d87d
- Build System:         CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     macOs & Ubunut
- Power Supply:         USB

## Problem Description
When building and flahsing example projects everything works just fine.
Later i started adding src files to the projec.
When building all files are compiled succesfully, 
while the project is been linked my src files cant be found.
Below code samples and build log.

### Steps to repropduce

1. Add dummy.c and dummy.h to blink example code.
2. include and call doNothing() function defined at dummy.h
3. Build

### Code to reproduce this issue

```cpp
// dummy.c
#include "dummy.h"

int doNothing.h(int i)
{
    return i;
}

```

```cpp
// dummy.h
#ifndef _DUMMY_H_
#define _DUMMY_H_

int doNothing(int i);

#endif

```
```cpp
//blink.c
/* Blink Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include "dummy.h"

/* Can use project configuration menu (idf.py menuconfig) to choose the GPIO to blink,
   or you can edit the following line and set a number here.
*/
#define BLINK_GPIO CONFIG_BLINK_GPIO
[... 25 lines cut ...]

## Debug Logs

```
Checking Python dependencies...
Python requirements from /Users/fede/esp/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory /Users/fede/esp/esp-idf/examples/get-started/blink/build
Executing "ninja all"...
[1/8] Performing build step for 'bootloader'
ninja: no work to do.
[5/6] Linking CXX executable blink.elf
FAILED: blink.elf
: && /Users/fede/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/blink.elf.dir/project_elf_src.c.obj  -o blink.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tls/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-idf/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/main/libmain.a -Wl,--cref -Wl,--Map=/Users/fede/esp/esp-idf/examples/get-started/blink/build/blink.map esp-idf/asio/libasio.a esp-idf/cbor/libcbor.a esp-idf/coap/libcoap.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/fatfs/libfatfs.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/libsodium/liblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp-tls/libesp-tls.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/protocomm/libprotocomm.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/nghttp/libnghttp.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjson.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libcore.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libespnow.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libmesh.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libpp.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/librtc.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libsmartconfig.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libcore.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libespnow.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libmesh.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libpp.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/librtc.a /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32/libsmartconfig.a -u esp_app_desc -L /Users/fede/esp/esp-idf/components/esp_wifi/lib_esp32 -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_includ [... cut ...]
/Users/fede/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(blink.c.obj): in function `app_main':
/Users/fede/esp/esp-idf/examples/get-started/blink/build/../main/blink.c:29: undefined reference to `doNothing'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code [1]
```

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3591173/sdkconfig.zip)

igrr, 2019-09-09:
Have you also modified the component CMakeLists.txt file to include the new source file? See https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#minimal-component-cmakelists for an example.

fvizzon, 2019-09-09:
Great!
that was the issue.

closed 2019-09-09 by fvizzon

gojimmypi, 2020-02-01:
fwiw - I had a similar issue with the `ninja: build stopped: subcommand failed.` and `Build finished with exit code 1`  non-descript errors. Turned out in my case that my default app to open `.py` files was VSCode and not Python.

See https://github.com/nanoframework/Home/issues/564