espressif/esp-idf #4039: git submodule update --init --recursive => 'fatal: reference is not a tree...'

mws-rmain, 2019-09-09:
updating repo to latest  with:
```
git pull
git submodule update --init --recursive
```
used to work, now results in the following error:

```
remote: Enumerating objects: 29, done.
remote: Counting objects: 100% (29/29), done.
remote: Total 50 (delta 29), reused 29 (delta 29), pack-reused 21
Unpacking objects: 100% (50/50), done.
From https://github.com/espressif/esp-idf
   148a269..6ccb4cf  release/v3.3 -> origin/release/v3.3
 * [new tag]         v3.3       -> v3.3
Already up-to-date.
fatal: reference is not a tree: 7f8c86e501e690301630029fa9bae22424adf618
Unable to checkout '7f8c86e501e690301630029fa9bae22424adf618' in submodule path 'components/coap/libcoap/ext/tinydtls'
Failed to recurse into submodule path 'components/coap/libcoap'
```

dobairoland, 2019-09-09:
Hi @mws-rmain. Please try this sequence of commands:
```
cd components/coap/libcoap
git submodule sync
git submodule update --init --recursive
cd $IDF_PATH
git submodule update --init --recursive
```

mws-rmain, 2019-09-09:
Thanks   @dobairoland, that fixed it.
So some URL(s) changed in the libcoap submodule?

dobairoland, 2019-09-09:
Yes, that was the issue. I'm happy that it got solved.

closed 2019-09-09 by dobairoland