espressif/esp-idf #4041: ESP32 S2 BETA Programming IDE

shveytank, 2019-09-10:
Hi i recently came to know that ESP32 S2 Beta version is released and i've got one, so i was asking which IDE you guys are using to test it or program it.

Alvin1Zhang, 2019-09-11:
@shveytank Thanks for reporting this.

huming2207, 2019-09-12:
CLion should work as it's still shares the same CMake build system. VSCode with vscode-cmake-tool plugin also works, but it is kinda outdated (should get updates later this month) and it won't handle the include path correctly sometimes.

shveytank, 2019-09-12:
@huming2207 would you like to share with me that how you programmed it because i am familiar with Arduino IDE, ESPIDF (a very little), avr studio but not much with VS.

huming2207, 2019-09-12:
@shveytank Just use `idf.py` in the console separately.

shveytank, 2019-09-20:
Please Can you provide me a little walk through how to program it. It would be very helpful .
Thanks in advance.

aallan, 2019-09-22:
Brief walkthrough on using `idf.py` in the console here, https://esp32.com/viewtopic.php?f=13&t=11265

igrr, 2020-09-09:
Closing as both the VS Code extension (https://github.com/espressif/vscode-esp-idf-extension) and Eclipse plugin (https://github.com/espressif/idf-eclipse-plugin) can be used to develop for ESP32-S2.

closed 2020-09-09 by igrr