espressif/esp-idf #4043: idf.py menuconfig failed.

phypy, 2019-09-10:
Hi,Something is wrong while executing the idf.py menuconfig:
Operation system:Windows 10. 
Python Version: Python 2.7.15 :: Anaconda,Inc.
Error info:
D:\Programs\Espressif\hello_world>python ../esp-idf/tools/idf.py menuconfig
Checking Python dependencies...
Python requirements from D:\Programs\Espressif\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory d:\programs\espressif\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 d:\programs\espressif\hello_world"...
Warn about uninitialized values.
-- Could NOT find Git (missing: GIT_EXECUTABLE)
CMake Warning at D:/Programs/Espressif/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  D:/Programs/Espressif/esp-idf/tools/cmake/idf.cmake:34 (include)
  D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

CMake Warning at D:/Programs/Espressif/esp-idf/tools/cmake/kconfig.cmake:39 (message):
  Failed to detect version of mconf-idf.  Return code was 1.
Call Stack (most recent call first):
  D:/Programs/Espressif/esp-idf/tools/cmake/build.cmake:161 (__kconfig_init)
  D:/Programs/Espressif/esp-idf/tools/cmake/idf.cmake:43 (__build_init)
  D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Warning at D:/Programs/Espressif/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain
  D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe
  version 5.2.0 is not the supported version 8.2.0.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  D:/Programs/Espressif/esp-idf/components/esp32/project_include.cmake:20 (gcc_version_check)
  D:/Programs/Espressif/esp-idf/tools/cmake/build.cmake:287 (include)
  D:/Programs/Espressif/esp-idf/tools/cmake/build.cmake:440 (__build_process_project_includes)
  D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Warning at D:/Programs/Espressif/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain
  D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe
  crosstool-ng version crosstool-ng-1.22.0-80-g6c4433a5 doesn't match
  supported version esp32-2019r1.  Check Getting Started documentation or
  proceed at own risk.
Call Stack (most recent call first):
  D:/Programs/Espressif/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  D:/Programs/Espressif/esp-idf/tools/cmake/build.cmake:287 (include)
[... 3 lines cut ...]

-- Found PythonInterp: D:/Programs/Anaconda3/envs/classic/python.exe (found version "2.7.15")
-- Found Perl: D:/Programs/DiaSemi/SmartSnippetsStudio/Tools/mingw64_targeting32/msys/1.0/bin/perl.exe (found version "5.8.8")
-- Adding linker script D:/Programs/Espressif/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script D:/Programs/Espressif/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-locale.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/Programs/Espressif/esp-idf/components/app_trace D:/Programs/Espressif/esp-idf/components/app_update D:/Programs/Espressif/esp-idf/components/asio D:/Programs/Espressif/esp-idf/components/bootloader D:/Programs/Espressif/esp-idf/components/bootloader_support D:/Programs/Espressif/esp-idf/components/bt D:/Programs/Espressif/esp-idf/components/cbor D:/Programs/Espressif/esp-idf/components/coap D:/Programs/Espressif/esp-idf/components/console D:/Programs/Espressif/esp-idf/components/cxx D:/Programs/Espressif/esp-idf/components/driver D:/Programs/Espressif/esp-idf/components/efuse D:/Programs/Espressif/esp-idf/components/esp-tls D:/Programs/Espressif/esp-idf/components/esp32 D:/Programs/Espressif/esp-idf/components/esp_adc_cal D:/Programs/Espressif/esp-idf/components/esp_common D:/Programs/Espressif/esp-idf/components/esp_eth D:/Programs/Espressif/esp-idf/components/esp_event D:/Programs/Espressif/esp-idf/components/esp_gdbstub D:/Programs/Espressif/esp-idf/components/esp_http_client D:/Programs/Espressif/esp-idf/components/esp_http_server D:/Programs/Espressif/esp-idf/components/esp_https_ota D:/Programs/Espressif/esp-idf/components/esp_https_server D:/Programs/Espressif/esp-idf/components/esp_local_ctrl D:/Programs/Espressif/esp-idf/components/esp_ringbuf D:/Programs/Espressif/esp-idf/components/esp_rom D:/Programs/Espressif/esp-idf/components/esp_websocket_client D:/Programs/Espressif/esp-idf/components/esp_wifi D:/Programs/Espressif/esp-idf/components/espcoredump D:/Programs/Espressif/esp-idf/components/esptool_py D:/Programs/Espressif/esp-idf/components/expat D:/Programs/Espressif/esp-idf/components/fatfs D:/Programs/Espressif/esp-idf/components/freemodbus D:/Programs/Espressif/esp-idf/components/freertos D:/Programs/Espressif/esp-idf/components/heap D:/Programs/Espressif/esp-idf/components/idf_test D:/Programs/Espressif/esp-idf/components/jsmn D:/Programs/Espressif/esp-idf/components/json D:/Programs/Espressif/esp-idf/components/libsodium D:/Programs/Espressif/esp-idf/components/log D:/Programs/Espressif/esp-idf/components/lwip D:/Programs/Espressif/hello_world/main D:/Programs/Espressif/esp-idf/components/mbedtls D:/Programs/Espressif/esp-idf/components/mdns D:/Programs/Espressif/esp-idf/components/mqtt D:/Programs/Espressif/esp-idf/components/newlib D:/Programs/Espressif/esp-idf/components/nghttp D:/Programs/Espressif/esp-idf/components/nvs_flash D:/Programs/Espressif/esp-idf/components/openssl D:/Programs/Espressif/esp-idf/components/partition_table D:/Programs/Espressif/esp-idf/components/protobuf-c D:/Programs/Espressif/esp-idf/components/protocomm D:/Pr [... cut ...] /Kconfig && D:\Programs\Anaconda3\envs\classic\python.exe D:/Programs/Espressif/esp-idf/tools/kconfig_new/confgen.py --kconfig D:/Programs/Espressif/esp-idf/Kconfig --sdkconfig-rename D:/Programs/Espressif/esp-idf/sdkconfig.rename --config D:/Programs/Espressif/hello_world/sdkconfig --env-file D:/Programs/Espressif/hello_world/build/config.env --env IDF_TARGET=esp32 --output config D:/Programs/Espressif/hello_world/sdkconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

dobairoland, 2019-09-10:
You need to update `mconf-idf` by either using the new version of the tools installer (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#esp-idf-tools-installer) or kconfig-frontends (https://github.com/espressif/kconfig-frontends/releases/tag/v4.6.0.0-idf-20190628).

phypy, 2019-09-11:
[Uploading Setup Log 2019-09-11 #003.txt…]()

phypy, 2019-09-11:
catched another error while running the esp-idf-tools-setup-2.0.exe,here is the log:
2019-09-11 14:45:53.293   Log opened. (Time zone: UTC+08:00)
2019-09-11 14:45:53.293   Setup version: Inno Setup version 6.0.1-beta (u)
2019-09-11 14:45:53.293   Original Setup EXE: C:\Users\hubei\Downloads\esp-idf-tools-setup-2.0.exe
2019-09-11 14:45:53.293   Setup command line: /SL5="$110F22,204901974,721408,C:\Users\hubei\Downloads\esp-idf-tools-setup-2.0.exe" 
2019-09-11 14:45:53.293   Windows version: 10.0.18362  (NT platform: Yes)
2019-09-11 14:45:53.293   64-bit Windows: Yes
2019-09-11 14:45:53.293   Processor architecture: x64
2019-09-11 14:45:53.293   User privileges: Administrative
2019-09-11 14:45:53.294   Administrative install mode: No
2019-09-11 14:45:53.294   Install mode root key: HKEY_CURRENT_USER
2019-09-11 14:45:53.296   64-bit install mode: Yes
2019-09-11 14:45:53.297   Created temporary directory: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp
2019-09-11 14:45:53.301   -- DLL function import --
2019-09-11 14:45:53.301   Function name: idpAddFile
2019-09-11 14:45:53.301   DLL name: files:idp.dll
2019-09-11 14:45:53.301   Extracting temporary file: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.311   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.311   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpAddFileComp
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpAddMirror
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpAddFtpDir
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpAddFtpDirComp
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpClearFiles
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpFilesCount
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpFtpDirsCount
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpFileDownloaded
2019-09-11 14:45:53.346   DLL name: files:idp.dll
2019-09-11 14:45:53.346   Dest DLL name: C:\Users\hubei\AppData\Local\Temp\is-KK6SF.tmp\idp.dll
2019-09-11 14:45:53.346   Importing the DLL function.
2019-09-11 14:45:53.346   Successfully imported the DLL function. Delay loaded? No
2019-09-11 14:45:53.346   -- DLL function import --
2019-09-11 14:45:53.346   Function name: idpFilesDownloaded
[... 451 lines cut ...]
                              raise ReadTimeoutError(self._pool, None, 'Read timed out.')
                          pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.
                          Creating a new Python environment in D:\Programs\Espressif\ESPIDF_Tools\.espressif\python_env\idf4.1_py3.6_env
                          Installing virtualenv
                          Installing Python packages from D:\Programs\Espressif\esp-idf\requirements.txt
                          Traceback (most recent call last):
                            File "D:\Programs\Espressif\esp-idf\tools\idf_tools.py", line 1317, in <module>
                              main(sys.argv[1:])
[... cut ...]
2019-09-11 14:59:42.314   --------
2019-09-11 14:59:42.314   Message box (OK):
                          Installation has failed with exit code 1
2019-09-11 14:59:58.452   User chose OK.
2019-09-11 14:59:58.467   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2019-09-11 14:59:59.813   User chose Yes.
2019-09-11 14:59:59.829   CurStepChanged raised an exception.

phypy, 2019-09-11:
thanks ，successed after installed a new python version!

closed 2019-09-11 by phypy