espressif/esp-idf #4044: Subscribe local model to group address (BLE mesh)

jkockx, 2019-09-10:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.0-dev-1191-g138c941fa
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I achieved to implement a provisioner which is able to:
- provision a node
- add and bind application keys to both local client models as the server models in nodes
- add subscription to group addresses to nodes

But I can't figure out how to subscribe a model local on my device to a group address. Could someone help me out on this?

Alvin1Zhang, 2019-09-11:
@jkockx Thanks for reporting this, we will look into it. Thanks.

Alvin1Zhang, 2019-10-11:
@jkockx Thanks for reporting this.

JacekDob, 2020-05-23:
Can we reopen it so using mention hack does not have to be used, and subscription shall be persitent like in functions:
esp_ble_mesh_provisioner_add_local_app_key
esp_ble_mesh_provisioner_bind_app_key_to_local_model
?

Campou, 2020-05-25:
Hi @JacekDob 

Thanks for this. We have reopened this, and will add functions for Provisioner to subscribe group addresses.

Thanks.

JacekDob, 2020-05-25:
Hi @Campou,

thank you for that.

Looking forward for this, because apart from provisioner functionality, I am using it also as a Bluetooth Mesh <> IP gateway, and after each reset this information is lost, so please support also persistence feature.

When may it be available more or less?

Thanks in advance,

Jacek

Campou, 2020-05-25:
Hi @JacekDob 

Have submitted a MR for this internally, now it's in the review progress. Will be synchronized to Github maybe this week.

Thanks.

JacekDob, 2020-06-12:
Hi,

any update for this? I don't see anything new in repo.
or patch also would be fine.

JacekDob, 2020-06-16:
Apart from above, how to set 

- network retransmit count and delay
- relay feature (enable/disable, count, delay)
- friend feature

for a provisioner locally?

Alvin1Zhang, 2020-06-16:
Thanks for reporting and updates and sorry for slow turnaround, the fix on master has been available at https://github.com/espressif/esp-idf/commit/35caf57919cd19e473063c9418b093ada3431c47 and the fix on v4.0 is available at https://github.com/espressif/esp-idf/commit/1ab23bfd3d1544f3fc828e443cf1cda9cf1d1022. Feel free to reopen if the issue still happens, and please help file a new ticket for new issue. Thanks.

closed 2020-06-16 by Alvin1Zhang

Campou, 2020-06-16:
Hi @JacekDob 

The implementation of subscribing group address by a local model has been synced to Github.

You can directly change the network transmit value within the Configuration Server of a Provisioner. Provisioner doesn't support relay currently. And for the friend feature, please enable the friend functionality in the menuconfig.

Thanks.

maheshvicky, 2022-05-16:
Hi @Campou , does the provisioner support relay function now??