espressif/esp-idf #4045: BLE secure connection issue

chegewara, 2019-09-10:
Hi @Weijian-Espressif 
i believe its my android smartphone issue, but i would like to confirm it. 
My esp32 works as BLE peripheral and i want to create secure connection with MITM, here is log i am getting:
```
W (12216) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (12256) BLEDevice: key type = ESP_LE_KEY_LENC
I (12256) BLEDevice: key type = ESP_LE_KEY_PENC
I (12256) BLEDevice: key type = ESP_LE_KEY_LID
I (12256) BLEDevice: key type = ESP_LE_KEY_LCSRK
E (12286) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (12286) BT_SMP: smp_derive_link_key_from_long_term_key failed

E (12296) BT_BTM: btm_proc_smp_cback received for unknown device
I (12306) BLEDevice: key type = ESP_LE_KEY_PID
I (12306) BLEDevice: key type = ESP_LE_KEY_PCSRK
I (12306) BLEDevice: ESP_GAP_BLE_AUTH_CMPL_EVT
W (15296) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

E (15306) BT_BTM: Device not found
```
Thanks for help.

Alvin1Zhang, 2019-09-11:
@chegewara Thanks for reporting, we will look into this. Thanks.

xiewenxiang, 2020-03-18:
Please confirm whether the bonding information of the mobile phone and the ESP32 is clear before doing this test?

You can use the API `esp_ble_get_bond_device_list()` and `esp_ble_remove_bond_device()` to delete the bonding information in ESP32.

If you can provide the conditions and steps to reproduce this issue,  it will helpful to analyze the problem

chegewara, 2020-03-18:
It was half year ago.
Smartphone samsung S9+ with android 9.
Yes, i can confirm that for 99% i did clear bonding data on smartphone and erase_flash on esp32 before that.

Alvin1Zhang, 2020-04-26:
@chegewara Thanks for sharing the updates, feel free to reopen if the issue gets reproduced. Thanks.

closed 2020-04-26 by Alvin1Zhang

caleedubya, 2020-07-04:
Hey guys, I'm trying to derive a key and I was wondering how you got this going?

I (12256) BLEDevice: key type = ESP_LE_KEY_LCSRK
E (12286) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (12286) BT_SMP: smp_derive_link_key_from_long_term_key failed

I know it says failed, but even to call the command to generate the derive_link key would be helpful. Thanks!

Jasonbroker, 2020-10-07:
[image]
wired, this happens.

TVAVAE, 2020-11-29:
> [image]
> wired, this happens.

Same problem ... But hangs on my side for a while ...

barish-malshikare, 2021-02-13:
Hello, We have observed the same problem while working with the Bluetooth provisioning.
We can successfully pair and communicate with the custom iOS app which is working on the iPhone 5S ( ios version 12.5.1) but facing the issue with the same application working on the iPhone 6S (ios version 12.5).
Also, the android application is available but in the android app also the same issue is observed. 

If we compile our code with esp.idf version 3.2 then the code works with all the mobile variants listed above!
But failed to work if compiled with esp.idf version v3.3.3, v4.2, and the latest version available.

The log is as follows:
******************************************************************************************************************
<DBG><../main/provisioning_handler.c><1689><Provisioning_Task>Free Heap Size After BLE Stop: 94268 bytes
<INFO><../main/provisioning_handler.c><1713><Provisioning_Task>Device is Not Provisioned, Starting Provisioning...
<DBG><../main/provisioning_handler.c><1762><Provisioning_Task>Free Heap Size Before BLE Start: 65156 bytes
<INFO><../main/provisioning_handler.c><278><show_bonded_devices>Bonded devices number : 0
<INFO><../main/provisioning_handler.c><1848><Provisioning_Task>BLE paired device list is Empty...
<INFO><../main/provisioning_handler.c><1855><Provisioning_Task>BLE Advertising started, Waiting for connection...
 E (45909) BT_SMP: Value for numeric comparison = 206803
<DBG><../main/provisioning_handler.c><807><gap_event_handler>ESP_GAP_BLE_NC_REQ_EVT, the passkey Notify number:206803
 W (51459) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
<DBG><../main/provisioning_handler.c><802><gap_event_handler>key type = ESP_LE_KEY_LENC
<DBG><../main/provisioning_handler.c><802><gap_event_handler>key type = ESP_LE_KEY_PENC
<DBG><../main/provisioning_handler.c><802><gap_event_handler>key type = ESP_LE_KEY_LID
 E (51729) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
 E (51739) BT_SMP: smp_derive_link_key_from_long_term_key failed

E (51749) BT_BTM: btm_proc_smp_cback received for unknown device
<DBG><../main/provisioning_handler.c><802><gap_event_handler>key type = ESP_LE_KEY_PID
<INFO><../main/provisioning_handler.c><814><gap_event_handler>BLE ADD: 44:4E:3A:74:FA:CA
<INFO><../main/provisioning_handler.c><815><gap_event_handler>Blutooth pair : Fail
<DBG><../main/provisioning_handler.c><949><sendBLEPacket>JSON Packet:	{"PAIR_STATUS": false }
 E (52809) BT_BTM: BTM_GetSecurityFlags false

 W (77469) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

 E (77469) BT_BTM: Device not found
 E (79839) BT_BTM: Device not found

<INFO><../main/provisioning_handler.c><814><gap_event_handler>BLE ADD: 44:4E:3A:74:FA:CA
<INFO><../main/provisioning_handler.c><815><gap_event_handler>Blutooth pair : Fail
 E (159799) BT_SMP: Value for numeric comparison = 306985
<DBG><../main/provisioning_handler.c><807><gap_event_handler>ESP_GAP_BLE_NC_REQ_EVT, the passkey Notify number:306985
 W (163629) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
<DBG><../main/provisioning_handler.c><802><gap_event_handler>key type = ESP_LE_KEY_LENC
<DBG><../main/provisioning_handler.c><802><gap_event_handler>key type = ESP_LE_KEY_PENC
<DBG><../main/provisioning_handler.c><802><gap_event_handler>key type = ESP_LE_KEY_LID
 E (163869) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
 E (163869) BT_SMP: smp_derive_link_key_from_long_term_key failed
 E (163879) BT_BTM: btm_proc_smp_cback received for unknown device

******************************************************************************************************************
Note: 
1) We have cleared the bonded device list from the mobile devices before performing the tests and also there is bonded device list in the esp device.