espressif/esp-idf #4046: Interrupt wakeup from sleep via GPIO

lachlanleslie1995, 2019-09-11:
IDF Ver: v4.0-dev-1476-g1b8ed9abf
build system: CMake
Compiler Ver: 8.2.0
OS: Windows 7

Scenario:
Attempting to use an accelerometer to wakeup esp32 from lightsleep with enabled external wakeup. Interrupt acts as normal when not in light sleep and only occurs once.

Expected behavior:
Wakes up and continues program from task

Actual behavior:
Interrupt doesn't get cleared and repeatedly gets called after waking from sleep (This does not occur if the esp32 is not sleeping, only occurs when waking from sleep)

Please see attached files below for more context:
Coredump:
[libespcoredump.zip](https://github.com/espressif/esp-idf/files/3598275/libespcoredump.zip) 
.elf:
[SourceCode.zip](https://github.com/espressif/esp-idf/files/3598279/SourceCode.zip)
sdkconfig:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3598285/sdkconfig.zip)

Alvin1Zhang, 2019-09-11:
@lachlanleslie1995 Thanks for reporting the issue, we will look into it. Thanks.

lachlanleslie1995, 2019-09-11:
Thanks @Alvin1Zhang ,

Let me know if you find the cause of the issue.

Cheers

lachlanleslie1995, 2019-09-11:
Hey @Alvin1Zhang ,

I found the cause of the issues. When you set the gpio_wakeup_enable it changes the interrupt type to high/low. And thus when I woke from sleep caused the interrupt to trigger repeatedly, maybe this should be highlighted in the documentation for people wanting to wakeup with an interrupt but also have an ISR attached to it?

Alvin1Zhang, 2019-09-11:
@lachlanleslie1995 Thanks for the updates. We will check your advice. Thanks.