espressif/esp-idf #4047: bluetooth master-slave integration

xielizyh, 2019-09-11:
Hello, I'm usin BLE as master and slave simultaneouly based on examples "gatt_client" and "gatt_server". When a device is connected, a connect  event will be generated to the “gattc_profile_event_handler” and “gatts_profile_event_handler” at the same time. My problem is how to distinguish whether the connected remote device is a master or a slave.

gengyuchao, 2019-09-11:
The master connection substate can only be entered from the initiating state. A device that becomes a
master must initiate the connection to the peer device. 
The slave connection substate can only be entered from the advertising state. A device that becomes a slave must have been advertising to the peer device.

WCCWCC, 2020-06-01:
Hi @xielizyh ,
 * The GATT protocol does not distinguish between master and slave, and directly operates through the method of Server and Client.

 * Normally, the device that sends connectable broadcast packets acts as a slave, and the device that initiates the connection actively acts as a master.

closed 2021-09-07 by Alvin1Zhang