espressif/esp-idf #4048: UART READ BYTES ERROR

Gohith, 2019-09-11:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    // v4.1-dev-58-g02c7c3885

- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.2.0
- Operating System:     [
LINUX]
- Power Supply:         [USB]

## Problem Description

//I had used UART_NUM_2 event type on esp32 before for reading data of less than 250 bytes and works fine.
But there is a problem in getting exact data of more than 330 bytes and the board started rebooting again and again after reading data.

### Expected Behavior
All the data should be copied and the board should not reboot.
### Actual Behavior
Problem in getting exact data of more than 330 bytes and the board started rebooting again and again after reading data.
### Steps to reproduce

1. build
2.flash

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#definr BUF_SIZE 1024
void uart_init(void)
{
	uart_config_t uart_config = {
        	.baud_rate = 9600,
        	.data_bits = UART_DATA_8_BITS,
        	.parity = UART_PARITY_DISABLE,
        	.stop_bits = UART_STOP_BITS_1,
        	.flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
			
    	};
	uart_param_config(UART_NUM_2, &uart_config);
	uart_set_pin(UART_NUM_2, 17, 16, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);  
	uart_driver_install(UART_NUM_2, BUF_SIZE * 3, BUF_SIZE * 3, 3072, &uart2_queue, 0);
}
static void uart_2_event_task(void *pvParameters)
{
	if(DEBUG) printf("UART EVENT +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++\n");
	uart_event_t event;
	uint8_t UART_data[2] = {0, 0};
	for(;;)
	{
		uart_flush(UART_NUM_2);
		if(xQueueReceive(uart2_queue, (void * )&event, (portTickType)portMAX_DELAY))
		{
			uint8_t data = 0;
			switch(event.type)
			{	
				case UART_DATA:	
						uart_read_bytes(UART_NUM_2, &UART_event_data, 334, portMAX_DELAY);
						if(DEBUG) printf("Received data from is ........... \n %s\n", UART_event_data);
						break;
	
				case UART_FIFO_OVF:
                	   			ESP_LOGI(TAG, "H/W fifo overflow");
                	    			uart_flush_input(UART_NUM_2);
                	   			xQueueReset(uart2_queue);
                	    			break;
	
				case UART_BUFFER_FULL:
                	   			ESP_LOGI(TAG, "ring buffer full");
                	    			uart_flush_input(UART_NUM_2);
                	   			xQueueReset(uart2_queue);
                	   			break;
				case UART_BREAK:
                	    			ESP_LOGI(TAG, "uart rx break");
                	   			break;
				
				case UART_PARITY_ERR:
                	    			ESP_LOGI(TAG, "uart parity error");
                	    			break;

                		case UART_FRAME_ERR:
                	    			ESP_LOGI(TAG, "uart frame error");
                	    			break;			

				default :
						ESP_LOGI(TAG, "Event raised: %d", event.type);
						break;
			}
			esp_event_loop_delete(uart2_queue);
		}	
	}
}

void app_main()
{
uart_init();
xTaskCreate(uart_2_event_task, "uart_2_event_task", 2048, NULL, 6, NULL);
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Received data from QR Code is ........... 
 {
“user_id”:“2",
“entity_id”:“1",
“dev_id”:“1",
“key”:“eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJ1c2VyX2lkIjoyLCJkYXRldGltZSI6IjIwMTktMDktMDUgMDk6NDI6NDcuNjgwMzE3IiwiZGV2X2lkIjoxLCJlbnRpdHlfaWQiOjF9.xGGpWhBESsqpxnPFflyZ7eQ6pPpXXotadaAAsPVc1Cg”,
“app_id”:“j5l1S4",
“app_time”:“05-09-2019-03-12-47"
}
/home/gohith/WorkForce/esp/esp-idf/components/freertos/queue.c:621 (xQueueTakeMutexRecursive)- assert failed!
abort() was called at PC 0x4008f06f on core 0
0x4008f06f: xQueueTakeMutexRecursive at /home/gohith/WorkForce/esp/esp-idf/components/freertos/queue.c:621 (discriminator 1)

ELF file SHA256: 5d990ea5107040cff6d53d78240913caf67dfb3296d50a590ca5941262b0d91a

Backtrace: 0x4008bf39:0x3ffd00b0 0x4008c2b1:0x3ffd00d0 0x4008f06f:0x3ffd00f0 0x4013d7d5:0x3ffd0110 0x400d5925:0x3ffd0140 0x4008e4b9:0x3ffd0170
0x4008bf39: invoke_abort at /home/gohith/WorkForce/esp/esp-idf/components/esp32/panic.c:157

0x4008c2b1: abort at /home/gohith/WorkForce/esp/esp-idf/components/esp32/panic.c:174

0x4008f06f: xQueueTakeMutexRecursive at /home/gohith/WorkForce/esp/esp-idf/components/freertos/queue.c:621 (discriminator 1)

0x4013d7d5: esp_event_loop_delete at /home/gohith/WorkForce/esp/esp-idf/components/esp_event/esp_event.c:604

0x400d5925: uart_2_event_task at /home/gohith/WorkForce/esp/Work_space/Work/build/../main/kea_beta.c:444

0x4008e4b9: vPortTaskWrapper at /home/gohith/WorkForce/esp/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

```

## Other items if possible

Alvin1Zhang, 2019-09-11:
@Gohith Thanks for reporting this, we will look into it. Thanks.

negativekelvin, 2019-09-11:
`#definr BUF_SIZE 1024` typo
`UART_event_data` undefined
`uart_flush(UART_NUM_2);` why?
`esp_event_loop_delete(uart2_queue);` what?

Gohith, 2019-09-11:
Sry uart_event_data was mentioned above, which I forgot to paste (part of
whole code).

And I used uart_flush(UART_NUM_2) for clearing buffer manually and I tried
without that api.

Thanks

negativekelvin, 2019-09-12:
What if you remove  `uart_flush(UART_NUM_2);` and `esp_event_loop_delete(uart2_queue);` and increase the stack size?

Gohith, 2019-09-12:
Getting GURU meditation error.

Gohith, 2019-09-12:
I got all data from uart and guru meditation error.

negativekelvin, 2019-09-12:
Show the log

Gohith, 2019-09-12:
Received data from QR Code is ........... 
 {
“user_id”:“2",
“entity_id”:“1",
“dev_id”:“1",
“key”:“eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJ1c2VyX2lkIjoyLCJkYXRldGltZSI6IjIwMTktMDktMDUgMDk6NDI6NDcuNjgwMzE3IiwiZGV2X2lkIjoxLCJlbnRpdHlfaWQiOjF9.xGGpWhBESsqpxnPFflyZ7eQ6pPpXXotadaAAsPVc1Cg”,
“app_id”:“j5l1S4",
“app_time”:“05-09-2019-03-12-47"
}
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400014fd  PS      : 0x00060730  A0      : 0x8015be48  A1      : 0x3ffd0130  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd00f0  
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x00000004  A13     : 0x00000000  
A14     : 0x3ffbd618  A15     : 0x39313032  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 3e7d719557f23b1c3059c6a736aeb12f44c5920e887e7a3d39048ea5ca2b1b9c

Backtrace: 0x400014fa:0x3ffd0130 |<-CORRUPTED

Rebooting...

negativekelvin, 2019-09-12:
Looks like memory corruption, you didn't show all the code so hard to say

koobest, 2019-10-18:
Did you delete `uart2_queue` in `esp_event_loop_delete()`?  This will be a problem when you call `xQueueReceive(uart2_queue,...)` again. This should be the cause for the first issue you reported.

koobest, 2019-10-28:
Hi, @Gohith
Have you solved this issue?

thanks

Alvin1Zhang, 2020-01-15:
@Gohith Thanks for reporting. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

Gohith, 2020-01-15:
No.
I couldn’t solve issue but I took another logic instead of it.

macdroid53, 2020-01-15:
Could it be similar to:
How to read UART bytes in queue after break? (IDFGH-2424) #4537

danpcurran, 2020-02-04:
I have a similar issue with the UART Rx loosing bytes of data over the when its sent consecutively by a host device with little or no delay between chars. 

Unexpected behaviour
**************************  
the default uart rx isr looses data when the receiving consecutive data over the UART_FULL_THRESH_DEFAULT (120 bytes)

IDF version used: v4.0-beta2-174-g99fb9a3f7-dirty
code is edited, compiled  and debugged using visualGDB version 5.5 ( preview 3, build 3458)
the dev board is a Esp WROVER V4.1

The UART code is based on 

https://github.com/espressif/esp-idf/blob/600d542f53ab6540d277fca4716824d168244c8c/examples/peripherals/uart/uart_events/main/uart_events_example_main.c

The uart_event_task() has a high priority.

I tried changing the changing the FIFO full threshold in uart.c  but this did not cure the problem, it still loses data on the FIFO full threshold. Tried setting it to 96 , 64 and it just moved the issue. 

when sending 130 bytes, the uart event handler task would get  0x78, 0  this ties in with the UART_FULL_THRESH_DEFAULT (120) but its missing characters.

Sending more data eg 150 was 0x78, 1A when it should have been 0x78,0x1E.

I modified the test file to 632 bytes and I see 0x78,0x78,0x78,0x78,0x78,0x18 when it should have ended 20

Looking at the buffered data the first 120 chars appears ok it, then looses a few in sequence after the first 120 char boundary, then is okay for the rest of the data to the next 120 byte boundary.
 A quick inspection of the data at the 120 boundaries sometimes is ok, sometimes is out of sequence 

So initially it looks like the ISR always struggles to populate the message que and receive characters in the driver for the first queued data and then randomly looses it for others. 

looked in the esp32 forum first for help but no significant contributions yet.
https://www.esp32.com/viewtopic.php?f=12&t=14078

koobest, 2020-03-17:
hi, @danpcurran
when the  rx ringbuffer size is small and the UART  working at high baud rate. If the task does not read the data in time, the buffer will be full soon. In this case the data will be lost

Alvin1Zhang, 2020-05-28:
@Gohith Thanks for reporting/updates and @danpcurran for sharing updates. Feel free to reopen if the issue still happens.

closed 2020-05-28 by Alvin1Zhang

beginner140BB, 2023-01-06:
@negativekelvin @Alvin1Zhang @koobest .

Does anyone knows what is actually causing 
 UART_FIFO_OVF,          /*!< UART FIFO overflow event*/

is there any way to solve  this:

```
xSemaphoreTake( xMutex, portMAX_DELAY );// Reading serial commands from palmtec
uart0_rx(Serial_data1);
xSemaphoreGive( xMutex )
```

- where uart reading is happening

- the function uart0_rx is given below

```
int uart0_rx(char *data)
{
	int length = 0;
	ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_0,(size_t*)&length));
	length=uart_read_bytes(UART_NUM_0,data,1024, 20 / portTICK_RATE_MS);
	uart_flush_input(UART_NUM_0);
	return length;
}
```

```
void uart_init(int BaudRate)
{
    const uart_config_t uart_config = {
    		.baud_rate = BaudRate,
    		.data_bits = UART_DATA_8_BITS,
    		.parity = UART_PARITY_DISABLE,
    		.stop_bits = UART_STOP_BITS_1,
    		.flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
			.source_clk = UART_SCLK_APB,
    };
    uart_driver_install(UART_NUM_0, 2048, 1024, 20, &uart0_queue, 0);
    uart_param_config(UART_NUM_0, &uart_config);
    uart_set_pin(UART_NUM_0, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
}
```

- where uart init takes place

does you guys have any suggestion on this as i am getting fifo overflow sometimes only? 

**the data is actually ~250 bytes with baudrate of 115200**