espressif/esp-idf #4049: about modbus master bug

1163167506, 2019-09-11:
When the device calls sense_modbus_read_value() to send the Modbus instruction,

If you read / set flash using NVS flash API nvs_get_str()/nvs_set_str(), the following error will be reported
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x40139964  PS      : 0x00060034  A0      : 0x400820e8  A1      : 0x3ffb0670  
A2      : 0x3ff5f000  A3      : 0xb0000000  A4      : 0x3ff42000  A5      : 0x4008e638  
A6      : 0x3ff42000  A7      : 0x700000bb  A8      : 0x800860fc  A9      : 0x00000001  
A10     : 0x40139964  A11     : 0xb0000000  A12     : 0x00060023  A13     : 0x3ffb0650  
A14     : 0x3ffb0c94  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
Core 0 was running in ISR context:
EPC1    : 0x40062230  EPC2    : 0x00000000  EPC3    : 0x400830c5  EPC4    : 0x40139964

Alvin1Zhang, 2019-09-16:
@1163167506 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

alisitsyn, 2019-10-01:
Hi 1163167506,
The issue has been reproduced and fixed. It will be merged in ESP-IDF rev v4.1. 
The Modbus master example for freemodbus component uses group timer to measure Modbus related timeouts. The current port code uses timer IRQ located in Flash memory (non-IRAM-safe interrupt), but during interrupt initialization the ESP_INTR_FLAG_IRAM was used that led example to crash when the task calls NVS read API which disable CPU cash. The simple solution is to change flag in interrupt initialization but it is better to provide flexible way to place related timer handlers into IRAM using Kconfig option what has been done during the fix.

alisitsyn, 2019-10-01:
Could you apply the change below to apply simple fix while it is not merged?
file: components/freemodbus/port/porttimer_m.c:
    xErr = timer_isr_register(usTimerGroupIndex, usTimerIndex,
line112:      vTimerGroupIsr, (void*)(uint32_t)usTimerIndex, ESP_INTR_FLAG_IRAM, NULL);
Change it to: 
line112:      vTimerGroupIsr, (void*)(uint32_t)usTimerIndex, ESP_INTR_FLAG_LEVEL3, NULL);
The same for components/freemodbus/port/porttimer.c:
Line 112: xErr = timer_isr_register(usTimerGroupIndex, usTimerIndex, vTimerGroupIsr, (void*)(uint32_t)usTimerIndex, ESP_INTR_FLAG_LEVEL3, NULL);

This will allow to avoid the "Guru Meditation Error: Core 0 panic'ed (Cache disabled but cached memory region accessed)".

Alvin1Zhang, 2019-10-08:
@1163167506 Thanks for reporting the issue, would you please help share if any updates for the issue? Thanks.

ClementMarrast, 2019-10-23:
Hello @alisitsyn and @Alvin1Zhang ,

I'm facing the same issue as @1163167506 , I've tried the modification you've suggested:
porttimer_m.c, line 116:  
``` 
// Register ISR for timer
xErr = timer_isr_register(usTimerGroupIndex, usTimerIndex,
vTimerGroupIsr, NULL, ESP_INTR_FLAG_LEVEL3, NULL); 
```  
`porttimer.c`, line 116:  
```
// Register ISR for timer
xErr = timer_isr_register(usTimerGroupIndex, usTimerIndex, vTimerGroupIsr, NULL, ESP_INTR_FLAG_LEVEL3, NULL);
```  

The `timer_isr_register` function is returning an error code:
```E (5) MB_PORT_COMMON: xMBMasterPortTimersInit(121): timer set value failure, timer_isr_register() returned (0x105).
E (14) MB_CONTROLLER_MASTER: mbc_serial_master_start(109): mb stack initialization failure, eMBInit() returns (0x3).
E (25) MB_CONTROLLER_MASTER: mbc_master_start(211): SERIAL master start failure error=(0x103).
E (34) HMI_MB: hmi_modbus_init(125): mb controller start fail, returns(0x103).```

ClementMarrast, 2019-10-23:
I answer to myself:
There is no more interrupt available for this level. I've experienced the same thing with functions components/freemodbus/port/portserial.c:xMBPortSerialInit() line 249:
```
 xErr = uart_driver_install(ucUartNumber, MB_SERIAL_BUF_SIZE, MB_SERIAL_BUF_SIZE,
            MB_QUEUE_LENGTH, &xMbUartQueue, ESP_INTR_FLAG_LEVEL3);
```
and  
components/freemodbus/port/portserial_m.c:xMBMasterPortSerialInit() line 244:
```
 xErr = uart_driver_install(ucUartNumber, MB_SERIAL_BUF_SIZE, MB_SERIAL_BUF_SIZE,
            MB_QUEUE_LENGTH, &xMbUartQueue, ESP_INTR_FLAG_LEVEL3);
```

For theses 4 cases, instead of using `ESP_INTR_FLAG_LEVEL3`, I've used `ESP_INTR_FLAG_LOWMED` which allow to use level 1, 2 or 3.
Do you see any problems doing that way?

Thanks for your feedback

alisitsyn, 2019-10-24:
Hi @ClementMarrast, 

This was my mistake and your change is good and will work good. There is one disadvantage of this is that when your code uses API to access flash intensively the interrupt handling will be delayed that may cause Modbus errors. I already implemented the feature to place timer interrupt and serial interrupt handler into IRAM with Kconfig option. It should be merged up to IDF v4.1.

Thank you.

Alvin1Zhang, 2020-05-28:
@1163167506 @ClementMarrast Thanks for reporting and sorry for slow turnaround. The fix has been available at https://github.com/espressif/esp-idf/commit/65c67faa6a84f63dc740111adaabea46ffc3bf89, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-28 by Alvin1Zhang