espressif/esp-idf #4050: SPIFFS: mount failed in restful_server example

derek-whisper, 2019-09-11:
## Environment

- Development Kit:      ESP32 Wrover Dev Board from Analog Lamb
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.1-dev-141-ga7e8d87d3
- Build System:  Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

### Expected Behavior
It is expected the the program works as described in the example [readme](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/restful_server)

### Actual Behavior
After the ESP32 connects to the WiFi network I get the following error
  ```
E (4526) SPIFFS: mount failed, -10025
E (4526) example: Failed to mount or format filesystem
```
### Steps to reproduce

1. Run `make menuconfig`
2. Set `Website deploy mode` to `Deploy website to SPI Nor Flash`
3. Setup `Example Connection Configuration` to use WiFi network
4. Run `make erase_flash flash monitor -j4`
5. After the ESP32 connects to the WiFi network I get the following error
  ```
E (4526) SPIFFS: mount failed, -10025
E (4526) example: Failed to mount or format filesystem
```

### Code to reproduce this issue
[Example code has not been modified](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/restful_server)

## Debug Logs
```
Toolchain path: /Users/derekdenman/toolchains//esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
Expected to see version: esp32-2019r1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /Users/derekdenman/toolchains/esp/esp-mdf/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6608
ho 0 tail 12 room 4
load:0x40078000,len:12008
ho 0 tail 12 room 4
load:0x40080400,len:6672
entry 0x40080770
I (32) boot: ESP-IDF v4.1-dev-141-ga7e8d87d3-dirty 2nd stage bootloader
I (32) boot: compile time 16:45:00
I (33) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
[... 798 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options
```

fernandosalomao, 2019-09-16:
I've changed this line to **.format_if_mount_failed = true**

[esp_rest_main.c](https://github.com/espressif/esp-idf/blob/9f145ff165099f32b92b7ee022a1debc5f0a583d/examples/protocols/http_server/restful_server/main/esp_rest_main.c#L97)

However, it cannot read the files:

W (4117) SPIFFS: mount failed, -10025. formatting...
E (9107) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (9107) task_wdt:  - IDLE0 (CPU 0)
E (9107) task_wdt: Tasks currently running:
E (9107) task_wdt: CPU 0: main
E (9107) task_wdt: CPU 1: IDLE1
E (14127) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (14127) task_wdt:  - IDLE0 (CPU 0)
E (14127) task_wdt: Tasks currently running:
E (14127) task_wdt: CPU 0: main
E (14127) task_wdt: CPU 1: IDLE1
I (17927) example: Partition size: total: 1920401, used: 0
I (17927) esp-rest: Starting HTTP Server
E (47787) esp-rest: Failed to open file : /www/index.html
W (47787) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Failed to read existing file
W (47797) httpd_uri: httpd_uri: uri handler execution failed

igrr, 2019-10-07:
@derek-whisper @fernandosalomao Thanks for reporting this. The issue is that SPIFFS filesystem image wasn't flashed to the board when running `make flash`. The fix is in review and will be merged soon. For now, you can apply the following change locally:

```patch
From 75f2948e02a498711b7080b094795ae2956ce866 Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Mon, 7 Oct 2019 13:28:38 +0200
Subject: [PATCH] examples/restful_server: fix Makefile syntax for flashing
 SPIFFS

Closes https://github.com/espressif/esp-idf/issues/4050
---
 examples/protocols/http_server/restful_server/Makefile | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)

diff --git a/examples/protocols/http_server/restful_server/Makefile b/examples/protocols/http_server/restful_server/Makefile
index c07edffd2..ba534aed1 100644
--- a/examples/protocols/http_server/restful_server/Makefile
+++ b/examples/protocols/http_server/restful_server/Makefile
@@ -7,7 +7,8 @@ include $(IDF_PATH)/make/project.mk
 ifdef CONFIG_EXAMPLE_WEB_DEPLOY_SF
 WEB_SRC_DIR = $(shell pwd)/front/web-demo
 ifneq ($(wildcard $(WEB_SRC_DIR)/dist/.*),)
-$(eval $(call spiffs_create_partition_image,www,$(WEB_SRC_DIR)/dist,FLASH_IN_PROJECT))
+SPIFFS_IMAGE_FLASH_IN_PROJECT := 1
+$(eval $(call spiffs_create_partition_image,www,$(WEB_SRC_DIR)/dist))
 else
 $(error $(WEB_SRC_DIR)/dist doesn't exist. Please run 'npm run build' in $(WEB_SRC_DIR))
 endif
-- 
2.21.0 (Apple Git-122)

```

Check that when running `make flash`, you see that something is getting flashed to address `0x00110000`, as in the example below:
```
Compressed 26080 bytes to 15615...
Wrote 26080 bytes (15615 compressed) at 0x00001000 in 0.2 seconds (effective 1085.9 kbit/s)...
Hash of data verified.
Compressed 733600 bytes to 459806...
Wrote 733600 bytes (459806 compressed) at 0x00010000 in 7.2 seconds (effective 815.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 116...
Wrote 3072 bytes (116 compressed) at 0x00008000 in 0.0 seconds (effective 1530.8 kbit/s)...
Hash of data verified.
Compressed 2097152 bytes to 529932...
Wrote 2097152 bytes (529932 compressed) at 0x00110000 in 15.3 seconds (effective 1097.0 kbit/s)...
```

closed 2019-10-13 by igrr