espressif/esp-idf #4052: failed to run idf.py menuconfig on windows10.

phypy, 2019-09-11:
Hi,I run the python idf.py menuconfig as a super manager on windows powershell but get errors llike this:
idf-tools version:2.0

PS D:\Programs\Espressif\hello_world> python D:\Programs\Espressif\esp-idf\tools\idf.py menuconfig
Checking Python dependencies...
Python requirements from D:\Programs\Espressif\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory d:\programs\espressif\hello_world\build
Executing "cmake -G 'MinGW Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 d:\programs\espressif\hello_world"...
Warn about uninitialized values.
-- Could NOT find Git (missing: GIT_EXECUTABLE)
CMake Warning at D:/Programs/Espressif/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  D:/Programs/Espressif/esp-idf/tools/cmake/idf.cmake:34 (include)
  D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "D:/Programs/Espressif/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "D:/Programs/Espressif/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
PS D:\Programs\Espressif\hello_world> python D:\Programs\Espressif\esp-idf\tools\idf.py menuconfig
Checking Python dependencies...
Python requirements from D:\Programs\Espressif\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory d:\programs\espressif\hello_world\build
Executing "cmake -G 'MinGW Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 d:\programs\espressif\hello_world"...
Warn about uninitialized values.
-- Could NOT find Git (missing: GIT_EXECUTABLE)
CMake Warning at D:/Programs/Espressif/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  D:/Programs/Espressif/esp-idf/tools/cmake/idf.cmake:34 (include)
  D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "D:/Programs/Espressif/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "D:/Programs/Espressif/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

dobairoland, 2019-09-11:
@phypy You need to run `install.sh` and `source export.sh` before you could use `idf.py menuconfig`.

phypy, 2019-09-11:
> @phypy You need to run `install.sh` and `source export.sh` before you could use `idf.py menuconfig`.

Hi,Do you means install.bat and export.bat ? I get the error like this:
D:\Programs\Espressif\hello_world>python D:\Programs\Espressif\esp-idf\tools\idf.py menuconfig
Checking Python dependencies...
Python requirements from D:\Programs\Espressif\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory d:\programs\espressif\hello_world\build
Executing "cmake -G 'MinGW Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 d:\programs\espressif\hello_world"...
Warn about uninitialized values.
-- Could NOT find Git (missing: GIT_EXECUTABLE)
CMake Warning at D:/Programs/Espressif/esp-idf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  D:/Programs/Espressif/esp-idf/tools/cmake/idf.cmake:34 (include)
  D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at D:/Programs/Espressif/esp-idf/tools/cmake/project.cmake:258 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "D:/Programs/Espressif/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "D:/Programs/Espressif/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

dobairoland, 2019-09-11:
The errors you are getting tells you that those executable are not accessible. 

If you run `get-command xtensa-esp32-elf-gcc` you should get something similar:
```
CommandType     Name                                               Version    Source
-----------     ----                                               -------    ------
Application     xtensa-esp32-elf-gcc.exe                           0.0.0.0    C:\Program Files\Espressif\ESP-IDF Tools\tools\bin\xtensa-esp32-elf-gcc.exe
```
The installer you've run here
https://github.com/espressif/esp-idf/issues/4043#issuecomment-530251164 should have installed it. The path to the tools need to be in PATH. Please check it. For example, I have there  `C:\Program Files\Espressif\ESP-IDF Tools\tools\bin` so if I run `xtensa-esp32-elf-gcc.exe` then the shell will know where to find it. 

You can add the path manually or the `export.bat` should do it for you.

phypy, 2019-09-12:
I cannot find a bin dir at path  ESP-IDF Tools\tool ...

dobairoland, 2019-09-12:
@phypy This is from your log (from https://github.com/espressif/esp-idf/issues/4043#issuecomment-530251164): 
```
-- Found assembler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe -- works
```

So `D:/Programs/Espressif/ESPIDF_Tools/tools/bin/xtensa-esp32-elf-gcc.exe` is not present in your computer? `D:/Programs/Espressif/ESPIDF_Tools/tools/bin` needs to be in your PATH.

Maybe you should do a clean reinstall of the tools since you have a couple of failures before.

phypy, 2019-09-15:
Hi，I fix this error by advance the PATH position to front of cmake.but get another problem:"'rm' 不是内部或外部命令，也不是可运行的程序或批处理文件。" Why it exist so much error???
this is the error msg:
D:\Programs\Espressif\hello_world>python D:\Programs\Espressif\esp-idf\tools\idf.py menuconfig
Checking Python dependencies...
Python requirements from D:\Programs\Espressif\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running mingw32-make in directory d:\programs\espressif\hello_world\build
Executing "mingw32-make -j 10 menuconfig"...
[  0%] Performing build step for 'mconf-idf'
'rm' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
CMakeFiles\mconf-idf.dir\build.make:124: recipe for target 'mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build' failed
mingw32-make[3]: *** [mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build] Error 1
CMakeFiles\Makefile2:2178: recipe for target 'CMakeFiles/mconf-idf.dir/all' failed
mingw32-make[2]: *** [CMakeFiles/mconf-idf.dir/all] Error 2
CMakeFiles\Makefile2:2249: recipe for target 'CMakeFiles/menuconfig.dir/rule' failed
mingw32-make[1]: *** [CMakeFiles/menuconfig.dir/rule] Error 2
Makefile:1047: recipe for target 'menuconfig' failed
mingw32-make: *** [menuconfig] Error 2
mingw32-make failed with exit code 2

dobairoland, 2019-09-16:
@phypy This will be again another problem with your PATH.

It should not do this `[ 0%] Performing build step for 'mconf-idf'` but use `mconf-idf.exe` from the directory of installed tools.

`get-command mconf-idf` should point to `<install_dir>\mconf-idf\mconf-idf.exe`. Is `<install_dir>\mconf-idf` in your path?

dobairoland, 2019-09-25:
Hi @phypy. Do you have any update regarding your issue?

dobairoland, 2019-10-03:
I'm closing this issue for lack of activity.

closed 2019-10-03 by dobairoland