espressif/esp-idf #4061: Switch between Wi-Fi AP and STA

jkockx, 2019-09-13:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.0-dev-1191-g138c941fa
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I'm trying to get info about the ssid and password of my wi-fi network over a wi-fi link. For this, I'm configuring my ESP as an AP first and running a HTTP server on it.  This way I get my ssid and psw through a POST message. 
But when I'm configuring as sta after that, I get the errors as listed below. The functions used for these configurations are:
void wifi_init_sta() {
	ESP_LOGI(TAGAPP, "%s", __func__);
	if (strlen(pwd_ap_internet) && strlen(ssid_ap_internet) != 0)
	{
		ESP_LOGI(TAGAPP, "init STA");

		tcpip_adapter_init();

		wifi_event_group = xEventGroupCreate();
		wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
		ESP_ERROR_CHECK(esp_wifi_init(&cfg));
		ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

		wifi_config_t wifi_config = {};
		strcpy((char *)wifi_config.sta.ssid, ssid_ap_internet);
		strcpy((char *)wifi_config.sta.password, pwd_ap_internet);

		ESP_LOGI(TAG, "WiFi %s ", wifi_config.sta.ssid);
		ESP_LOGI(TAG, "PSW %s ", wifi_config.sta.password);
		ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
		ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
		ESP_LOGI(TAGAPP, "esp_wifi_start: as sta");
		ESP_ERROR_CHECK(esp_wifi_start());

		//wait a portMAX_DELAY for CONNECTED_BIT to become set
		xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);
	}
}
`
void wifi_init_softap()
{
	ESP_LOGI(TAG, "%s", __func__);
	if (strlen(ssid_ap_internet) != 0) {
		ESP_LOGI(TAGAPP, "ssid valid: initialize sta");
		wifi_init_sta();
		return;
	}
	ESP_LOGI(TAGAPP, "ssid invalid: get ssid and pwd with http");
	// configuration as access point + web server

	tcpip_adapter_init();

	s_wifi_event_group = xEventGroupCreate();

	if (!event_loop_inited) {
		ESP_ERROR_CHECK(esp_event_loop_create_default());
		ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL)); //-> just here to get rid of startup errors
		event_loop_inited = 1;
	}

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();       //= WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));

	ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL));

	wifi_config_t wifi_config = {
		.ap = {
			.ssid = WIFI_SSID,
			.ssid_len = strlen(WIFI_SSID),
			.password = WIFI_PASS,
			.max_connection = 1,
			.authmode = WIFI_AUTH_WPA_WPA2_PSK},
	};

	if (strlen(WIFI_PASS) == 0) //EXAMPLE_ESP_WIFI_PASS
	{
		wifi_config.ap.authmode = WIFI_AUTH_OPEN;
	}

	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
	int err;
	err = esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config);

	if (!err) {
		ESP_LOGI(TAG, "wifi ap config:ssid:%s,pass:%s,authmode:%d", wifi_config.ap.ssid, wifi_config.ap.password, wifi_config.ap.authmode);
	} else {
		ESP_LOGI(TAG, "wifi ap config:failed");
	}

	uint8_t ap_ip[4];
	ap_ip[0]=192;
	ap_ip[1]=168;
	ap_ip[2]=1;
	ap_ip[3]=1;
	tcpip_adapter_ip_info_t ip_info;
	IP4_ADDR(&ip_info.ip, ap_ip[0], ap_ip[1], ap_ip[2], ap_ip[3]);
	IP4_ADDR(&ip_info.gw, ap_ip[0], ap_ip[1], ap_ip[2], ap_ip[3]);
	IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);

	uint8_t ap_mac[6];
	esp_wifi_get_mac(ESP_IF_WIFI_AP, ap_mac);
	// ESP_LOGI(TAG, "tcpip_adapter_ap_start:MAC=%s", print_mac(ap_mac));
	ESP_LOGI(TAGAPP, "tcpip_adapter_ap_start:IP=%d %d %d %d", ap_ip[0], ap_ip[1], ap_ip[2], ap_ip[3]);
	//initializes aut a dhcp server
	//function called aut when an AP is started
	tcpip_adapter_ap_start(ap_mac, &ip_info);

	ESP_ERROR_CHECK(esp_wifi_start());

    start_webserver();
}
`
## Debug Logs
`I (1334) APPLICATION: Starting server on port: '80'
I (1334) APPLICATION: wifi_event_handler: non of the registered events, event 12
I (1344) APPLICATION: wifi_event_handler
I (1354) APPLICATION: event_handler
I (1354) APPLICATION: Registering URI handlers
I (48124) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (48124) wifi: station: cc:20:e8:82:8e:29 join, AID=1, bgn, 20
I (48154) APPLICATION: station cc:20:e8:82:8e:29 join, AID=1
I (48154) APPLICATION: wifi_event_handler
I (48154) APPLICATION: event_handler
I (48484) tcpip_adapter: softAP assign IP to station,IP is: 192.168.1.2
I (48484) APPLICATION: event_handler
I (50014) wifi: ampdu: ignore deleting tx BA0
I (54754) ESP32_Server: pwd: DPTechnics
I (54754) ESP32_Server: ssid: makkelijk
I (54754) APPLICATION: stop wifi access point
I (54764) wifi: station: cc:20:e8:82:8e:29 leave, AID = 1, bss_flags is 134243, bss:0x3ffbac94
I (54764) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (54774) APPLICATION: wifi_event_handler: non of the registered events, event 3
I (54784) APPLICATION: wifi_event_handler
I (54784) APPLICATION: event_handler
I (54794) APPLICATION: station cc:20:e8:82:8e:29 leave, AID=1
I (54794) APPLICATION: wifi_event_handler
I (54804) APPLICATION: event_handler
I (54804) wifi: flush txq
I (54804) wifi: stop sw txq
I (54814) wifi: lmac stop hw txq
I (54814) APPLICATION: wifi_event_handler: non of the registered events, event 13
I (54814) APPLICATION: wifi_init_sta
I (54824) APPLICATION: wifi_event_handler
I (54824) APPLICATION: init STA
I (54834) APPLICATION: event_handler
W (54834) event: handler already registered, overwriting
I (54844) ESP32_Server: WiFi DPTechnics 
I (54854) ESP32_Server: PSW makkelijk 
I (54854) APPLICATION: esp_wifi_start: as sta
I (54874) wifi: mode : sta (24:0a:c4:04:55:ec)
I (54874) APPLICATION: WIFI_EVENT_STA_START
I (54874) APPLICATION: wifi_event_handler
assertion "netif already added" failed: file "/home/programmer/esp/esp-idf/components/lwip/lwip/src/core/netif.c", line 395, function: netif_add
abort() was called at PC 0x40101d4b on core 1
0x40101d4b: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: bef5245ef3d90279b2f58c060ac347802ddd78bf4a5d77dc4d273f7c137a6f53

Backtrace: 0x40091dcd:0x3ffd2b00 0x40092145:0x3ffd2b20 0x40101d4b:0x3ffd2b40 0x4013288b:0x3ffd2b70 0x40115d33:0x3ffd2ba0 0x40115dda:0x3ffd2bf0 0x40115bde:0x3ffd2c10 0x40128d29:0x3ffd2c30 0x40128dd0:0x3ffd2c50 0x400953bd:0x3ffd2c80
0x40091dcd: invoke_abort at /home/programmer/esp/esp-idf/components/esp32/panic.c:155

0x40092145: abort at /home/programmer/esp/esp-idf/components/esp32/panic.c:172

0x40101d4b: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x4013288b: netif_add at /home/programmer/esp/esp-idf/components/lwip/lwip/src/core/netif.c:395 (discriminator 1)

0x40115d33: tcpip_adapter_start at /home/programmer/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:196

0x40115dda: tcpip_adapter_start_api at /home/programmer/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:255

0x40115bde: tcpip_adapter_api_cb at /home/programmer/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:86

0x40128d29: tcpip_thread_handle_msg at /home/programmer/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:168

0x40128dd0: tcpip_thread at /home/programmer/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

0x400953bd: vPortTaskWrapper at /home/programmer/esp/esp-idf/components/freertos/port.c:143
`

espxiehang, 2019-11-04:
hi @jkockx ，Can you give me your project to reproduce this problem? If the project is not confidential.

freakyxue, 2019-11-08:
hi @jkockx  
 This should be the assertion that the sta or AP pattern is initialized twice.
 
At the beginning, the netif interface is added to the netif list. If it is reinitialized, the netif interface will be added to the list again. If the network interface has been saved in the list, an assertion will be generated.

Can you send in your project? Let me check where the mode is reinitialized?

afcec, 2019-11-21:
Hi, any news? I am experiencing the same problem. First connection in 'station' mode works fine, but then if I disconnect and try to start again in either 'station' or 'ap' mode I am getting this assertion error.

These are some fragments of my code (the event loop is created somewhere else):

**Wi-Fi Connection in Station Mode**

`wifi_event_group = xEventGroupCreate();
tcpip_adapter_init();
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK( esp_wifi_init(&cfg) );

ESP_ERROR_CHECK( esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL) );
ESP_ERROR_CHECK( esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_event_handler, NULL) );
ESP_ERROR_CHECK( esp_event_handler_register(SC_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL) );

esp_wifi_get_config(WIFI_IF_STA, &wifi_config);

ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_FLASH) );

esp_wifi_get_config(WIFI_IF_STA, &wifi_config);
ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
ESP_ERROR_CHECK( esp_wifi_start() );`

**Wi-Fi Connection in AP mode**

`wifi_event_group = xEventGroupCreate();
tcpip_adapter_init();
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK( esp_wifi_init(&cfg) );

ESP_ERROR_CHECK( esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL) );
ESP_ERROR_CHECK( esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_event_handler, NULL) );
ESP_ERROR_CHECK( esp_event_handler_register(SC_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL) );

wifi_config_t wifi_config = {
	.ap = {
		.ssid = MY_SSID,
		.ssid_len = strlen(MY_SSID),
		.password = MY_PASS,
		.max_connection = 1,
		.authmode = WIFI_AUTH_WPA_WPA2_PSK
	},
};
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
ESP_ERROR_CHECK(esp_wifi_start());`

**Wi-Fi disconnect**

`esp_wifi_disconnect();
esp_wifi_stop();
esp_wifi_deinit();`

altcuim, 2019-12-16:
Try update to esp_netif. and deinit STA when switch to AP:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/tcpip_adapter_migration.html#updating-network-connection-code

freakyxue, 2019-12-16:
@jkockx @afcec YES,before switch to ap or station , must call esp_wifi_stop.

freakyxue, 2020-01-06:
@jkockx Do you have any questions about this

Alvin1Zhang, 2020-01-20:
@jkockx Thanks for reporting and sharing the updates @tekaiot Will close the ticket due to short of feedback. Feel free to reopen if the issue still happens. Thanks.

closed 2020-01-20 by Alvin1Zhang

Zeni241, 2020-02-26:
Using stable version 4.0 of IDF, while switching from AP mode to Station mode, I was getting below error:
```
Connecting to wifi
I (38201) wifi: mode : sta (cc:50:e3:95:49:b8)
wifi Station started
assertion "netif already added" failed: file "C:/Users/faraz/esp-idf/components/lwip/lwip/src/core/netif.c", line 395, function: netif_add
abort() was called at PC 0x4012c45b on core 1 
```
I added these lines at the end of AP mode , (before switching to station mode.)

esp_wifi_disconnect();
esp_wifi_stop();
esp_wifi_deinit();
 vTaskDelay(1000 / portTICK_PERIOD_MS);

This solved the problem.

freakyxue, 2020-02-27:
yes  before change model.
must call  esp_wifi_stop();

ChrisEAlfred, 2021-03-22:
NO! The problem is related to the call to esp_wifi_deinit().

Should it be ok to do cycles of:
- esp_wifi_init
- esp_wifi_start
- do wifi stuff
- esp_wifi_stop
- esp_wifi_deinit

I have found that if you have the esp_wifi_init and esp_wifi_deinit in the loop, you get "netif already added".
If you do esp_wifi_init once at startup, and esp_wifi_deinit only if you reset, all is good.

Is this what you expect/intend?