espressif/esp-idf #4062: SPP Bluetooth Classic Security Problem

mdolgich, 2019-09-14:
Closely related to: #2935, no resolution though.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:   Adafruit Feather Huzzah32
- Kit version -
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v3.3 LTS
- Build System:         Make
- Compiler version 
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
 and
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
- Operating System:    Linux and macOS
- Power Supply:         [USB/Battery]

## Problem Description
Setting security mask as : 
- ESP_SPP_SEC_ENCRYPT
- ESP_SPP_SEC_AUTHENTICATE
- ESP_SPP_SEC_MODE4_LEVEL4

The Android Phone fails to reconnect.
Setting the security mask to:
- ESP_SPP_SEC_NONE
- ESP_SPP_SEC_MITM
the Android Phone can connect to the ESP without issue.

We are using the SPP API with the legacy pairing method, to in order establish a connection from an Android App to the ESP. The reconnection after pairing does not result in any SPP or GAP events and fails.

### Expected Behavior
The device gets connected to the ESP.

### Actual Behavior
The device does not get a connection to the ESP.

### Steps to repropduce

1. Go to the example_spp_acceptor_demo.c
2. Disable SSP in menuconfig.
3. Pair your device to the ESP.
4. Disconnect your device.
5. Try to reconnect your device.

### Code to reproduce this issue

See example_spp_acceptor_demo.c

## Other items if possible

- [X] [sdkconfig.txt](https://github.com/espressif/esp-idf/files/3612751/sdkconfig.txt)
- [X] [bt_spp_acceptor_demo.elf.txt](https://github.com/espressif/esp-idf/files/3612752/bt_spp_acceptor_demo.elf.txt)
- [ ] coredump (This provides stacks of tasks.) 
I have added a .txt ending to the file in order to satisfies the GitHub upload rules.

Alvin1Zhang, 2019-09-16:
@mdolgich Thanks for reporting the issue, we will look into. Thanks.

Alvin1Zhang, 2020-05-28:
@mdolgich Thanks for reporting and sorry for slow turnaround. The fix on release/3.3 is available at https://github.com/espressif/esp-idf/commit/1a657c255c8562d838f31c9c109dcfd0ea8b627b. Feel free to reopen if the issue still happens. Thanks.

closed 2020-05-28 by Alvin1Zhang