espressif/esp-idf #4063: esp-idf-tools-setup-2.0 fails exit code (1)

ccvelandres, 2019-09-14:
Been trying to install esp-idf using the setup on win 10 for about a day. I give up trying to troubleshoot myself :(. I'm running it as administrator but still fails saying access is denied. I haven't seen other issues similar to mine. What am I doing wrong? 

Log below
[Setup Log 2019-09-14 #012.txt](https://github.com/espressif/esp-idf/files/3612853/Setup.Log.2019-09-14.012.txt)

ccvelandres, 2019-09-14:
Just found #3819 
Kinda fixed it i think.

What to do: 
Ready to edit idf_tools_fallback.py by replacing line 261 `os.rename(path, tmp_path)` into 
```
for retry in range(100):
        try:
            os.rename(path, tmp_path)
            break
        except:
            print("rename failed, retrying...")
```
when the idf tool setup is running and saying its extracting stuff in .espressif directory. Since the setup is overwriting idf_tools_fallback.py so need to edit it after it overwrites it.

hazee007, 2019-09-18:
Hey man,
How do you start the installation process again after you edited that line..

hazee007, 2019-09-18:
anyways i edited the idf_tools_fallback.py as you said and try the installation process again but got the same error again.. 

here is my error file
[Setup Log 2019-09-18 #003.txt](https://github.com/espressif/esp-idf/files/3625423/Setup.Log.2019-09-18.003.txt)

ccvelandres, 2019-09-18:
@hazee007 you need to edit idf_tools_fallback.py **WHILE** the installation process is running (while it says extracting and while the progress bar is around the middle)

msysmilu, 2019-09-27:
Try uninstalling the driver that Windows installed by default. Worked for me.

closed 2019-09-30 by igrr

hrecently, 2020-02-18:
Hello, 
I tried to install the esp32 idf, assuming meanwhile no problems but I failed with the first step of the "getting startet", the Installation of the pre-requisites. I have this exit code problem described above. What is the next step after I downloaded the new Version? Starting the installation with the Link fails again ...
Thank you very much for help

igrr, 2020-02-18:
@hrecently please open a new issue, and attach the error log file which is created by the installer.