espressif/esp-idf #4064: Kconfig retative path includes

no1seman, 2019-09-15:
- Development Kit:  TTGO T8
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used: ESP32-D0WDQ6
- IDF version: v4.1-dev-256-g9f145ff16
- Build System: CMake
- Compiler version: xtensa-esp32-1.22.0-98-g4638c4f-5.2.0-20190827
- Operating System: Linux
- Power Supply:  (USB|external 5V)&Battery

## Problem Description
Simplified project folder structure:
[PROJECT ROOT]
   [components]
       [node]
          Kconfig.in
   [main]
       Kconfig.projbuild

Contents of Kconfig.projbuild:
```
...
 source "../components/node/Kconfig.in" 
...
```
When running 'idf.py menuconfig' from [PROJECT ROOT] config menu successfully built, but when i'm running 'idf.py build' I've got the following error:
IOError: /home/user/esp/beacon.new/main/Kconfig.projbuild:55: Could not open '../components/node/Kconfig.in' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).

If there any way to use [PROJECT ROOT] relative paths in Kconfig? I'm using the same set of my own components in different projects and wanna use them without changing paths in Kconfig.* files.

dobairoland, 2019-09-16:
Hi @no1seman. We will update ASAP to upstream `kconfiglib` so this behavior will change. I'm not sure what will be final accepted solution.

Could you define an environment variable, e.g. MY_PROJECT and then use it for sourcing Kconfigs as follows?

```
source "$MY_PROJECT/components/node/Kconfig.in" 
```

no1seman, 2019-09-16:
@dobairoland,

thank you for answer. 
I know that way but as I have already said I have several projects and it's little bit annoying to change environment variables every time I switched from one project to another. Is it possible to add global variable $PROJECT_ROOT (like $BUILD_ROOT) that will be pointed to the root of current project that will be avaible in all cmake scripts?

negativekelvin, 2019-09-16:
Since node is a component can't it have its own Kconfig.projbuild?

https://github.com/espressif/esp-idf/tree/master/examples/common_components/protocol_examples_common

no1seman, 2019-09-16:
@negativekelvin,
because it's way to make disorder like in IDF, where config values related to the same funtionality plased in 2-3-4 different places, for example debug settings are in: ESP32-specific, GDB stub, Core dump, Freertos, Log output, Bootloder config. I want to collect all regarded settings of my components into clear intuitive structure.

dobairoland, 2019-09-25:
Hi @no1seman. The `rsource` option from upstream kconfiglib (https://github.com/ulfalizer/Kconfiglib) will be available as soon as we update. This is a work-in-progress but I'll keep open this issue and notify you when it is available.

no1seman, 2019-09-25:
@dobairoland,
wow, much thanks for attention to my problem, will wait for update. It also will be great if you will add that case into manuals (https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html)

closed 2019-11-14 by espressif-bot

dobairoland, 2019-11-14:
@no1seman The "rsource" option is now available in the master branch from https://github.com/ulfalizer/Kconfiglib. There is one limitiation: it is not yet available from MSYS2 Mingw32 terminal. Please see the documentation of the upstream `kconfiglib` for the complete list of new features.

no1seman, 2019-11-14:
@dobairoland, 
thanks for information, will try a little bit later.

shwnyao-ultron, 2020-11-24:
Dear @dobairoland ,

Despite `rsource` has been working for me, I was forced to use `source` due my team's decision on staying backward-compatible with ESP8266 Kconfiglib, which does not support `rsource`.

What I would like to ask, is that is there a way to set environment variable, like `Project_PATH` in project Cmake, and use that variable in Kconfig? My code below, however, does not work for me.

CMakelists.txt (Project level)
```
cmake_minimum_required(VERSION 3.17)
set(ENV{PROJ_PATH} ${CMAKE_CURRENT_SOURCE_DIR})
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(MyProject)
```

$PROJ_PATH/componenets/mycomp/Kconfig.projbuild (Component level)
```
source "$PROJ_PATH/componenets/mycomp/subdir/Kconfig"
```

shwnyao-ultron, 2020-11-24:
Dear @dobairoland ,

Despite `rsource` has been working for me, I was forced to use `source` due my team's decision on staying backward-compatible with ESP8266 Kconfiglib, which does not support `rsource`.

What I would like to ask, is that is there a way to set environment variable, like `PROJ_PATH` in project Cmake, and use that variable in Kconfig? My code below, however, does not work for me. 

ESP_IDF: v4.1-rc

CMakelists.txt (Project level)
```
cmake_minimum_required(VERSION 3.17)
set(ENV{PROJ_PATH} ${CMAKE_CURRENT_SOURCE_DIR})
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(MyProject)
```

$PROJ_PATH/componenets/mycomp/Kconfig.projbuild (Component level)
```
source "$PROJ_PATH/componenets/mycomp/subdir/Kconfig"
```

dobairoland, 2020-11-24:
I'm linking https://www.esp32.com/viewtopic.php?f=13&t=18249&p=68052#p68052. I'll answer there.