espressif/esp-idf #4065: make monitor occurs error

happynet95, 2019-09-15:
Hi. I'm trying [gatts_client](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_client/main/gattc_demo.c) example and I typed `make flash` and `make monitor`.
Then my terminal shows me this error code.

However, my Arduino Serial Monitor performs well so I guess this is not a problem of my esp-idf code. Maybe something's wrong with python serial code(But still dont get what it is.)

Does anyone know how to deal with this?

```
--- idf_monitor on /dev/cu.usbserial-14630 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0xLB�=]ION_RESET),boot:0x13 (SPI_FAST_FLASH%==Q�
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:��0,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock V��2
load:0x3fff0018,len:4
load:0x3fff001c,len:6364
load:0x40078000,len:11304
loX����0080400,len:6700
entry 0x40080764
I (28) boot: ESP-IDF v4.0-dev-459-gba1ff1692-dirty 2nd stage bootloader
I (28) boot: compile time 17:32:38
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x20c9c (134300) map
I (141) esp_image: segment 1: paddr=0x00030cc4 vaddr=0x3ffbdb60 size=0x02b78 ( 11128) load
I (145) esp_image: segment 2: paddr=0x00033844 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/junghaemin/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (148) esp_image: segment 3: paddr=0x00033c4c vaddr=0x40080400 size=0x0c3c4 ( 50116) load
I (177) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x709fc (461308) map
0x400d0018: _flash_cache_start at ??:?

I (339) esp_image: segment 5: paddr=0x000b0a1c vaddr=0x4008c7c4 size=0x050c8 ( 20680) load
0x4008c7c4: esp_rom_spiflash_read at /Users/junghaemin/esp/esp-idf/components/spi_flash/spi_flash_rom_patch.c:539

I (358) boot: Loaded app from partition at offset 0x10000
I (358) boot: Disabling RNG early entropy source...
I (359) cpu_start: Pro cpu up.
I (362) cpu_start: Application information:
I (367) cpu_start: Project name:     gatt_client_demo
I (373) cpu_start: App version:      v4.0-dev-459-gba1ff1692-dirty
I (380) cpu_start: Compile time:     Sep 15 2019 17:32:41
I (386) cpu_start: ELF file SHA256:  87d0c746ceb23d8a...
I (392) cpu_start: ESP-IDF:          v4.0-dev-459-gba1ff1692-dirty
I (398) cpu_start: Starting app cpu, entry point is 0x40081030
0x40081030: call_start_cpu1 at /Users/junghaemin/esp/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (409) heap_init: Initializing. RAM available for dynamic allocation:
I (416) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (422) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (428) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (434) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (440) heap_init: At 3FFC8DA8 len 00017258 (92 KiB): DRAM
I (446) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (453) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (459) heap_init: At 4009188C len 0000E774 (57 KiB): IRAM
I (465) cpu_start: Pro cpu start user code
I (148) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (209) BTDM_INIT: BT controller compile version [484c057]
I (209) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/junghaemin/esp/esp-idf/tools/idf_monitor.py", line 136, in _run_outer
    self.run()
  File "/Users/junghaemin/esp/esp-idf/tools/idf_monitor.py", line 229, in run
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/Users/junghaemin/Library/Python/2.7/lib/python/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
SerialException: read failed: [Errno 6] Device not configured

```

gengyuchao, 2019-12-01:
Maybe your USB connector is not stable, you can try to replace it with a good USB cable

Alvin1Zhang, 2019-12-04:
@happynet95 Thanks for reporting. Would you please help try with another USB cable (suggested by @gengyuchao ) and help share if the issue still exists? Thanks

Alvin1Zhang, 2020-03-12:
@happynet95 Thanks for reporting. Will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2020-03-12 by Alvin1Zhang