espressif/esp-idf #4066: Error on Kconfig and when build a project on Ubuntu

noyelseth, 2019-09-15:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
   v4.1-dev-256-g9f145ff
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
when I build a application i getting error on Kconfig.

Generating esp32.project.ld
/home/noyelseth/esp/esp-idf/Kconfig:21: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename IDF_TARGET_ENV to IDF_TARGET (so that the symbol name matches the environment variable name).
Traceback (most recent call last):
File "/home/noyelseth/esp/esp-idf/tools/ldgen/ldgen.py", line 150, in
main()
File "/home/noyelseth/esp/esp-idf/tools/ldgen/ldgen.py", line 114, in main
sdkconfig = SDKConfig(kconfig_file, config_file)
File "/home/noyelseth/esp/esp-idf/tools/ldgen/sdkconfig.py", line 50, in init
self.config = kconfiglib.Kconfig(kconfig_file)
File "/usr/local/lib/python2.7/dist-packages/kconfiglib.py", line 1038, in init
self._parse_block(None, self.top_node, self.top_node)
File "/usr/local/lib/python2.7/dist-packages/kconfiglib.py", line 2875, in _parse_block
if self.srctree else "unset or blank"))
kconfiglib.KconfigError: /home/noyelseth/esp/esp-idf/Kconfig:67: '/home/noyelseth/esp/esp-idf/components/app_update/Kconfig.projbuild /home/noyelseth/esp/esp-idf/components/bootloader/Kconfig.projbuild /home/noyelseth/esp/esp-idf/components/esptool_py/Kconfig.projbuild /home/noyelseth/esp/esp-idf/components/partition_table/Kconfig.projbuild' not found (in 'source "$COMPONENT_KCONFIGS_PROJBUILD"'). Check that environment variables are set correctly (e.g. $srctree, which is set to '/home/noyelseth/esp/esp-idf'). Also note that unset environment variables expand to the empty string.
/home/noyelseth/esp/esp-idf/components/esp32/Makefile.projbuild:17: recipe for target '/home/noyelseth/esp/esp_workspace/hello_world/build/esp32/esp32.project.ld' failed
make: *** [/home/noyelseth/esp/esp_workspace/hello_world/build/esp32/esp32.project.ld] Error 1

[image]

- sdkconfig file
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3614257/sdkconfig.txt)

dobairoland, 2019-09-16:
Hi @noyelseth. Currently, ESP-IDF uses an incompatible `kconfiglib` with the upstream version. You've go this error because you have a globally installed package. The fix is not yet in the master branch. It will be fixed by commit 1dcdc56a7fec9cf200cf7fd4c2644aaf9cc32433 as soon as it get pushed.

A workaround until then is to remove kconfiglib (if you don't need it for anything else):
```
python -m pip uninstall kconfiglib
```

dobairoland, 2019-09-25:
Hi @noyelseth. The fix is in the `master` branch now. Is your issue fixed?

dobairoland, 2019-10-03:
I'm closing this for lack of activity. Please feel free to reopen it if the issue is still present, or create a new issue.

closed 2019-10-03 by dobairoland

ustccw, 2020-10-27:
> Hi @noyelseth. Currently, ESP-IDF uses an incompatible `kconfiglib` with the upstream version. You've go this error because you have a globally installed package. The fix is not yet in the master branch. It will be fixed by commit [1dcdc56](https://github.com/espressif/esp-idf/commit/1dcdc56a7fec9cf200cf7fd4c2644aaf9cc32433) as soon as it get pushed.
> 
> A workaround until then is to remove kconfiglib (if you don't need it for anything else):
> 
> ```
> python -m pip uninstall kconfiglib
> ```

It works, thanks

rustyx, 2021-06-14:
If this happens after an upgrade of ESP-IDF version, don't forget to `python -m pip install -r requirements.txt` after each upgrade.

NeilMacMullen, 2021-07-27:
A couple of extra comments for windows users since I just ran into this after upgrading to 3.3.0 of the  ESP32 platform then trying to revert to 1.11...

- PlatformIO uses it's own version of Python so make sure to run...

`c:\users\[your user name]\.platformio\penv\scripts\python.exe  -m pip uninstall kconfiglib`

(Similarly if you want to try the "requirements" step above you want something like...

`c:\users\[your user name]\.platformio\penv\scripts\python.exe  -m pip install -r c:\users\[your user name]\.platformio\packages\[package name]\requirements.txt`

- If you see a code-page error from python, it's probably because you are using the Windows Terminal Preview - try running from a Cmd prompt or old PowerShell IDE window.